tttapa/Control-Surface #149: Issue using CCPotentiometer over ESP32 BLE-MIDI

IsThisNameGoodEnough, 2020-03-27:
I've been testing the BLE-MIDI function for the ESP32 and seem to have problems sending commands using the CCPotentiometer class.  I've included a simple test program below.  If I run the code as shown I can see both the button and potentiometer events in the debug window of the Hairless MIDI app.  If I switch over to the bluetooth interface and run the code I can see the button event in MidiBerry, but no potentiometer events.  I've confirmed the lack of pot events on multiple computers and monitoring apps, so pretty confident it's real.  Any thoughts on what I might be doing wrong?

Thanks again for all the work you've put into this (and help troubleshooting), it's greatly appreciated!

```cpp
#include <Control_Surface.h>
#include <MIDI_Interfaces/BluetoothMIDI_Interface.hpp>

HairlessMIDI_Interface midi;
//BluetoothMIDI_Interface midi;
 
CCPotentiometer potentiometer = { 4, {MIDI_CC::Channel_Volume, CHANNEL_1}, };
CCButton button = { 0, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}, };

void setup() {
  Control_Surface.begin();
}
 
void loop() {
  Control_Surface.loop();
}
```

tttapa, 2020-03-27:
This is a hardware limitation of the ESP32: You cannot use ADC2 when the radio is in use. (GPIO4 is connected to ADC2.) You should be able to use analog pins connected to ADC1.

IsThisNameGoodEnough, 2020-03-27:
Wow, I thought I read through the ESP32 datasheet pretty thoroughly but I guess I missed that.  Thanks for the quick response!

closed 2020-03-27 by IsThisNameGoodEnough