tttapa/Control-Surface #150: Modifing RelativeCCsender message

sucrz, 2020-03-28:
Hi, 

I try to use the lib to control a Max4Live patch with rotary encoders but it don't work. 
I've test a couple of things and I found the problem using my beatstep pro : 

With the lib the encoder send two values, for example if the encoder is in "TwosComplement" it send a "127" to increment and "1" to decrement. 
the beatstep pro send two messages, first it send a "0" value message and then a "127" to increment and "1" to decrement. 

Whitout the "0" value message the "ctlin" object in M4L get stuck with the "127" or "1" message and doesn't increment or decrement. 

I know I'll have to modify to lib to match my need can someone point me to the file or line to modify to send a "0" message before the increment or decrement?

sucrz, 2020-03-28:
I found a solution, I've modify the "RelativeCCSender.hpp" file, I've add a "        Control_Surface.MIDI().sendCC(address, 0x00);" in the send void. I don't know if it the good way to do but It's works.

tttapa, 2020-03-28:
Yes, RelativeCCSender is the right file to edit. Glad to hear you got it working!

closed 2020-03-28 by tttapa