tttapa/Control-Surface #152: Map on bankable pot

diegogauffin, 2020-03-30:
hi! Hope you are fine ! 
I have question about this issue, i have a list of bankable potentiometers (on a MUX), and want to MAP one of them, how could i write the code for that ? I ll put here the code i am using now, you ll see the mapped potentiometer is not bankable. Thank you a lot ! 

```
#include <Control_Surface.h> 

 #include <Selectors/IncrementDecrementSelector.hpp>
 
USBMIDI_Interface midi;  
 
CD74HC4067 mux = {
  A3,       
  {2, 3, 4, 5} 
};

Bank<4> bank(2);

IncrementDecrementSelector<4> selector = {
    bank,       // Bank to manage
    {6, 16},     // push button pins (increment, decrement)
    Wrap::Wrap, // Wrap around
};
IncrementDecrementSelectorLEDs<4> bankSelector = {
  bank,
  {16, 6},         // button pins
  {0,1,19,20}, // LED pins
};

Bankable::CCPotentiometer myPots[] = {
    {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(0),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_1,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
    {                             // myPot1
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(1),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_2,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
    {                             // myPot2
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(2),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_3,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
     {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(3),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_4,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
     {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(4),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_5,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
    
    {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(5),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_6,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
     {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(6),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_7,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
    {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(7),                   // (2)
        {                             // (3)
            MIDI_CC::General_Purpose_Controller_8,      // (3.1)
            CHANNEL_1,                    // (3.2)
        },
    },
    {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(8),                   // (2)
        {                             // (3)
            MIDI_CC::Sound_Controller_1,      // (3.1)
            CHANNEL_2,                    // (3.2)
        },
    },
{                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(9),                   // (2)
        {                             // (3)
            MIDI_CC::Sound_Controller_2,      // (3.1)
            CHANNEL_2,                    // (3.2)
        },
    },
  {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(10),                   // (2)
        {                             // (3)
            MIDI_CC::Sound_Controller_3,      // (3.1)
            CHANNEL_2,                    // (3.2)
        },
    },
    {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(11),                   // (2)
        {                             // (3)
            MIDI_CC::Sound_Controller_6,      // (3.1)
            CHANNEL_2,                    // (3.2)
        },
    },
    {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(12),                   // (2)
        {                             // (3)
            MIDI_CC::Sound_Controller_4,      // (3.1)
            CHANNEL_2,                    // (3.2)
        },
    },
    {                             // myPot0
        {                             // (1)   
            bank,                         // (1.1)
            BankType::CHANGE_CHANNEL,     // (1.2)
        },
        mux.pin(13),                   // (2)
        {                             // (3)
            MIDI_CC::Sound_Controller_5,      // (3.1)
            CHANNEL_2,                    // (3.2)
        },
    },
    
    
    
  
};
CCPotentiometer potentiometer = {
  mux.pin(15) , {MIDI_CC::Channel_Volume, CHANNEL_1}
};

constexpr analog_t minimumValue = 20* 128;

constexpr analog_t maximumValue = 33* 128;

analog_t mappingFunction(analog_t raw) {
    
    raw = constrain(raw, minimumValue, maximumValue);
   
    return map(raw, minimumValue, maximumValue, 0, 16383);
   
};

Bankable::NoteButtonLatching switches[] = {
  {{bank, BankType::CHANGE_ADDRESS}, 7, MCU::MUTE_1},
  {{bank, BankType::CHANGE_ADDRESS}, 8, MCU::MUTE_2},
  {{bank, BankType::CHANGE_ADDRESS}, 9, MCU::MUTE_3},
  {{bank, BankType::CHANGE_ADDRESS}, 10, MCU::MUTE_4},
  {{bank, BankType::CHANGE_ADDRESS}, 18, MCU::SELECT_2},
  {{bank, BankType::CHANGE_ADDRESS}, 15, MCU::SELECT_1},
};

void setup (){
  potentiometer.map(mappingFunction);
Control_Surface.begin(); 

}
 
void loop() {
  Control_Surface.loop(); 
}
 

```

tttapa, 2020-03-30:
Hi,
There are two things you can do:

If you want to bank one of the potentiometers in the array, you can simply access it using the array subscript operator:

```cpp
void setup() {
  myPots[0].map(mappingFunction); // map only the first potentiometer in the array
  Control_Surface.begin();
}
```

You can also make the separate potentiometer bankable:
```cpp
Bankable::CCPotentiometer potentiometer = {
  {bank, BankType::CHANGE_CHANNEL},
  mux.pin(15) , {MIDI_CC::Channel_Volume, CHANNEL_1}
};

void setup () {
  potentiometer.map(mappingFunction);
  Control_Surface.begin(); 
}
```

diegogauffin, 2020-03-31:
Nice! Thankyou!