tttapa/Control-Surface #153: DAW Plugin Controller "ENCODER ISSUE with values"

forsakenrecordingstudios, 2020-03-30:
Hello,
   I am building a MIDI Controller to control my EQ Plugin. I use the plugin on multiple channels in the DAW. My issue is when i switch to a different DAW channel after making a adjustment, the encoder value carries over from the previous channel and throws the  new setting value right to the previous setting value from the last channel. The whole reason i got rotary encoders where so there was no conflict from DAW Channel to DAW Channel. Is there a fix to this? 

I only have one encoder written in the code at the moment since i am waiting for my MUX to arrive to add more.

This happens in both global and not global mode.

```cpp

// Your code here

#include <Encoder.h> // Include the Encoder library.

// Include the library
#include <Control_Surface.h>

// Instantiate a MIDI Interface to use
USBMIDI_Interface midi;

// Buttons
const int LF = A0;
const int MF = A1;
const int HF = A2;
const int LC = A3;
const int LMF = A4;
const int HMF = A5;
const int HC = A6;
const int ShowCurve = A7;
const int HighQuality = A8;
const int GainAuto = A9;

// LEDS
//const int LFLED = 44;
//const int MFLED = 45;
//const int HFLED = 46;
//const int LCLED = 47;
//const int LMFLED = 48;
//const int HMFLED = 49;
//const int HCLED = 50;
//const int ShowCurveLED = 51;
//const int HighQualityLED = 52;
//const int GainAutoLED = 53;

const pin_t ledPins[] =  {44, 45, 46, 47, 48, 49, 50, 51, 52, 53};

//Encoders
int LFQa = 2;
int LFQb = 3;

 

//Freq Activation
CCButtonLatched buttons[] =  {
  {LF, MCU::SOLO_1},
  {MF, MCU::SOLO_2},
  {HF, MCU::SOLO_3},
  {LC, MCU::SOLO_4},
  {LMF, MCU::SOLO_5},
  {HMF, MCU::SOLO_6},
  {HC, MCU::SOLO_7},
  {ShowCurve, MCU::SOLO_8},
  {HighQuality, MCU::MUTE_1},
  {GainAuto, MCU::MUTE_2},
  
};

CCAbsoluteEncoder enc = {
  {LFQa, LFQb},MIDI_CC::Sound_Controller_1,1,       // pins
           
};

  

// Initialize the Control Surface
void setup() {
 
 //LED Controll
 for (pin_t ledPin : ledPins)
    pinMode(ledPin, OUTPUT);

 //Encoders
    pinMode(LFQa, INPUT_PULLUP);
    pinMode(LFQb, INPUT_PULLUP);

  
  
  // Use the Mackie Control protocol for sending relative MIDI CC messages.
  RelativeCCSender::setMode (MACKIE_CONTROL_RELATIVE);
  Serial.begin(115200);
  Control_Surface.begin(); // Initialize Control Surface
}

//Led Controll
template <class T, size_t N> 
constexpr size_t len(const T (&)[N]) { return N; }

// Update the Control Surface
void loop() {

// Update the control surface
  Control_Surface.loop();

 //LED Controll
 static_assert(len(ledPins) == len(buttons), "Use the same number of LEDs as the number of buttons!");
  for (size_t i = 0; i < len(buttons); ++i)
    digitalWrite(ledPins[i], buttons[i].getState());
}

```

tttapa, 2020-03-30:
Hi,  
The problem is the use of the `CCAbsoluteEncoder` class, if you use a [`CCRotaryEncoder`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d62/classCCRotaryEncoder.html), this shouldn't be an issue, because it uses relative MIDI messages. `CCAbsoluteEncoder` keeps track of the position of the encoder, regardless of what channel is active.

If your DAW doesn't support relative CC modes, you'll have to hack your way around it by remembering the position for each channel separately, and then listen to the DAW to know what channel is active.

The solution will depend on what DAW you're using.

> I only have one encoder written in the code at the moment since i am waiting for my MUX to arrive to add more.

Sadly, encoders are not supported using multiplexers, because of their tight timing requirements. You can read more about it here: https://github.com/tttapa/Control-Surface/issues/140#issuecomment-602068153

Depending on what Arduino you're using, you'll probably find a solution that does work, but it's not included in the library because there's no one solution that works in all cases.

forsakenrecordingstudios, 2020-03-31:
Thank You,

  When using CCRotaryEncoder I get an error message.

could not convert '{LFQa, LFQb}' from '<brace-enclosed initializer list>' to 'CS::CCRotaryEncoder'

.
.
.
.

Also using MultiPlex for additional encoders, there are lots of guys doing this, one guy i know personally. Making the same controllers. I am now confused.

tttapa, 2020-03-31:
> When using CCRotaryEncoder I get an error message.
> 
> could not convert '{LFQa, LFQb}' from '' to 'CS::CCRotaryEncoder'

Please see this example: https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/d1f/RotaryEncoder_8ino-example.html

> Also using MultiPlex for additional encoders, there are lots of guys doing this, one guy i know personally. Making the same controllers. I am now confused.

Sure, you can do it, but it's not included in the library.  
If you combine the multiplexed encoders with other things like a couple of potentiometers or a display, you can run into timing issues.

forsakenrecordingstudios, 2020-03-31:
Okay,

   I am using Presonus Studio One, and for some reason i can not get a single encoder to increment, it is always just 2 values. I tried every relative. is there just something wrong with my code? Ive seen others use your library with studio one with no issues with this.

forsakenrecordingstudios, 2020-03-31:
I am really just wanting this to send generic MIDI CC messages of my choice.

`#include <Encoder.h> // Include the Encoder library.

// Include the library
#include <Control_Surface.h>

// Instantiate a MIDI Interface to use
USBMIDI_Interface midi;

// Buttons
const int LF = A0;
const int MF = A1;
const int HF = A2;
const int LC = A3;
const int LMF = A4;
const int HMF = A5;
const int HC = A6;
const int ShowCurve = A7;
const int HighQuality = A8;
const int GainAuto = A9;

// LEDS
//const int LFLED = 44;
//const int MFLED = 45;
//const int HFLED = 46;
//const int LCLED = 47;
//const int LMFLED = 48;
//const int HMFLED = 49;
//const int HCLED = 50;
//const int ShowCurveLED = 51;
//const int HighQualityLED = 52;
//const int GainAutoLED = 53;

const pin_t ledPins[] =  {44, 45, 46, 47, 48, 49, 50, 51, 52, 53};

//Encoders
int LFQa = 2;
int LFQb = 3;

 

//Freq Activation
CCButtonLatched buttons[] =  {
  {LF, MCU::SOLO_1},
  {MF, MCU::SOLO_2},
  {HF, MCU::SOLO_3},
  {LC, MCU::SOLO_4},
  {LMF, MCU::SOLO_5},
  {HMF, MCU::SOLO_6},
  {HC, MCU::SOLO_7},
  {ShowCurve, MCU::SOLO_8},
  {HighQuality, MCU::MUTE_1},
  {GainAuto, MCU::MUTE_2},
  
};

CCRotaryEncoder enc1 = {
  {LFQa, LFQb}, 
  MIDI_CC::Pan,
  1,
    
};

  

// Initialize the Control Surface
void setup() {
 
 //LED Controll
 for (pin_t ledPin : ledPins)
    pinMode(ledPin, OUTPUT);

 //Encoders
    pinMode(LFQa, INPUT_PULLUP);
    pinMode(LFQb, INPUT_PULLUP);

  
  
  // Use the Mackie Control protocol for sending relative MIDI CC messages.
  RelativeCCSender::setMode (TWOS_COMPLEMENT);
  Serial.begin(115200);
  Control_Surface.begin(); // Initialize Control Surface
  //Control_Surface.setMIDIInputCallbacks(channelMsgCallback, nullptr, nullptr); //MIDI Read
}

//Led Controll
template <class T, size_t N> 
constexpr size_t len(const T (&)[N]) { return N; }

// Update the Control Surface
void loop() {

// Update the control surface
  Control_Surface.loop();

 //LED Controll
 static_assert(len(ledPins) == len(buttons), "Use the same number of LEDs as the number of buttons!");
  for (size_t i = 0; i < len(buttons); ++i)
    digitalWrite(ledPins[i], buttons[i].getState());
}`

forsakenrecordingstudios, 2020-03-31:
`#include <Encoder.h> // Include the Encoder library.

// Include the library
#include <Control_Surface.h>

// Instantiate a MIDI Interface to use
USBMIDI_Interface midi;

// Buttons
const int LF = A0;
const int MF = A1;
const int HF = A2;
const int LC = A3;
const int LMF = A4;
const int HMF = A5;
const int HC = A6;
const int ShowCurve = A7;
const int HighQuality = A8;
const int GainAuto = A9;

// LEDS
//const int LFLED = 44;
//const int MFLED = 45;
//const int HFLED = 46;
//const int LCLED = 47;
//const int LMFLED = 48;
//const int HMFLED = 49;
//const int HCLED = 50;
//const int ShowCurveLED = 51;
//const int HighQualityLED = 52;
//const int GainAutoLED = 53;

const pin_t ledPins[] =  {44, 45, 46, 47, 48, 49, 50, 51, 52, 53};

//Encoders
int LFQa = 2;
int LFQb = 3;

 

//Freq Activation
CCButtonLatched buttons[] =  {
  {LF, MCU::SOLO_1},
  {MF, MCU::SOLO_2},
  {HF, MCU::SOLO_3},
  {LC, MCU::SOLO_4},
  {LMF, MCU::SOLO_5},
  {HMF, MCU::SOLO_6},
  {HC, MCU::SOLO_7},
  {ShowCurve, MCU::SOLO_8},
  {HighQuality, MCU::MUTE_1},
  {GainAuto, MCU::MUTE_2},
  
};

CCRotaryEncoder enc1 = {
  {LFQa, LFQb}, 
  MIDI_CC::Pan,
  1,
    
};

  

// Initialize the Control Surface
void setup() {
 
 //LED Controll
 for (pin_t ledPin : ledPins)
    pinMode(ledPin, OUTPUT);

 //Encoders
    pinMode(LFQa, INPUT_PULLUP);
    pinMode(LFQb, INPUT_PULLUP);

  
  
  // Use the Mackie Control protocol for sending relative MIDI CC messages.
  RelativeCCSender::setMode (TWOS_COMPLEMENT);
  Serial.begin(115200);
  Control_Surface.begin(); // Initialize Control Surface
  //Control_Surface.setMIDIInputCallbacks(channelMsgCallback, nullptr, nullptr); //MIDI Read
}

//Led Controll
template <class T, size_t N> 
constexpr size_t len(const T (&)[N]) { return N; }

// Update the Control Surface
void loop() {

// Update the control surface
  Control_Surface.loop();

 //LED Controll
 static_assert(len(ledPins) == len(buttons), "Use the same number of LEDs as the number of buttons!");
  for (size_t i = 0; i < len(buttons); ++i)
    digitalWrite(ledPins[i], buttons[i].getState());
}`

I am really just wanting this to send generic MIDI CC messages of my choice over USB

tttapa, 2020-03-31:
> I am using Presonus Studio One, and for some reason i can not get a single encoder to increment, it is always just 2 values. I tried every relative. is there just something wrong with my code? Ive seen others use your library with studio one with no issues with this.

If you're getting just two values, this indicates that your DAW is expecting absolute values. You have to select relative mode in Presonus as well.  
If Presonus doesn't support it, you can use `CCAbsoluteEncoder`.

> I am really just wanting this to send generic MIDI CC messages of my choice over USB

You can use [`Control_Surface.sendCC(address, value)`](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d45/classControl__Surface__.html#a82b919f54a86238cc55ae5ed7eeac578).

forsakenrecordingstudios, 2020-03-31:
while using CCAbsoluteEncoder my settings jump between DAW Channels which is no good,

is there an example of 'Control_Surface.sendCC(address, value)' being used, not to sure what this is?

forsakenrecordingstudios, 2020-03-31:
so you mean use " Control_Surface.sendCC(address, value); " instead of "  RelativeCCSender::setMode (TWOS_COMPLEMENT); " ?

tttapa, 2020-03-31:
See the last post in this thread: https://forums.presonus.com/viewtopic.php?t=17130&p=218622#p218622

forsakenrecordingstudios, 2020-03-31:
So it looks as if Studio one dont support Relative, so wouldn't there be a way to transmit back to the encoder in absolute to were it knows the new value when switching to another channel so it wont jump the setting?

tttapa, 2020-03-31:
> so wouldn't there be a way to transmit back to the encoder in absolute to were it knows the new value when switching to another channel so it wont jump the setting?

Yes, but then you still rely on Studio One transmitting the value to the Arduino. There's no standard way to "request" a value from the DAW.

It seems like Studio One has both options though (relative mode and re-transmitting), but you have to specify it in an XML control surface description. I've never used Studio One, but the link in my previous post seems to explain at least some of it.

tttapa, 2020-03-31:
I'm afraid I can't help you with that, I've never used Studio One.

forsakenrecordingstudios, 2020-04-01:
O well,

So back to the Control_Surface.sendCC(...) then. 

Would i need this code on every button and knob?

forsakenrecordingstudios, 2020-04-01:
I am not 100% sure what serial send and receive actually is. 

Im at least ecstatic i got the button and leds functioning properly lol.

tttapa, 2020-04-01:
I'm afraid `Control_Surface.sendCC(...)` isn't gonna help here. It's just a function that allows you to send custom MIDI control change messages. If you can't configure your DAW to interpret these messages correctly, there's nothing you can do, even if you have full control over what MIDI messages the Arduino is sending.

`CCButton`, `CCRotaryEncoder` and `CCAbsoluteEncoder` already make use of `sendCC` under the hood. If `CCButton` already works correctly, there's no reason to replace it with manual calls to `sendCC`.

To recap, there are 5 main MIDI Control Change modes:

- Absolute 7-bit
- Absolute 14-bit
- Relative two's complement
- Relative sign+magnitude
- Relative binary offset

The default that's used for "MIDI learn" or manual MIDI mappings in most DAWs is absolute 7-bit. Absolute 14-bit is not really used for DAWs, but it is used for some synthesizers.  
The three relative modes send relative deltas, not absolute positions. This means they don't jump when changing settings/tracks/plugins. When sending deltas, you need some way to encode negative values, when the knob is turned to the left. There are three ways to encode the sign of a number, and hence three relative modes.  
There's no way for a MIDI receiver to distinguish between relative and absolute modes, the MIDI messages are the same, they just have a different meaning. This means that you have to select the right mode in your DAW, otherwise, it simply cannot work.

forsakenrecordingstudios, 2020-04-01:
Thank You So Much! issue resolved, that Studio One Fix worked editing the XML file,

closed 2020-04-01 by forsakenrecordingstudios