tttapa/Control-Surface #154: Bank para CCAbsoluteEncoder.

Joshbas1, 2020-03-31:
hola Pieter, tu libreria es fantastica, felicitaciones.
Elmotivo de este hilo es si se pudiera bankear la funcion CCAbsoluteEncoder, ya que esta funciona para cualquier mnsaj midi y mackie, mientras que CCRotaryEncoder no esta funcionando como deberia, he probado todos los ejemplos, cambiado el relativesender y no funciona, su comportiento es extraño,al girar solo da tres estados minimo, centro o maximo, no vasubiendo de uno en uno de 0 a 127, se queda en 0, 63 o 127. Podrias revisar este bug. 
Muchas gracias, sigue adelante.

tttapa, 2020-03-31:
> Elmotivo de este hilo es si se pudiera bankear la funcion CCAbsoluteEncoder, ya que esta funciona para cualquier mnsaj midi y mackie

Yes, this is possible, but it's not implemented yet. I'll see if I can add it later today.

> mientras que CCRotaryEncoder no esta funcionando como deberia, he probado todos los ejemplos, cambiado el relativesender y no funciona, su comportiento es extraño,al girar solo da tres estados [...]

This is the expected behavior, CCRotaryEncoder sends relative MIDI messages. You have to select relative mode in your DAW. If you select MCU mode in your DAW, it should work automatically, but you have to use the right CC addresses ([`MCU::V_POT_n`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/dc7/namespaceMCU.html#a9400848eec9735623e1502aef5bb7cb3)), and you have to [use the right relative mode](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/d1f/RotaryEncoder_8ino-example.html#a3) (`relativeCCmode::MACKIE_CONTROL_RELATIVE`).

You can find the expected values here: https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d24/RelativeCCSender_8hpp.html#a793b4d05f50a2c3af3ddeeef50e7a71d

Joshbas1, 2020-03-31:
Con CCRotaryEncoder va perfecto en los V_Pot, pero si en su lugar pongo VOLUME_n ya no funciona a pesar de seleccionar MCU en el daw (probado en ableton, fl studio, studio one) y de usar  (relativeCCmode::MACKIE_CONTROL_RELATIVE).   Mientras que si uso CCAbsoluteEncoder trabaja perfecto.
Si pudieses agregarle el bank seria grandioso. Estare a la espectativa de su implementacion. Agradezco mucho tu tiempo y trabajo en esta super libreria.

tttapa, 2020-03-31:
> Con CCRotaryEncoder va perfecto en los V_Pot, pero si en su lugar pongo VOLUME_n ya no funciona a pesar de seleccionar MCU en el daw (probado en ableton, fl studio, studio one) y de usar (relativeCCmode::MACKIE_CONTROL_RELATIVE). Mientras que si uso CCAbsoluteEncoder trabaja perfecto.

That's because the MCU protocol only describes 8 encoders, VOLUME_n is not an encoder in the MCU specification.

Most DAWs I've worked with allow you to select relative mode in the MIDI mapping. For example, in Reaper and Traktor (two that I happen to have screenshots for):
[image]
[image]

Joshbas1, 2020-03-31:
Buscare más a fondo las configuraciones de mi daw, sin embargo te agradeceria la implementacion de Bank para CCAbsoluteEncoder.

Joshbas1, 2020-04-01:
Hola, estuve investigando en fl studio y ableton y no permiten mas que seleccionar el protocolo o script para funcionar el controlador midi. Por lo que estuve tratando de modificar la libreria para añadir banco a CCAabsoluteEncoder pero al compilar me tira muchos errores.

tttapa, 2020-04-01:
I'm pretty busy at the moment, and I need to do some cleanup before I can add the Bankable::CCAbsoluteEncoder. I could maybe post a quick-and-dirty solution here.

Joshbas1, 2020-04-02:
Vale, te entiendo, si hay una solucion temporal y rapida que tengas a mano bienvenida sea, si no es asi esperaré a la implementacion

tttapa, 2020-04-04:
I've pushed some commits to the `share-encoder` branch, you can pull it using git, or you can [download it as a ZIP](https://github.com/tttapa/Control-Surface/archive/share-encoder.zip). It is very important that you use my patched fork of the Encoder library: https://github.com/tttapa/Encoder  
Delete the old version of the Encoder library in your `~/Arduino/libraries` folder, and replace it with this one.

This is the code I used to test it:
```cpp
#include <Encoder.h>
#include <Control_Surface.h>

USBDebugMIDI_Interface midi;

Bank<4> bank;
IncrementSelector<4> sel = {bank, 4};
Bankable::CCAbsoluteEncoder<4> enc = {
  bank, 
  {2, 3},
  {7, CHANNEL_1},
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```