tttapa/Control-Surface #156: Include MIDIUSB Library in Conrtol Surface

kentforth, 2020-04-03:
Hello Pieter!
If I remove library MIDIUSB from libraries I get this error

```
In file included from C:\Users\kentforth\Documents\Arduino\libraries\Control-Surface-master\src\MIDI_Interfaces\USBMIDI\USBMIDI.cpp:32:0:

C:\Users\kentforth\Documents\Arduino\libraries\Control-Surface-master\src\MIDI_Interfaces\USBMIDI\USBMIDI_MIDIUSB.ipp:2:21: fatal error: MIDIUSB.h: No such file or directory
```

If I need to use encoders I have to download library for encoders.

**Can you include all necessary libraries to Control Surface library, so people don't need to install all other libraries except your?**

tttapa, 2020-04-04:
Including them as a copy would mean that I have to update and maintain them manually, which is not something I want to do.  
Including them as a git submodule would be easier to maintain, but this is not supported by the Arduino library system, so I would still have to fork it to fix include paths etc.

Personally, I don't think it's worth the hassle. You can install both the MIDI USB library and the Encoder library in a couple of clicks using Arduino's library manager.
MIDIUSB and Encoder are both non-essential libraries. Only a limited number of boards requires the MIDIUSB library, and you only need the Encoder library if you want to use encoders.

Besides, where does this end? Should I include the FastLED library, Adafruit_GFX and Adafruit_SSD1306 as well? All of these libraries are required for some optional modules of the library.

Finally, while both the MIDIUSB and the Encoder library are released under free and open-source licenses, I don't think just taking their work and pasting it into my library does the original authors justice.

kentforth, 2020-04-04:
Thank you for your answer.
You are totally right, I didn't think that it would be easy enough to do that.
Your approach is best variant in this case.

closed 2020-04-04 by kentforth