tttapa/Control-Surface #158: Does the library support I2C multiplexers?

d1One, 2020-04-03:
Hello, sorry if this is not the right place to ask but I am wondering if the library supports I2C multiplexers? 
Such as a TCA9548A, basically to be able to use 4 SSD1306 oled screens instead of 2.

tttapa, 2020-04-04:
Displays are managed through `DisplayInterface`s. You can easily change how the data is actually written to the display by overriding the `display` method:

https://github.com/tttapa/Control-Surface/blob/6ef0fc9e3cf15830989c2f0f75880261be74cfb7/src/Display/DisplayInterfaces/DisplayInterfaceSSD1306.hpp#L20

For example (incomplete and untested)

```cpp
#include <Display/DisplayInterfaces/DisplayInterfaceSSD1306.hpp>

class MySSD1306_DisplayInterface : public SSD1306_DisplayInterface {
 public:
  MySSD1306_DisplayInterface(Adafruit_SSD1306 &display, uint8_t muxCh)
    : SSD1306_DisplayInterface(display), muxCh(muxCh) {}
 
  void begin() override {
    selectMux();
    // Initialize the Adafruit_SSD1306 display
    if (!disp.begin())
      FATAL_ERROR(F("SSD1306 allocation failed."), 0x1306);
 
    // If you override the begin method, remember to call the super class method
    SSD1306_DisplayInterface::begin();
  }

  void display() override {
    selectMux();
    SSD1306_DisplayInterface::display();
  }
 
  void drawBackground() override { disp.drawLine(1, 8, 126, 8, WHITE); }
 
 private:
  uint8_t muxCh;

  void selectMux() {
    // Implement this, use I2C to select the right multiplexer channel
    Wire.write(...);
    Wire.write(muxCh);
  }
} display = ssd1306Display;
```
In theory, only the `begin` and `display` methods communicate with the display over I2C, so these are the only methods that you have to override and add `selectMux()` to.

anikhossain285, 2020-07-13:
I tried the above solution but nothing shows on display. I have tried adding Serial.print() to see whether it's selecting mux or not. But nothing works  
```
#include <Control_Surface.h> // Include the Control Surface library
// Include the display interface you'd like to use
#include <Display/DisplayInterfaces/DisplayInterfaceSSD1306.hpp>
#include <Wire.h>
// ----------------------------- MIDI Interface ----------------------------- //
USBMIDI_Interface midi;

// ----------------------------- Display setup ------------------------------ //
constexpr uint8_t SCREEN_WIDTH = 128;
constexpr uint8_t SCREEN_HEIGHT = 64;
constexpr int8_t OLED_RESET = -1; // Use the external RC circuit for reset
Adafruit_SSD1306 ssd1306Display_L(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
Adafruit_SSD1306 ssd1306Display_R(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
Adafruit_SSD1306 ssd1306Display_R2(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET); //3rd screen
Adafruit_SSD1306 ssd1306Display_R3(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET); //4th screen

// --------------------------- Display interface ---------------------------- //
constexpr int8_t muxCh_L = 0; // multiplexer channel for Left Screen
constexpr int8_t muxCh_R = 1; // multiplexer channel for Right Screen
constexpr int8_t muxCh_R2 = 2; //3rd screen
constexpr int8_t muxCh_R3 = 3; //4th screen

class MySSD1306_DisplayInterface : public SSD1306_DisplayInterface {
  public:
    MySSD1306_DisplayInterface(Adafruit_SSD1306 &display, uint8_t muxCh)
      : SSD1306_DisplayInterface(display), muxCh(muxCh) {}

    void begin() override {
      selectMux();
      // Initialize the Adafruit_SSD1306 display
      if (!disp.begin())
        FATAL_ERROR(F("SSD1306 allocation failed."), 0x3C);

      SSD1306_DisplayInterface::begin();
      //SSD1306_DisplayInterface::begin(SSD1306_SWITCHCAPVCC, 0x3C);
    }

    void display() override {
      selectMux();
      SSD1306_DisplayInterface::display();
    }

    void drawBackground() override {
      selectMux();
      disp.drawLine(1, 8, 126, 8, WHITE);
    }

  private:
    uint8_t muxCh;
    //how the heck do i make this work :(
    void selectMux() {
      Wire.beginTransmission(0x70);
      Wire.write(1 << muxCh);
      Wire.endTransmission();
      Serial.print("Selected mux:");
      Serial.print(muxCh);
    }
} display_L = {ssd1306Display_L, muxCh_L}, display_R = {ssd1306Display_R, muxCh_R}, display_R2 = {ssd1306Display_R2, muxCh_R2}, display_R3 = {ssd1306Display_R3, muxCh_R3};

// ------------------------------- Bank setup ------------------------------- //
// ========================================================================== //

Bank<2> bank(4); // Create a new bank with four tracks per bank

// Create a new bank selector to control the bank using a push button
IncrementSelector<2> bankselector = {bank, 5};

// -------------------------- MIDI Input Elements --------------------------- //
[... 153 lines cut ...]
  select_chnl(3);
 // Wire.setClock(3400000); // Set the I²C clock to 1.8 MHz for faster refresh / 1800000 was the default, changed to 3400000 not sure if this even works this fast.
  ssd1306Display_R3.setRotation(0); // Normal screen orientation.
}
void select_chnl(int ch)
{
  Wire.beginTransmission(0x70);
  Wire.write(1 << ch);
  Wire.endTransmission();
  Serial.print("Selected mux:");
  Serial.print(ch);
}
void setup() {
  Serial.begin(9600);
  initializeDisplay();     // Start the OLED display
  Control_Surface.begin(); // Initialize Control Surface
}

// ---------------------------------- Loop ---------------------------------- //
// ========================================================================== //

void loop() {
  Control_Surface.loop(); // Refresh all elements
}
```

tttapa, 2020-07-13:
I don't have the necessary hardware to help you debugging, I only have a single I2C display, I always use SPI OLED displays for my projects.

That being said, the first step should be to get this working without the Control Surface library. Use just the Adafruit library directly and maybe a library that controls the mux if it requires one, and do something extremely simple, like printing "1" to the first display, "2" to the second, etc., just enough to show that it's working.

Once that's working, I can show you how to do the same thing using the Control Surface library.

You might want to try using IO pins to drive the reset lines of the displays instead of using an RC circuit, it seems to be more reliable if you have multiple displays. First try a separate IO pin for the reset line of each OLED.

The code you posted tries to initialize the displays twice, once in the `MySSD1306_DisplayInterface::begin()` method, and once in the `initializeDisplay()` function. You don't need the `initializeDisplay()` function, by calling `Control_Surface.begin()`, you're already implicitly calling the `MySSD1306_DisplayInterface::begin()` method.  
Initializing the displays multiple times will probably do more harm than good.

Good luck!

anikhossain285, 2020-07-13:
I have tried single initialisation too. I will remove `initializeDisplay()` I have been successful in using the mux for the four displays. I got it to work for a single display without mux using control surface library. The only problem occurs when I try to use multiple displays in control surface library. Maybe I am doing something wrong in Display interface.

Code for working display with mux 
```
//#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
#define OLED_RESET -1 // Reset pin # (or -1 if sharing Arduino reset pin)
#define TCAADDR 0x70 // Define address for TCA

// Create an object for each OLED display
Adafruit_SSD1306 display0(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
Adafruit_SSD1306 display1(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
Adafruit_SSD1306 display2(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
Adafruit_SSD1306 display3(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

void TCA9548A(uint8_t bus)
{
  Wire.beginTransmission(TCAADDR); //TCA9548A address is 0x70
  Wire.write(1 << bus);         // send byte to select bus
  Serial.println("TCA Loop "); Serial.println(bus);
  Wire.endTransmission();
}

void setup() {
  Serial.begin(9600);
//  Serial.println(("Start"));

  // start wire library for I2C
  Wire.begin();

//initialize OLED at ox3C (set multiplexer to channel 1 and initialize OLED-0 with I2C addr 0x3C
    TCA9548A(0);
    display0.begin(SSD1306_SWITCHCAPVCC, 0x3C);
//    Serial.println(("Initialize 0"));

    TCA9548A(1);
    display1.begin(SSD1306_SWITCHCAPVCC, 0x3C);
//    Serial.println(("Initialize 1"));

    TCA9548A(2);
    display1.begin(SSD1306_SWITCHCAPVCC, 0x3C);
//    Serial.println(("Initialize 2"));

    TCA9548A(3);
    display1.begin(SSD1306_SWITCHCAPVCC, 0x3C);
//    Serial.println(("Initialize 3"));
}

void oled1() {
TCA9548A(0);
display0.clearDisplay();//clear display
display0.setTextColor(WHITE);//set color (always white)
display0.setTextSize(2);//set font size
display0.setCursor(0,0);//set the cursor coordinates
display0.print("disp 1");
display0.display();
//Serial.println(("Write Display One"));
delay(1000);}

void oled2() {
TCA9548A(1);
display1.clearDisplay();
display1.setTextColor(WHITE);//set color (always white)
display1.setTextSize(2);//set font size
display1.setCursor(0,0);//set the cursor coordinates
[... 92 lines cut ...]
  {display, VUMeters[0], {2 + 16 * 0, 50}, 12, 3, 1, WHITE},
  {display, VUMeters[1], {2 + 16 * 1, 50}, 12, 3, 1, WHITE},
  {display, VUMeters[2], {2 + 16 * 2, 50}, 12, 3, 1, WHITE},
  {display, VUMeters[3], {2 + 16 * 3, 50}, 12, 3, 1, WHITE},
  {display, VUMeters[4], {2 + 16 * 4, 50}, 12, 3, 1, WHITE},
  {display, VUMeters[5], {2 + 16 * 5, 50}, 12, 3, 1, WHITE},
  {display, VUMeters[6], {2 + 16 * 6, 50}, 12, 3, 1, WHITE},
  {display, VUMeters[7], {2 + 16 * 7, 50}, 12, 3, 1, WHITE},
};

 
// --------------------------------- Setup ---------------------------------- //
 
void setup() {
    initializeDisplay();     // Start the OLED display
    Control_Surface.begin(); // Initialize Control Surface
}

// ---------------------------------- Loop ---------------------------------- //

void loop() {
    Control_Surface.loop(); // Refresh all elements
}

```

tttapa, 2020-07-13:
In your code without Control Surface, you're calling
```cpp
    display0.begin(SSD1306_SWITCHCAPVCC, 0x3C);
```
But in your Control Surface code, you're calling it without any arguments:
```cpp
    void begin() override {
      selectMux();
      // Initialize the Adafruit_SSD1306 display
      if (!disp.begin()) //                                                              ← No arguments here 
        FATAL_ERROR(F("SSD1306 allocation failed."), 0x3C);

      SSD1306_DisplayInterface::begin();
      //SSD1306_DisplayInterface::begin(SSD1306_SWITCHCAPVCC, 0x3C);
    }
```

anikhossain285, 2020-07-13:
I tried this too. I am not sure it's a correct format or not.

```
void begin() override {
      selectMux();
      // Initialize the Adafruit_SSD1306 display
      if (!disp.begin(SSD1306_SWITCHCAPVCC,0x3C))
        FATAL_ERROR(F("SSD1306 allocation failed."), 0x3C);

      SSD1306_DisplayInterface::begin();
      //SSD1306_DisplayInterface::begin(SSD1306_SWITCHCAPVCC, 0x3C);
    }
```

tttapa, 2020-07-13:
I think the problem is that you forgot to call `Wire.begin()` before setting the mux channel. If you only use a single display, the `Adafruit_SSD1306::begin()` method will call `Wire.begin()` internally, but since you're talking I2C to the mux _before_ initializing the display, you have to initialize the I2C interface first, manually.

The following works for me (without a mux, both displays in the code draw to the same physical display):
```cpp
#include <Control_Surface.h> // Include the Control Surface library
// Include the display interface you'd like to use
#include <Display/DisplayInterfaces/DisplayInterfaceSSD1306.hpp>

// ----------------------------- MIDI Interface ----------------------------- //

USBMIDI_Interface midi;

// ----------------------------- Display setup ------------------------------ //

constexpr uint8_t SCREEN_WIDTH = 128;
constexpr uint8_t SCREEN_HEIGHT = 64;
 
constexpr int8_t OLED_reset = -1; // Use the external RC circuit for reset
 
// Instantiate the displays
Adafruit_SSD1306 ssd1306Display_L = {
  SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_reset,
};
Adafruit_SSD1306 ssd1306Display_R = {
  SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_reset,
};

constexpr uint8_t muxCh_L = 0; // multiplexer channel for Left Screen
constexpr uint8_t muxCh_R = 1; // multiplexer channel for Right Screen

// --------------------------- Display interface ---------------------------- //

class MySSD1306_DisplayInterface : public SSD1306_DisplayInterface {
  public:
    MySSD1306_DisplayInterface(Adafruit_SSD1306 &display, uint8_t muxCh)
      : SSD1306_DisplayInterface(display), muxCh(muxCh) {}

    void begin() override {
      selectMux();
      
      // Initialize the Adafruit_SSD1306 display
      if (!disp.begin(SSD1306_SWITCHCAPVCC, 0x3C))
        FATAL_ERROR(F("SSD1306 allocation failed."), 0x1306);

      SSD1306_DisplayInterface::begin();

      disp.setTextColor(WHITE);
      disp.setTextSize(2);
    }

    void display() override {
      selectMux();
      SSD1306_DisplayInterface::display();
      delay(2000); // slow down, easier to debug
    }

    void drawBackground() override {}

  private:
    uint8_t muxCh;

    void selectMux() {
      Wire.beginTransmission(0x70);
      Wire.write(1 << muxCh);
      Wire.endTransmission();
      Serial.print("Selected mux:");
      Serial.println(muxCh);
    }
} display_L = {ssd1306Display_L, muxCh_L},
  display_R = {ssd1306Display_R, muxCh_R};

// ---------------------------- Display Elements ---------------------------- //

// Simple display element that just prints some text
struct TestDisplayElement : DisplayElement {
  TestDisplayElement(DisplayInterface &display, const char *message)
   : DisplayElement(display), message(message) {}
   
  void draw() override {
    getDisplay().setCursor(8, 8);
    getDisplay().print(message);
  }

  const char *message;
} dispElem_L = {display_L, "LEFT"}, 
  dispElem_R = {display_R, "RIGHT"};

// --------------------------------- Setup ---------------------------------- //

void setup() {
  Serial.begin(115200);
  Wire.begin(); // This is important! Otherwise, you can't select the mux channel
  Control_Surface.begin(); // Initialize Control Surface
}

// ---------------------------------- Loop ---------------------------------- //

void loop() {
  Control_Surface.loop(); // Refresh all elements
}
```

anikhossain285, 2020-07-14:
I knew it, it would be something small like that. Thank you so much. It's working now.