tttapa/Control-Surface #159: How to use MCU?

kentforth, 2020-04-04:
I didn't find examples with MCU

**How can I use MCU in Reaper DAW?**

**Can you show me a sketch  example?**

tttapa, 2020-04-04:
What MCU features do you want to use?

The MCU protocol is just a protocol on top of MIDI that was designed specifically for the Mackie Control Universal.  
It assigns special meaning to some note numbers etc. and it defines some MIDI messages for VU meters, LED rings and track names.

The best overview I could find about the MCU protocol (it's proprietary, after all), is this user manual: https://images.thomann.de/pics/prod/151261_manual.pdf

Most of it is implemented in the Control Surface library. You can find the documentation here: https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/dc7/namespaceMCU.html

kentforth, 2020-04-04:
I thought MCU is like for example : if I use mcu: scrub it means scrub will be understandable in all DAWs. 
If I assign MCU message on any element will a daw understand that message as midi message?

tttapa, 2020-04-05:
Yes, you can do that.

kentforth, 2020-04-05:
Thanx a lot!

closed 2020-04-05 by kentforth