tttapa/Control-Surface #162: Bank  and program change line 6 load setlist

pandeagus, 2020-04-08:
Hi.. 
[image]
I need help... I dont know how to acces setlist using combine bank change cc0(bank msb) and cc32( LSB).. can you teach me the code ?

tttapa, 2020-04-08:
This should do the trick:

```cpp
uint16_t bank = 1234;
Control_Surface.sendCC(MIDI_CC::Bank_Select, (bank >> 7) & 0x7F);
Control_Surface.sendCC(MIDI_CC::Bank_Select_LSB, bank & 0x7F);
```

pandeagus, 2020-04-08:
[image]
 
Im sorry maybe  i dont understand what you mean.. can you help me..

pandeagus, 2020-04-08:
[image]
Like this get error... can you help me

tttapa, 2020-04-08:
Please don't post images of text. Post text.

The first error is because you're just throwing together some code. You cannot use CCButton like that, you have to use the right arguments. Please see the [documentation](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/daa/classCCButton.html#a8ff77ca457f85462169de01ca7e02238).

The second error seems to indicate that your Arduino installation is messed up, but it's impossible to help you without seeing the full error message.  
Also, you cannot use `Control_Surface.sendCC(...)` without first calling `Control_Surface.begin()`.

pandeagus, 2020-04-08:
yups thanks for help the second error... after i upload the code nothing happen... may be i wrong

code :
#include <Control_Surface.h>
auto &serial = Serial;
SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD};

uint16_t bank = 1234;

void setup() {
  Control_Surface.begin();
}
void loop() { 
   Control_Surface.sendCC(MIDI_CC::Bank_Select, (bank>>7) & 0x7F);
 Control_Surface.sendCC(MIDI_CC::Bank_Select_LSB, bank & 0x7F);
}

i change bank = 1; still nothing happen...
sorry take your time for help me.

tttapa, 2020-04-08:
What do you expect to happen? It just selects a bank of presets/programs. You'll probably still have to send a Program Change afterwards. For example: https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d7f/Program-Changer_8ino-example.html