tttapa/Control-Surface #163: Bankable potentiometers behavior - Disable midi output until it reaches the same value

wayna, 2020-04-08:
Hello!

I'm trying to build a little midi controller using your awesome library (thank you for all the amazing work by the way).
I would like to build a simple fader control surface, with 10 bankable faders, and with two buttons to change the banks, so i looked into your examples and found the right code to it.

I'm trying to figure out a solution for the following problem:
Let's say i just use 1 bankable potentiometer (two bank, banktype channel) for the example

- i put up the potentiometer to 127 in channel 1
- I change the Channel to 2
- I put the potentiometer to 57
- I put back the channel to 1
- When i move the potentiometer, the value jumps from 127 to 57

Is there a way to change this behavior? 
I would like the value to change only if the potentiometer has been put to the same position as it was before in this bank, so there is no jump in values.

I tried to look into the documentation but didn't found out what could solve this problem, but my coding level isn't really good so maybe i missed something.

Hope this post is understandable, sorry for my poor english
Many Thanks

Wayna

tttapa, 2020-04-08:
That's a good idea.  
It's not supported out of the box, but I'll look into adding it.

wayna, 2020-04-08:
Oh great thanks!
Looking forward to this new feature.

Thanks again for all the great work you put together!

tttapa, 2020-04-10:
I added it yesterday, with an example here: https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/da4/Bankable-Smart-Control-Change-Potentiometer_8ino-example.html

What do you think?

wayna, 2020-04-10:
This is great, works perfectly fine!

Thank you!

closed 2020-04-20 by wayna