tttapa/Control-Surface #168: CC Button single value

crazymaca69, 2020-04-14:
I'd like to have two CC Buttons both mapped to CC9, one always sending value '0' and one always sending value '127'

Is this possible?

crazymaca69, 2020-04-14:
and how do you map a momentary cc button? so when I press, it sends value 127 and when I release it sends value 0

tttapa, 2020-04-15:
> I'd like to have two CC Buttons both mapped to CC9, one always sending value '0' and one always sending value '127'

You can define a custom MIDI sender to do this:
```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

// A custom sender to use later. It has to declare two methods:
// - sendOn(MIDIAddress): will be called when the button is pressed
// - sendOff(MIDIAddress): will be called when the button is released
class CustomCCSender {
  public:
    CustomCCSender(uint8_t onValue)
      : onValue(onValue) {}

    void sendOn(MIDIAddress address) {
      Control_Surface.sendCC(address, onValue);
    }

    void sendOff(MIDIAddress) {
      // nothing here
    }

  private:
    uint8_t onValue;
};

// Now tell the MIDIButton class template (included with the Control
// Surface library) that it has to use your custom sender class.
//
// We wrap it in another class so we can easily construct it later, 
// without having to write `MIDIButton<CustomCCSender>` all the time,
// and so we have more control over the constructor arguments.
// The colon (:) indicates inheritance.
struct CustomCCButton : MIDIButton<CustomCCSender> {
  // Constructor
  CustomCCButton(pin_t pin, MIDIAddress address, uint8_t onValue = 0x7F)
    : MIDIButton(pin, address, {onValue}) {}
  //  ^~~~~~~~~~ Initialization of the base class MIDIButton
};

// Now we can instantiate an object of our custom class.
// The three arguments match the ones of the CustomCCButton
// constructor we wrote a couple of lines back.
CustomCCButton button1 = {
  2,      // button pin
  0x09,   // MIDI address
  0x00,   // on value
};
CustomCCButton button2 = {
  2,      // button pin
  0x09,   // MIDI address
  0x7F,   // on value
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

> and how do you map a momentary cc button? so when I press, it sends value 127 and when I release it sends value 0

Isn't this what CCButton does already?

crazymaca69, 2020-04-15:
Thanks! could you possibly drop some example data in there. I am not an expert and I'm not sure which parts of the code I edit for my own purposes.

crazymaca69, 2020-04-15:
Like straight away I'm getting the following error
Infinity_Remote:34: error: 'MIDIAddress' has not been declared
     void sendOn(MIDIAddress address) {
                 ^
Infinity_Remote:38: error: 'MIDIAddress' has not been declared
     void sendOff(MIDIAddress) {
                  ^

tttapa, 2020-04-15:
You'll either have to upgrade to the master version, or replace `MIDIAddress` with `MIDICNChannelAddress`.

crazymaca69, 2020-08-23:
Just getting back round to this and now have it working. I have an RGB LED attached to pins 21,22 and 23.
I'd like to change colour to red when CC9 with a value of 0 is selected and blue when CC9 with a value of 127 is selected.

`#include <Encoder.h>
#include <Control_Surface.h>

HardwareSerialMIDI_Interface midi = {Serial1, MIDI_BAUD};

CCButton TopRow[] = {
  {2, {30, CHANNEL_5}}, //CC30 Pitch Down 
  {3, {29, CHANNEL_5}}, //CC29 Pitch Up
  {4, {78, CHANNEL_5}}, //CC78 Reverse
  {5, {75, CHANNEL_5}}, //CC75 Stutter
 /* {7, {24, CHANNEL_5}}, //CC24 Loop 1 Record*/
  {8, {79, CHANNEL_5}}, //CC79 Undo
  {9, {89, CHANNEL_5}}, //CC89 Erase
};

/*
CCButtonLatched ES8[] = {
  {6, {9, CHANNEL_5}}, //CC9 Loop Aging
};
*/
CCPotentiometer fader[] = {
  {A0, {7, CHANNEL_5}}, //CC07 Global Volume
};

const int speedMultiplier = 1; //increase this number if the fader does not respond at the desired speed.

// A custom sender to use later. It has to declare two methods:
// - sendOn(MIDIAddress): will be called when the button is pressed
// - sendOff(MIDIAddress): will be called when the button is released
class CustomCCSender {
  public:
    CustomCCSender(uint8_t onValue)
      : onValue(onValue) {}

    void sendOn(MIDIAddress address) {
      Control_Surface.sendCC(address, onValue);
    }

    void sendOff(MIDIAddress) {
      // nothing here
    }

  private:
    uint8_t onValue;
};

// Now tell the MIDIButton class template (included with the Control
// Surface library) that it has to use your custom sender class.
//
// We wrap it in another class so we can easily construct it later, 
// without having to write `MIDIButton<CustomCCSender>` all the time,
// and so we have more control over the constructor arguments.
// The colon (:) indicates inheritance.
struct CustomCCButton : MIDIButton<CustomCCSender> {
  // Constructor
  CustomCCButton(pin_t pin, MIDIAddress address, uint8_t onValue = 0x7F)
    : MIDIButton(pin, address, {onValue}) {}
  //  ^~~~~~~~~~ Initialization of the base class MIDIButton
};

// Now we can instantiate an object of our custom class.
// The three arguments match the ones of the CustomCCButton
// constructor we wrote a couple of lines back.
CustomCCButton button1 = {
  6,      // button pin
  {9, CHANNEL_5},   // MIDI address
  00,   // on value
};
CustomCCButton button2 = {
  7,      // button pin
  {9, CHANNEL_5},   // MIDI address
  127,   // on value
};

void setup() {
  Control_Surface.begin();
}
 
void loop() { // Refresh all inputs
  Control_Surface.loop();
}`