tttapa/Control-Surface #169: CCRotaryEncoderExtIO

IsThisNameGoodEnough, 2020-04-15:
Hi,

I was wondering if there's been any discussions on implementing a rotary encoder class that's compatible with ExtIO.  I understand that it couldn't use interrupts and may be prone to missing pulses due to the slow nature of polling, but it seems that this might be a nice feature for people using simple turn knobs.  The basic logic in Paul's encoder library seems pretty straightforward (copied below) and I would imagine a CCRotaryEncoderExtIO could use the existing Button class to poll the states, so to me (albeit not someone with a programming background) it looks pretty straightforward to implement.  Perhaps I'm missing something, which is not hard to believe at all.

```cpp

//                           _______         _______       
//               Pin1 ______|       |_______|       |______ Pin1
// negative <---         _______         _______         __      --> positive
//               Pin2 __|       |_______|       |_______|   Pin2

		//	new	new	old	old
		//	pin2	pin1	pin2	pin1	Result
		//	----	----	----	----	------
		//	0	0	0	0	no movement
		//	0	0	0	1	+1
		//	0	0	1	0	-1
		//	0	0	1	1	+2  (assume pin1 edges only)
		//	0	1	0	0	-1
		//	0	1	0	1	no movement
		//	0	1	1	0	-2  (assume pin1 edges only)
		//	0	1	1	1	+1
		//	1	0	0	0	+1
		//	1	0	0	1	-2  (assume pin1 edges only)
		//	1	0	1	0	no movement
		//	1	0	1	1	-1
		//	1	1	0	0	+2  (assume pin1 edges only)
		//	1	1	0	1	-1
		//	1	1	1	0	+1
		//	1	1	1	1	no movement
/*
	// Simple, easy-to-read "documentation" version :-)
	//
	void update(void) {
		uint8_t s = state & 3;
		if (digitalRead(pin1)) s |= 4;
		if (digitalRead(pin2)) s |= 8;
		switch (s) {
			case 0: case 5: case 10: case 15:
				break;
			case 1: case 7: case 8: case 14:
				position++; break;
			case 2: case 4: case 11: case 13:
				position--; break;
			case 3: case 12:
				position += 2; break;
			default:
				position -= 2; break;
		}
		state = (s >> 2);
	}
*/

```

IsThisNameGoodEnough, 2020-04-15:
As a follow-up, I would imagine the debounce feature could be reduced  or possibly removed when polling the button states, although I don't know if that adds complexity.

tttapa, 2020-04-15:
There's some experimental work going on on the `share-encoder` branch.

 - [MCP23017-RotaryEncoder-No-Interrupts](https://tttapa.github.io/Control-Surface-doc/share-encoder/Doxygen/d7/dac/MCP23017-RotaryEncoder-No-Interrupts_8ino-example.html)
 - [MCP23017-RotaryEncoder-Interrupts](https://tttapa.github.io/Control-Surface-doc/share-encoder/Doxygen/dd/d56/MCP23017-RotaryEncoder-Interrupts_8ino-example.html)

I don't have any intentions to add a general ExtIO variant for the encoders, simply because for most available ExtIO types, it doesn't make much sense. For instance, polling an encoder connected using an analog multiplexer would be way too slow anyway.

The MCP23017 works perfectly on Teensy because it allows you to use I²C inside of interrupt service routines. On most other boards, it still resorts to polling, but it can handle all 16 inputs simultaneously, and it only has to poll the actual inputs if the MCP23017's interrupt pin is low, so it's still pretty fast.

For ExtIO, all pins of the different encoders could be in completely different ExtIOElements - in theory at least.  
Each pin number has to be resolved to an ExtendedIOElement, the ExtendedIOElement would have to poll the pin ... Compare this with polling just a single pin in the case of the MCP23017 encoders I mentioned above.  
You could use the new buffered ExtIO functions I've added, but, it's still going to be slower, and not really worth it. You might just as well write a custom Encoder driver for the specific port expander like I've done for the MCP23017, it takes the same amount of time to write, and will yield much better results in the long run.

IsThisNameGoodEnough, 2020-04-15:
Quick question regarding the MCP23017 work: it looks like the way it's written it's not currently compatible with using button's on the MCP23017 as well.  Is this correct?  Is there any plans to allow for a combination of buttons and encoders within an MCP23017 object?

The application I'm working on has a dynamic allocation of buttons and encoders, and I'm trying to figure out out to scale the number upwards.  The encoders are used for slow moving dials, so I'm not so worried about polling vs interrupt (although perhaps I'm being naive and the performance of polling really is not adequate).  It seems one of the benefits of ExtIO is how it works transparently between real pins and virtual pins.

tttapa, 2020-04-16:
> Quick question regarding the MCP23017 work: it looks like the way it's written it's not currently compatible with using button's on the MCP23017 as well. Is this correct? Are there any plans to allow for a combination of buttons and encoders within an MCP23017 object?

That's correct. I don't know your specific situation of course, but I'd try to move as many of the encoders to real Arduino pins, and move everything else to the MCP.  
Do you really need to be able to share an MCP between buttons and encoders?

You could write a simple Encoder class using ExtIO yourself, of course, I can help you with that if you get stuck, but it's going to be a bit harder to optimize.  
I'd recommend using the `share-encoder` branch, it allows different types of encoders, all other branches have the PJRC Encoder class hard-coded. `share-encoder` should be fairly up-to-date with `master`.  
If you need a push in the right direction, I'll have a look at it tomorrow.

IsThisNameGoodEnough, 2020-04-16:
Yeah, I'm trying to set up a set number of digital inputs and let the user decide if the inputs are buttons or encoders.  The inputs span pins and a couple muxes.  I'm going to tell the user to prioritize the direct pins first, but don't have control on the final combination they choose.

Appreciate the help regarding the ExtIO based encoder!  My programming skills aren't very good, but I might take a crack at it over the next few days.  Will probably reach out once I'm banging my head on the wall.  Kind of interested to implement it just to see how much of a problem the polling latency actually plays.  When I setup an encoder on the mux as two buttons I don't have any dropped pulses due to the slow nature of the dial.

IsThisNameGoodEnough, 2020-04-18:
Ok, have my first question if you don't mind.  My approach is to create two new objects:

- CCRotaryEncoderExtIO
- MIDIRotaryEncoderExtIO

I'm still using "const EncoderPinList &pins" as an input to CCRotaryEncoderExtIO to keep consistency with the other types of encoders.  My plan is to have MIDIRotaryEncoderExtIO initialize two buttons and use them to check the states.  Seems pretty straight forward.  MIDIButtons does this by having CCButtons pass an array of buttons which are then initialized in MIDIButtons.

So my question is: is there an elegant way to pass the EncoderPinList into CCRotaryEncoderExtIO  and turn that into a 2 element array that can used to initialize a buttons object?  You can see my attempt in the code below, which (as expected) results in a few errors.  Any help would be greatly appreciated!

**\src\MIDI_Outputs\CCRotaryEncoderExtIO.hpp**
```cpp
#pragma once

#include <MIDI_Outputs/Abstract/MIDIRotaryEncoderExtIO.hpp>
#include <MIDI_Senders/RelativeCCSender.hpp>

BEGIN_CS_NAMESPACE

/**
 * @brief   A class of MIDIOutputElement%s that read the input of a **quadrature
 *          (rotary) encoder** and send out relative MIDI **Control Change**
 *          events.
 * 
 * This version cannot be banked.
 *
 * @note    This class uses polling of ExtIO pins to determine encoder position
 *			and has a high likelyhood of missing encoder pulses.  It is
 *			recommended only to use this version if an encoder must be attached
 *			via a multiplexer and is very slow to change position (e.g. a hand
 *			dial).
 *
 * @ingroup MIDIOutputElements
 */
class CCRotaryEncoderExtIO : public MIDIRotaryEncoderExtIO<RelativeCCSender> {
  public:
    /**
     * @brief   Construct a new CCRotaryEncoder object with the given pins, 
     *          address, channel, speed factor, and number of pulses per step.
     * 
     * @param   pins
     *          A list of the two pins connected to the A and B outputs of the
     *          encoder. 
	 *			The internal pull-up resistor will be enabled.
     * @param   address
     *          The MIDI address containing the controller number [0, 119], 
     *          channel [CHANNEL_1, CHANNEL_16], and optional cable number 
     *          [0, 15].
     * @param   speedMultiply
     *          A constant factor to increase the speed of the rotary encoder.
     *          The difference in position will just be multiplied by this 
     *          factor. 
     * @param   pulsesPerStep
     *          The number of pulses per physical click of the encoder.
     *          For a normal encoder, this is 4. If you want to increase the
     *          resolution, for the use of Jog wheels, for example, you can go
     *          as 1.  
     *          Whereas a greater speedMultiplier factor will increase the 
     *          speed, increasing the number of pulsesPerStep will result in a 
     *          lower speed.
     */
    CCRotaryEncoderExtIO(const EncoderPinList &pins, const MIDIAddress &address,
                    int8_t speedMultiply = 1, uint8_t pulsesPerStep = 4)
        : MIDIRotaryEncoderExtIO(pins, address, speedMultiply, pulsesPerStep, {}) {}
};

END_CS_NAMESPACE
```

**\src\MIDI_Outputs\Abstract\MIDIRotaryEncoderExtIO.hpp**
```cpp
[... 160 lines cut ...]

C:\Users\mrjes\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Outputs/Abstract/MIDIRotaryEncoderExtIO.hpp:42:21: error: 'button' was not declared in this scope

    buttonState[i] = button.update();

                     ^

C:\Users\mrjes\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Outputs/Abstract/MIDIRotaryEncoderExtIO.hpp:74:40: error: there are no arguments to 'indices' that depend on a template parameter, so a declaration of 'indices' must be available [-fpermissive]

   for (unsigned i : indices(buttonState)) {

                                        ^

C:\Users\mrjes\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Outputs/Abstract/MIDIRotaryEncoderExtIO.hpp: In instantiation of 'void CS::MIDIRotaryEncoderExtIO<Sender>::update() [with Sender = CS::RelativeCCSender]':

C:\Users\mrjes\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Outputs/Abstract/MIDIRotaryEncoderExtIO.hpp:41:28: error: 'indices' was not declared in this scope

   for (unsigned i : indices(buttons)){

                            ^

C:\Users\mrjes\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Outputs/Abstract/MIDIRotaryEncoderExtIO.hpp:74:28: error: 'indices' was not declared in this scope

   for (unsigned i : indices(buttonState)) {
```

IsThisNameGoodEnough, 2020-04-19:
As a follow-up I wanted make sure I was using the initialization process in MIDIButtons properly so I thought I'd make a simple example using CCButtons, but am getting an argument list error.  I believe I'm providing the correct inputs.  Any thoughts on why this won't compile?

**Script**
```cpp
#include <Control_Surface.h>

HairlessMIDI_Interface midi;
 
CCButtons buttons = {
  { 0, 4, 16 }, // Array of digital inputs
  { 0x02, CHANNEL_1 }, // Base MIDI address
  2 // MIDI address increment
};
 
void setup() {
  Control_Surface.begin();
}
 
void loop() {
  Control_Surface.loop();
}
```

**Compiling Error**
```
CCButtons:5:1: error: invalid use of template-name 'CS::CCButtons' without an argument list

 CCButtons buttons = {

 ^
```

tttapa, 2020-04-19:
You have to declare `NumButtons` in your code. Either hardcode it to be 2, or make it a template parameter of your `MIDIRotaryEncoderExtIO` class.

I would strongly advise against using the `Button` class for rotary encoder inputs. It debounces the inputs, which is absolutely not what you want for an encoder. Some debouncing may be necessary, but definitely not 25 milliseconds ([BUTTON_DEBOUNCE_TIME](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d69/namespaceAH.html#a838b0bb32d176701ae1362e3188c578d)).

I think it would be best to read the pins directly using ExtIO::digitalRead. 

```cpp
        for (unsigned i : indices(buttons)){
            buttonState[i] = button.update();
        }
```
Unless you've defined `indices` somewhere (the compiler says you haven't), this won't work. See https://en.cppreference.com/w/cpp/language/range-for.

You could write it yourself, of course:
```cpp
#include <Control_Surface.h>
#include <AH/STL/iterator>

class indices {
  public:
    template <class It>
    indices(It &iterable) : size(std::end(iterable) - std::begin(iterable)) {}
    struct iterator { 
      size_t index;
      bool operator!=(iterator that) const { return this->index != that.index; }
      size_t &operator*() { return index; }
      iterator &operator++() { return ++index, *this; }
    };
    iterator begin() { return {0}; }
    iterator end() { return {size}; }
  private:
    size_t size;
};

void setup() {
  Serial.begin(115200);
  while(!Serial);
  const char *strings[] = {"one", "two", "three"};
  for (size_t i : indices(strings))
    Serial.println(strings[i]);
}

void loop() {}
```

[`CCButtons` is a class template](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/def/classBankable_1_1CCButtons.html#details). You have to provide the number of buttons as a template argument:
```cpp
#include <Control_Surface.h>

HairlessMIDI_Interface midi;
 
CCButtons<3> buttons = {
  { 0, 4, 16 }, // Array of digital inputs
  { 0x02, CHANNEL_1 }, // Base MIDI address
  2 // MIDI address increment
};
 
void setup() {
  Control_Surface.begin();
}
 
void loop() {
  Control_Surface.loop();
}
```

IsThisNameGoodEnough, 2020-04-24:
Thanks for the feedback.  I agree about the debounce; my plan is to get it working as buttons (since I've already tested using the encoder as two separate buttons with success), and then change over to digitalRead if it's not fast enough for my application.

At least that was my plan, but I can't seem to get it working.  I've tweaked the code, specifically:

- Making a local variable for the encoder pin list
- Hardcoding the NumButtons for the button array
- Using a class constructor to set the pin list for the button array to the encoder pin list
- Fixing the indices errors

It compiles successfully, but it doesn't look like I'm getting any info from the buttons when button.update() is called.  I placed Serial.Println(buttonState[i]); immediately after button.update() is called and I'm getting 0 at all times.  My guess is that I'm not passing the encoder pin list to the button array properly, so the pin addresses being used are wrong.  Any suggestions on how to debug this?  Updated code is:

```cpp
#pragma once

#include <Def/Def.hpp>
#include <MIDI_Outputs/Abstract/MIDIOutputElement.hpp>

BEGIN_CS_NAMESPACE

/**
 * @brief   An abstract class for rotary encoders that send MIDI events.
 */
template <class Sender>
class MIDIRotaryEncoderExtIO : public MIDIOutputElement {
  protected:
    /**
     * @brief   Construct a new MIDIRotaryEncoderExtIO.
     *
     * @todo    Documentation
     */
    MIDIRotaryEncoderExtIO(const EncoderPinList &pins,
                      const MIDIAddress &address,
                      int8_t speedMultiply, uint8_t pulsesPerStep,
                      const Sender &sender)
        : pins{pins}, address(address),
          speedMultiply(speedMultiply),
          pulsesPerStep(pulsesPerStep), sender(sender) {}

  public:
    void begin() final override {
        for (auto &button : buttons)
            button.begin();
    }
    void update() final override {
        AH::Button::State buttonState[2];
        int priorState[2] = {0, 0};
        int newState[2] = {0, 0};
        long currentPosition = 0;
        int i = 0;
        
        for (auto &button : buttons) {
            buttonState[i] = button.update();
            i++;
        }
        
        // Structuring code to determine position based off of Paul Stoffregen's
        // encoder library comments, as copied here:
        //
        //                           _______         _______       
        //               Pin1 ______|       |_______|       |______ Pin1
        // negative <---         _______         _______         __      --> positive
        //               Pin2 __|       |_______|       |_______|   Pin2

        //    new    new    old    old
        //    pin2    pin1    pin2    pin1    Result
        //    ----    ----    ----    ----    ------
        //    0    0    0    0    no movement
        //    0    0    0    1    +1
        //    0    0    1    0    -1
        //    0    0    1    1    +2  (assume pin1 edges only)
        //    0    1    0    0    -1
        //    0    1    0    1    no movement
        //    0    1    1    0    -2  (assume pin1 edges only)
        //    0    1    1    1    +1
        //    1    0    0    0    +1
        //    1    0    0    1    -2  (assume pin1 edges only)
        //    1    0    1    0    no movement
        //    1    0    1    1    -1
        //    1    1    0    0    +2  (assume pin1 edges only)
        //    1    1    0    1    -1
        //    1    1    1    0    +1
        //    1    1    1    1    no movement

        i = 0;
        for (auto &button : buttons) {
            if(buttonState[i] == AH::Button::Released || buttonState[i] == AH::Button::Falling ) {
                priorState[i] = 1;
            }
            if(buttonState[i] == AH::Button::Released || buttonState[i] == AH::Button::Rising ) {
                newState[i] = 1;
            }
            i++;
        }
        
        unsigned int s = priorState[0] * 1 + priorState[1] * 2 + newState[0] * 4 + newState[1] * 8;
        
        switch (s) {
            case 0: case 5: case 10: case 15:
                break;
            case 1: case 7: case 8: case 14:
                currentPosition++; break;
            case 2: case 4: case 11: case 13:
                currentPosition--; break;
            case 3: case 12:
                currentPosition += 2; break;
            default:
                currentPosition -= 2; break;
        }
        
        long difference = (currentPosition - previousPosition) / pulsesPerStep;
        if (difference) {
            sender.send(difference * speedMultiply, address);
            previousPosition += difference * pulsesPerStep;
        }
    }
    MIDIRotaryEncoderExtIO(){
        buttons = pins;
    }

  private:
    CS::EncoderPinList pins;
    Array<AH::Button, 2> buttons;
	const MIDIAddress address;
    const int8_t speedMultiply;
    const uint8_t pulsesPerStep;
    long previousPosition = 0;

  public:
    Sender sender;
};

END_CS_NAMESPACE
```

IsThisNameGoodEnough, 2020-04-24:
I also tried hardcoding the encoder pins in the class constructor (see code below) with the same result.

```cpp
MIDIRotaryEncoderExtIO(){
    buttons = {22, 23};
}
```

tttapa, 2020-04-24:
Adding another constructor won't help, unless you're using delegating constructors, only a single constructor will be used at any time.

You don't need the `pins` member variable either. You just need a single constructor that correctly initializes the buttons member:

```cpp
    MIDIRotaryEncoderExtIO(const EncoderPinList &pins,
                      const MIDIAddress &address,
                      int8_t speedMultiply, uint8_t pulsesPerStep,
                      const Sender &sender)
        : buttons{pins.A, pins.B}, address(address),
          speedMultiply(speedMultiply),
          pulsesPerStep(pulsesPerStep), sender(sender) {}
```

A general remark on architecture: I don't think the MIDIRotaryEncoder class should handle the actual reading of the encoder, it should just act as a proxy between the actual encoder class and the MIDI sender.

You can clearly see this on the share-encoder branch: https://tttapa.github.io/Control-Surface-doc/share-encoder/Doxygen/db/d50/Abstract_2MIDIRotaryEncoder_8hpp_source.html  
The MIDI encoder class just reads from a generic encoder class, without worrying how the encoder is actually read.  
This allows you to use all kinds of encoders interchangeably, for example, the standard `Encoder` (with a patch to add a working move constructor), references to encoders, actual encoders, encoders connected to an MCP23017, your ExtIO encoder, etc.  
For example:
https://tttapa.github.io/Control-Surface-doc/share-encoder/Doxygen/dd/d56/MCP23017-RotaryEncoder-Interrupts_8ino-example.html

The main reason why the EncoderPinList proxy exists, is because the PJRC `Encoder` class has a defective move constructor. This means that you cannot pass an encoder into the constructor to then save it as a member variable. You have to construct it in-place as the member variable.

(Upon construction, the `Encoder` class writes a pointer to itself into a global ISR context array. The move constructor should change that pointer when moving the object, otherwise, it would still point to the old encoder, the one that has been moved from. PJRC`s `Encoder` doesn't do this, that's why the `EncoderPinList` workaround is needed.)

IsThisNameGoodEnough, 2020-04-25:
Very much appreciate the help.  As you can probably tell, I'm a novice when it comes to C++ (or coding in general for that matter).  I thought I had tried passing {pin.A, pin.B} previously without success.  Also didn't realize that an initialization list is considered a constructor and using it means that you can't call a separate constructor function.  Good thing to know!

Understand your point regarding the use of MIDIRotaryEncoder.  My thought was that it's essentially two buttons with a little logic, and so it would be a similar approach as MIDIButtons.  Now that the first pass is working, I'll play around with it and most likely roll it into the MIDI encoder class along with any changes.

IsThisNameGoodEnough, 2020-05-03:
I've updated the code to use ExtIO::digitalRead and it's working well.  I haven't had an issue with false triggering due to the removal of any debounce filtering.  I'm not going to port it over to use the generic encoder class, as it's easier to implement it in its current form in my particular application.  That said, here's the updated code in case anyone wants to in the future.  Thanks again for the help tttapa!

**\src\MIDI_Outputs\Abstract\MIDIRotaryEncoderExtIO.hpp**

```cpp
#pragma once

#include <Def/Def.hpp>
#include <MIDI_Outputs/Abstract/MIDIOutputElement.hpp>

BEGIN_CS_NAMESPACE

/**
 * @brief   An abstract class for rotary encoders that send MIDI events.
 */
template <class Sender>
class MIDIRotaryEncoderExtIO : public MIDIOutputElement {
  protected:
    /**
     * @brief   Construct a new MIDIRotaryEncoderExtIO.
     *
     * @todo    Documentation
     */
    MIDIRotaryEncoderExtIO(const EncoderPinList &pins,
                      const MIDIAddress &address,
                      int8_t speedMultiply, uint8_t pulsesPerStep,
                      const Sender &sender)
        : pins(pins), address(address),
          speedMultiply(speedMultiply),
          pulsesPerStep(pulsesPerStep), sender(sender) {}

  public:
    void begin() final override {
        Array<AH::Button, 2> buttons = {pins.A, pins.B};
        
        for (auto &button : buttons)
            button.begin();
    }
    void update() final override {
        int newState[2] = {1, 1};

        newState[0] = AH::ExtIO::digitalRead(pins.A);
        newState[1] = AH::ExtIO::digitalRead(pins.B);
        
        // Structuring code to determine position based off of Paul Stoffregen's
        // encoder library comments, as copied here:
        //
        //                           _______         _______       
        //               Pin1 ______|       |_______|       |______ Pin1
        // negative <---         _______         _______         __      --> positive
        //               Pin2 __|       |_______|       |_______|   Pin2

        //    new    new    old    old
        //    pin2    pin1    pin2    pin1    Result
        //    ----    ----    ----    ----    ------
        //    0    0    0    0    no movement
        //    0    0    0    1    +1
        //    0    0    1    0    -1
        //    0    0    1    1    +2  (assume pin1 edges only)
        //    0    1    0    0    -1
        //    0    1    0    1    no movement
        //    0    1    1    0    -2  (assume pin1 edges only)
        //    0    1    1    1    +1
        //    1    0    0    0    +1
        //    1    0    0    1    -2  (assume pin1 edges only)
        //    1    0    1    0    no movement
        //    1    0    1    1    -1
        //    1    1    0    0    +2  (assume pin1 edges only)
        //    1    1    0    1    -1
        //    1    1    1    0    +1
        //    1    1    1    1    no movement
        
        unsigned int s = priorState[0] + priorState[1] * 2 + newState[0] * 4 + newState[1] * 8;
                
        switch (s) {
            case 0: case 5: case 10: case 15:
                break;
            case 1: case 7: case 8: case 14:
                currentPosition++; break;
            case 2: case 4: case 11: case 13:
                currentPosition--; break;
            case 3: case 12:
                currentPosition += 2; break;
            default:
                currentPosition -= 2; break;
        }
        
        long difference = (currentPosition - previousPosition) / pulsesPerStep;
        if (difference) {
            sender.send(difference * speedMultiply, address);
            previousPosition += difference * pulsesPerStep;
        }
        
        priorState[0] = newState[0];
        priorState[1] = newState[1];
    }

  private:
    CS::EncoderPinList pins;
    Array<AH::Button, 2> buttons;
    const MIDIAddress address;
    const int8_t speedMultiply;
    const uint8_t pulsesPerStep;
    long currentPosition = 0;
    long previousPosition = 0;
    int priorState[2] = {1, 1};

  public:
    Sender sender;
};

END_CS_NAMESPACE
```

closed 2020-05-03 by IsThisNameGoodEnough

tttapa, 2020-05-03:
Thanks for sharing!

Alexandr010360, 2024-01-08:
> Я обновил код для использования ExtIO::digitalRead, и он работает хорошо. У меня не было проблем с ложным срабатыванием из-за удаления какой-либо фильтрации дребезга. Я не собираюсь переносить его для использования общего класса кодировщика, поскольку в моем конкретном приложении его проще реализовать в его текущей форме. Тем не менее, вот обновленный код на случай, если кто-то захочет это сделать в будущем. Еще раз спасибо за помощь, ттапа!
> 
> **\src\MIDI_Outputs\Abstract\MIDIRotaryEncoderExtIO.hpp**
> 
> ```c++
> #pragma once
> 
> #include <Def/Def.hpp>
> #include <MIDI_Outputs/Abstract/MIDIOutputElement.hpp>
> 
> BEGIN_CS_NAMESPACE
> 
> /**
>  * @brief   An abstract class for rotary encoders that send MIDI events.
>  */
> template <class Sender>
> class MIDIRotaryEncoderExtIO : public MIDIOutputElement {
>   protected:
>     /**
>      * @brief   Construct a new MIDIRotaryEncoderExtIO.
>      *
>      * @todo    Documentation
>      */
>     MIDIRotaryEncoderExtIO(const EncoderPinList &pins,
>                       const MIDIAddress &address,
>                       int8_t speedMultiply, uint8_t pulsesPerStep,
>                       const Sender &sender)
>         : pins(pins), address(address),
>           speedMultiply(speedMultiply),
>           pulsesPerStep(pulsesPerStep), sender(sender) {}
> 
>   public:
>     void begin() final override {
>         Array<AH::Button, 2> buttons = {pins.A, pins.B};
>         
>         for (auto &button : buttons)
>             button.begin();
>     }
>     void update() final override {
>         int newState[2] = {1, 1};
> 
>         newState[0] = AH::ExtIO::digitalRead(pins.A);
>         newState[1] = AH::ExtIO::digitalRead(pins.B);
>         
>         // Structuring code to determine position based off of Paul Stoffregen's
>         // encoder library comments, as copied here:
>         //
>         //                           _______         _______       
>         //               Pin1 ______|       |_______|       |______ Pin1
>         // negative <---         _______         _______         __      --> positive
>         //               Pin2 __|       |_______|       |_______|   Pin2
> 
>         //    new    new    old    old
>         //    pin2    pin1    pin2    pin1    Result
>         //    ----    ----    ----    ----    ------
>         //    0    0    0    0    no movement
>         //    0    0    0    1    +1
>         //    0    0    1    0    -1
>         //    0    0    1    1    +2  (assume pin1 edges only)
>         //    0    1    0    0    -1
>         //    0    1    0    1    no movement
>         //    0    1    1    0    -2  (assume pin1 edges only)
>         //    0    1    1    1    +1
>         //    1    0    0    0    +1
>         //    1    0    0    1    -2  (assume pin1 edges only)
>         //    1    0    1    0    no movement
>         //    1    0    1    1    -1
>         //    1    1    0    0    +2  (assume pin1 edges only)
>         //    1    1    0    1    -1
>         //    1    1    1    0    +1
>         //    1    1    1    1    no movement
>         
>         unsigned int s = priorState[0] + priorState[1] * 2 + newState[0] * 4 + newState[1] * 8;
>                 
>         switch (s) {
>             case 0: case 5: case 10: case 15:
>                 break;
>             case 1: case 7: case 8: case 14:
>                 currentPosition++; break;
>             case 2: case 4: case 11: case 13:
>                 currentPosition--; break;
>             case 3: case 12:
>                 currentPosition += 2; break;
>             default:
>                 currentPosition -= 2; break;
>         }
>         
>         long difference = (currentPosition - previousPosition) / pulsesPerStep;
>         if (difference) {
>             sender.send(difference * speedMultiply, address);
>             previousPosition += difference * pulsesPerStep;
>         }
>         
>         priorState[0] = newState[0];
>         priorState[1] = newState[1];
>     }
> 
>   private:
>     CS::EncoderPinList pins;
>     Array<AH::Button, 2> buttons;
>     const MIDIAddress address;
>     const int8_t speedMultiply;
>     const uint8_t pulsesPerStep;
>     long currentPosition = 0;
>     long previousPosition = 0;
>     int priorState[2] = {1, 1};
> 
>   public:
>     Sender sender;
> };
> 
> END_CS_NAMESPACE
> ```

Здравствуйте! При компиляции ошибка не называет тип. Вот скетч. Что тут не так? 

#include <Control_Surface.h>

USBMIDI_Interface midi; // Instantiate a MIDI over USB interface.

MIDIRotaryEncoderExtIO buttons {
  {2, 3},               // Контакты кнопки: 5 шагов, 6 шагов
 {MIDI_CC::Channel_Volume, CHANNEL_3}, // Увеличение/Уменьшение адреса CC
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface (calls button.begin())
  Serial.begin(57600);
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface (calls button.update())
}

tttapa, 2024-01-08:
Note that with the current version of Control Surface, you can simplify the above code as follows:

```cpp
#include <Control_Surface.h>
#include <AH/Hardware/RegisterEncoders.hpp>

// Rotary encoder class that can be used with ExtIO pins.
// Note that polling many ExtIO pins may too slow to work reliably.
class ExtIOEncoder {
  public:
    using position_t = uint8_t;
    ExtIOEncoder(EncoderPinList pins) : pins(pins) {}

    // Enable pull-up resistors. You may want to extend this so it also
    // initializes the state with the current state of the pins.
    void begin() {
      ExtIO::pinMode(pins.A, INPUT_PULLUP);
      ExtIO::pinMode(pins.B, INPUT_PULLUP);
    }

    // Read the two encoder pins and update the position.
    position_t read() {
      uint8_t pinstate = (ExtIO::digitalRead(pins.A) == HIGH ? 1 : 0)
                       | (ExtIO::digitalRead(pins.B) == HIGH ? 2 : 0);
      state.update(pinstate);
      return state.read(0);
    }

  private:
    // Extended IO pin numbers of the encoder's A and B pins
    // (common should be connected to ground).
    EncoderPinList pins;
    // Keeps track of the position of the encoder.
    AH::RegisterEncoders<uint8_t, 1, position_t, false> state;
};

// Example of using the ExtIOEncoder class as the encoder for a MIDI element.
struct CCExtIOEncoder : GenericMIDIRotaryEncoder<ExtIOEncoder, RelativeCCSender> {
  CCExtIOEncoder(EncoderPinList pins, MIDIAddress address, int16_t speedMultiply = 1, uint8_t pulsesPerStep = 4)
  : GenericMIDIRotaryEncoder({pins}, address, speedMultiply, pulsesPerStep, {}) {}
};

USBMIDI_Interface midi;

CCExtIOEncoder enc {
  {{0, 1}}, // encoder pins
  MIDI_CC::Channel_Volume, // address
};

void setup() {
  Control_Surface.begin();
}

void loop() {
    Control_Surface.loop();
}
```