tttapa/Control-Surface #170: Plantilla base para crear nuevas funciones pertenecientes al Bankable

alejamelissa, 2020-04-16:
Hola, estoy por usar sensores de presion o fuerza para simular el uso de pads, con el siguiente codigo, los susodichos sensores de fuerza comportan de manera correcta y tienen un desempeño a mi gusto sorprendente en la medida que se añade expresividad a las notas segun la intensidad de fuerza que se aplique con los dedos, caracteristica que no se logra con piezoelectricos. Dicho lo anterior me gustaria poder usar la funcion transposer para cambiar las notas asignadas a los pads, funcion que me ha sido imposible de usar al ser este segun entiendo un arreglo que no pertenece al bankable. Si pudieras orientarme de cuales archivos de la libreria se deben citar o cual seria la estructura para lograr el cambio de nota en este arreglo.

#include <Control_Surface.h>

USBMIDI_Interface midi;

class NotePot : public MIDIOutputElementPotentiometer {
  public:
    NotePot(pin_t analogPin, MIDICNChannelAddress midinote)
      : analog(analogPin), midinote(midinote) {}

    void begin() override {}
    void update() override {
      if (analog.update()) {
        uint8_t newValue = analog.getValue();
        if (newValue > 0) {
          if (noteOnSent) {
            midi.sendKP(midinote, newValue); // MIDI Key Pressure
          } else {
            midi.sendNoteOn(midinote, newValue);
            noteOnSent = true;
          }
        } else {
          midi.sendNoteOff(midinote, 0x7F);
          noteOnSent = false;
        }
      }
    }

    void map(MappingFunction fn) { analog.map(fn); }
    void invert() { analog.invert(); }

  private:
    FilteredAnalog<7> analog;
    MIDICNChannelAddress midinote;
    bool noteOnSent = false;
};

using namespace MIDI_Notes;
NotePot notepots[] = {
  {A0, {note(C, 4), CHANNEL_1}},
  {A1, {note(Db, 4), CHANNEL_1}},
  {A2, {note(D, 4), CHANNEL_1}},
  {A3, {note(Eb, 4), CHANNEL_1}},
  // Etc.
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}

tttapa, 2020-04-16:
Apologies, I seem to have missed the notification for your previous issue. I'll have a look at it later today.

Basically, you have to replace your `MIDICNChannelAddress` with a `Bankable::SingleAddress`.

tttapa, 2020-04-16:
This should work:

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

BEGIN_CS_NAMESPACE
namespace Bankable {

class NotePot : public MIDIOutputElementPotentiometer {
  public:
    NotePot(OutputBankConfig<> config, pin_t analogPin, MIDIAddress midinote)
      : analog(analogPin), address(config, midinote) {}

    void begin() override {}
    void update() override {
      if (analog.update()) {
        uint8_t newValue = analog.getValue();
        if (newValue > threshold) {
          if (noteOnSent) {
            midi.sendKP(address.getActiveAddress(), newValue); // MIDI Key Pressure
          } else {
            address.lock();
            midi.sendNoteOn(address.getActiveAddress(), newValue);
            noteOnSent = true;
          }
        } else {
          midi.sendNoteOff(address.getActiveAddress(), 0x7F);
          address.lock();
          noteOnSent = false;
        }
      }
    }

    void map(MappingFunction fn) { analog.map(fn); }
    void invert() { analog.invert(); }

  private:
    FilteredAnalog<7> analog;
    SingleAddress address;
    bool noteOnSent = false;
    constexpr static const uint8_t threshold = 1;
};

} // namespace Bankable
END_CS_NAMESPACE

Bank<4> bank;

using namespace MIDI_Notes;
Bankable::NotePot notepots[] = {
  {bank, A0, {note(C, 4), CHANNEL_1}},
  {bank, A1, {note(Db, 4), CHANNEL_1}},
  {bank, A2, {note(D, 4), CHANNEL_1}},
  {bank, A3, {note(Eb, 4), CHANNEL_1}},
  // Etc.
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```
You need a fairly recent master version. 

If changing banks doesn't have any effect, that could be because the analog value never goes below the threshold, so it never unlocks the bank address. You can solve this by changing the threshold or by using a mapping function.