tttapa/Control-Surface #171: readme.md error

mididreamer, 2020-04-16:
```cpp

// Your code here

```hello,  it seems that input and output are inverted in the readme.md'''

tttapa, 2020-04-16:
I guess it depends on how you look at it. From the Arduino's perspective, buttons and potentiometers generate MIDI output, while LEDs listen for MIDI input.

What would you change to make it more clear?

mididreamer, 2020-04-16:
In your midi controler librairy it seems to me that
You said button is input and led is output (wich is seems to me more
logical)

tttapa, 2020-04-16:
The MIDI controller library talks about input from potentiometers and MIDI output, which is consistent with the terminology used in Control Surface.

Personally, I find this the most logical option, but I'm open to improvement if you have a specific proposal.

Do keep in mind that push buttons, potentiometers, etc. are `MIDIOutputElement`s, and LEDs and VU meters are `MIDIInputElement`s,  so I can't change the terminology in the documentation without breaking the entire API.

mididreamer, 2020-04-17:
Ok, no need to change anything, i'm ok with your clarification. Thanks to
taking Time to answer me.

closed 2020-04-17 by mididreamer