tttapa/Control-Surface #172: Custom Callback with TM1637

wayna, 2020-04-16:
Hello!

I'm trying to get a TM1637 to work with the library.
I use the [TM1637 Display Library](https://github.com/avishorp/TM1637)  from avisharp.
I'd like to use it with a selector, the goal would be to display the bank number.
So i tried to understand the custom selector callbacks with my poor coding level.
I tried modify the example you provide with the library, with no success.
I think i have to modify the declaration of the constructor, but i really don't know how.

Could you explain me how to write it?
Thanks a lot

```cpp

#include <Control_Surface.h>
// ------------------------- TM1637 CONFIG -------------------------------------------
#include <Arduino.h>
#include <TM1637Display.h>
// Module connection pins (Digital Pins)
#define CLK 5
#define DIO 6

const uint8_t SEG_PAGE1[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_B | SEG_C }; // P , -- , 0 , 1
const uint8_t SEG_PAGE2[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_D | SEG_E | SEG_E | SEG_G }; // P , -- , 0 , 2
const uint8_t SEG_PAGE3[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C | SEG_D | SEG_G }; 
const uint8_t SEG_PAGE4[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_B | SEG_C | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE5[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_C | SEG_D | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE6[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_C | SEG_D | SEG_E | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE7[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C }; 
const uint8_t SEG_PAGE8[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C | SEG_D |  SEG_E | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE9[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C | SEG_D | SEG_F | SEG_G }; 
TM1637Display display(CLK, DIO);

// ------------------------- TM1637 CONFIG -------------------------------------------

// Custom callback to handle output for a selector.
class MySelectorCallback {
  public:
    // Constructor
    MySelectorCallback(pin_t redLED, pin_t greenLED, pin_t blueLED)
      : redLED(redLED), greenLED(greenLED), blueLED(blueLED) {}

    // Begin function is called once by Control Surface.
    // Use it to initialize everything.
    void begin() {
      }

    // Update function is called continuously by Control Surface.
    // Use it to implement things like fading, blinking ...
    void update() {}

    // Update function with arguments is called when the setting
    // changes.
    // Use it to update the LEDs.
    void update(setting_t oldSetting, setting_t newSetting) {
      (void) oldSetting; // unused in this example
      show(newSetting);
    }

  private:
    void show(setting_t setting) {
      switch (setting) {
        case 0: display.setSegments(SEG_PAGE1);
        case 1: display.setSegments(SEG_PAGE2);
        case 2: display.setSegments(SEG_PAGE3);
        case 3: display.setSegments(SEG_PAGE4);
        case 4: display.setSegments(SEG_PAGE5);
        case 5: display.setSegments(SEG_PAGE6);
        default: display.setSegments(SEG_PAGE9);
      }
    }

  private:
    // Member variables to remember the pin numbers of the LEDs.
    pin_t redLED, greenLED, blueLED;
};

// :::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: //

USBMIDI_Interface midi;

// Create 6 banks with one track per bank.
Bank<6> bank = 1;

// Create a selector that uses our custom callback, to control the bank.
GenericIncrementDecrementSelector<6, MySelectorCallback> selector = {
  bank,         // bank to manage
  {14, 15, 16}, // red, green, blue LED pins 
                // (this is the MySelectorCallback constructor defined above)
  {2, 3},       // incr/decr button pins
  Wrap::Wrap,   // wrap around when reaching setting 6
};

void setup() {
  Control_Surface.begin(); // initialize
}

void loop() {

  display.setBrightness(1);
  Control_Surface.loop(); // update
  
}

```

tttapa, 2020-04-16:
Can you control the display without using Control Surface, just using the examples from the TM1637 library?

wayna, 2020-04-17:
Hello,

Yes the library example works well

tttapa, 2020-04-17:
It looks like you forgot the `break` statements in your switch:

```cpp
// ------------------------- TM1637 CONFIG -------------------------------------------
#include <TM1637Display.h>

const uint8_t SEG_PAGE1[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_B | SEG_C }; // P , -- , 0 , 1
const uint8_t SEG_PAGE2[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_D | SEG_E | SEG_E | SEG_G }; // P , -- , 0 , 2
const uint8_t SEG_PAGE3[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C | SEG_D | SEG_G }; 
const uint8_t SEG_PAGE4[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_B | SEG_C | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE5[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_C | SEG_D | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE6[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_C | SEG_D | SEG_E | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE7[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C }; 
const uint8_t SEG_PAGE8[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C | SEG_D |  SEG_E | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE9[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C | SEG_D | SEG_F | SEG_G }; 

// ------------------------- Control Surface CONFIG -------------------------------------------
#include <Control_Surface.h>

// Custom callback to handle output for a selector.
class MySelectorCallback {
  public:
    // Constructor
    MySelectorCallback(uint8_t clk, uint8_t dio)
      : display(clk, dio) {}

    // Begin function is called once by Control Surface.
    // Use it to initialize everything.
    void begin() {
      display.setBrightness(1);
    }

    // Update function is called continuously by Control Surface.
    // Use it to implement things like fading, blinking ...
    void update() {}

    // Update function with arguments is called when the setting
    // changes.
    // Use it to update the LEDs.
    void update(setting_t oldSetting, setting_t newSetting) {
      (void) oldSetting; // unused in this example
      show(newSetting);
    }

  private:
    void show(setting_t setting) {
      switch (setting) {
        case 0: display.setSegments(SEG_PAGE1); break;
        case 1: display.setSegments(SEG_PAGE2); break;
        case 2: display.setSegments(SEG_PAGE3); break;
        case 3: display.setSegments(SEG_PAGE4); break;
        case 4: display.setSegments(SEG_PAGE5); break;
        case 5: display.setSegments(SEG_PAGE6); break;
        default: display.setSegments(SEG_PAGE9); 
      }
    }

  private:
    TM1637Display display;
};

// :::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: //

USBMIDI_Interface midi;

// Create 6 banks with one track per bank.
Bank<6> bank = 1;

// Create a selector that uses our custom callback, to control the bank.
GenericIncrementDecrementSelector<6, MySelectorCallback> selector = {
  bank,         // bank to manage
  {5, 6},       // clk and dio pins
                // (this is the MySelectorCallback constructor defined above)
  {2, 3},       // incr/decr button pins
  Wrap::Wrap,   // wrap around when reaching setting 6
};

void setup() {
  Control_Surface.begin(); // initialize
}

void loop() {
  Control_Surface.loop(); // update
}
``` 

You could probably clean up your code by moving the `SEG_PAGEn` variables into a (2D) array.

wayna, 2020-04-17:
Works great, Thanks!

I will try to clean the code as you suggest

wayna, 2020-04-19:
Hello!

Still tweaking with this part of code.
Now i'd like to send a different midi note corresponding to the active bank.

So i had some success with the following code (still need to clean the code though).
But i wondered if there is a more efficient (or reliable) way to do this? 
Or does it looks fine to you?

```
// ------------------------- TM1637 CONFIG -------------------------------------------
#include <TM1637Display.h>

const uint8_t SEG_PAGE1[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_B | SEG_C }; // P , -- , 0 , 1
const uint8_t SEG_PAGE2[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_D | SEG_E | SEG_E | SEG_G }; // P , -- , 0 , 2
const uint8_t SEG_PAGE3[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C | SEG_D | SEG_G }; 
const uint8_t SEG_PAGE4[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_B | SEG_C | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE5[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_C | SEG_D | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE6[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_C | SEG_D | SEG_E | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE7[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C }; 
const uint8_t SEG_PAGE8[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C | SEG_D |  SEG_E | SEG_F | SEG_G }; 
const uint8_t SEG_PAGE9[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F , SEG_A | SEG_B | SEG_C | SEG_D | SEG_F | SEG_G }; 
const uint8_t SEG_GO[] = {false, false, SEG_A | SEG_C | SEG_D | SEG_E | SEG_F, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F}; 

// ------------------------- Control Surface CONFIG -------------------------------------------

#include <Control_Surface.h>
USBMIDI_Interface midi; 
using namespace MIDI_Notes;
// :::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: //
// Custom callback to handle output for a selector.
class MySelectorCallback {
  public:
    // Constructor
    MySelectorCallback(uint8_t clk, uint8_t dio)
      : display(clk, dio) {}

    void begin() {
      display.setBrightness(1);
      display.setSegments(SEG_GO);
      delay(2000);
    }

    void update() {}

    void update(setting_t oldSetting, setting_t newSetting) {
      (void) oldSetting; // unused in this example
      show(newSetting);
    }

  private:
    void show(setting_t setting) {
      switch (setting) {
        case 0: display.setSegments(SEG_PAGE1); break;
        case 1: display.setSegments(SEG_PAGE2); break;
        case 2: display.setSegments(SEG_PAGE3); break;
        case 3: display.setSegments(SEG_PAGE4); break;
        case 4: display.setSegments(SEG_PAGE5); break;
      }
    }

  private:
    TM1637Display display;
};

// :::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: //

// Define Page buttons pins
Button pageUpButton = {2};
Button pageDownButton = {3};

// Counter parameters
int pageCounter = 1; // Variable for the page number
const int PAGEAMOUNT = 5; // Quantity of pages

Bank<PAGEAMOUNT> bankFaders = 1; // Create N banks with one track per bank.

// Create a selector that uses our custom callback, to control the bank.
GenericIncrementDecrementSelector<PAGEAMOUNT, MySelectorCallback> selector = {
    bankFaders,         // bank to manage
    {6, 7},       // clk and dio pins
                // (this is the MySelectorCallback constructor defined above)
    {pageUpButton, pageDownButton},       // incr/decr button pins
    Wrap::Clamp,   // wrap around when reaching setting 6
};

//Smart Faders
Bankable::CCSmartPotentiometer<PAGEAMOUNT> playbackFaders[] = {
  {{bankFaders, BankType::CHANGE_CHANNEL}, A3, {0x10, CHANNEL_1}},
  {{bankFaders, BankType::CHANGE_CHANNEL}, A4, {0x11, CHANNEL_1}},
  {{bankFaders, BankType::CHANGE_CHANNEL}, A5, {0x12, CHANNEL_1}},
  {{bankFaders, BankType::CHANGE_CHANNEL}, A6, {0x13, CHANNEL_1}},
};

void setup() {
  Control_Surface.begin(); // initialize
}

void loop() {
  Control_Surface.loop(); // update
  pageUpButton.update();   {
   if (pageUpButton.getState() == Button::Falling && pageCounter != (PAGEAMOUNT + 1)) {
     pageCounter++;
     midi.sendNoteOn ({note(C, pageCounter ), CHANNEL_1}, 0x7F);
     }      
   else {}
  }
  
  pageDownButton.update(); {
   if (pageDownButton.getState() == Button::Falling && pageCounter != 2) {
     pageCounter--;
     midi.sendNoteOn ({note(C, pageCounter), CHANNEL_1}, 0x7F);
   }      
   else {}
  }
}

```

tttapa, 2020-04-19:
The code in your loop seems unnecessary. The selector already keeps track of the page counter. You should be able to just add the `midi.sendNoteOn` calls inside of the `update` method of the selector callback, where you update your display.

wayna, 2020-04-19:
Oh yes, that was obvious! :)

Thank you

wayna, 2020-04-20:
Final code

```cpp
// ------------------------- TM1637 CONFIG -------------------------------------------
#include <TM1637Display.h>
byte SEG_GO[] = {false, false, SEG_A | SEG_C | SEG_D | SEG_E | SEG_F, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F}; // _ , _ , G , 0
byte SEG_PAGE[] = {SEG_A | SEG_B | SEG_E | SEG_F | SEG_G, false, SEG_A | SEG_B | SEG_C | SEG_D | SEG_E | SEG_F, };    // P, _ , 0

// ------------------------- Control Surface CONFIG -------------------------------------------
#include <Control_Surface.h>
USBMIDI_Interface midi; 
using namespace MIDI_Notes;

// :::::::::::::::::::::::::::::::Custom selector callback::::::::::::::::::::::::::::::::::::::::::: //
class MySelectorCallback {
  public:
    // Constructor
    MySelectorCallback(uint8_t clk, uint8_t dio)
      : display(clk, dio) {}

    void begin() {
      display.setBrightness(1);
      display.setSegments(SEG_GO);
      delay(1000);
    }

    void update() {}

    void update(setting_t oldSetting, setting_t newSetting) {
      midi.sendNoteOn ({note(C, newSetting), }, 0x7F);    // Send midi Note according to the active bank 
      display.setSegments(SEG_PAGE, 3, 0);                // Display  "P_0" message on the TM 1637
      display.showNumberDec(newSetting + 1, false, 1, 3); // Display  "N" number according to the active bank on the TM 1637
    }
    
  private:
    TM1637Display display;
};

// :::::::::::::::::::::::::::::::Custom selector callback::::::::::::::::::::::::::::::::::::::::::: //

const int PAGEAMOUNT = 5; // Quantity of pages
Bank<PAGEAMOUNT> bankFaders = 1; // Create N banks with one track per bank.

// Create a selector that uses our custom callback, to control the bank.
GenericIncrementDecrementSelector<PAGEAMOUNT, MySelectorCallback> selector = {
    bankFaders,         // bank to manage
    {6, 7},       // clk and dio pins
                // (this is the MySelectorCallback constructor defined above)
    {2, 3},       // incr/decr button pins
    Wrap::Clamp,   // wrap around when reaching setting 6
};

//Smart Faders
Bankable::CCSmartPotentiometer<PAGEAMOUNT> playbackFaders[] = {
  {{bankFaders, BankType::CHANGE_CHANNEL}, A3, {0x10, CHANNEL_1}},
  {{bankFaders, BankType::CHANGE_CHANNEL}, A4, {0x11, CHANNEL_1}},
  {{bankFaders, BankType::CHANGE_CHANNEL}, A5, {0x12, CHANNEL_1}},
  {{bankFaders, BankType::CHANGE_CHANNEL}, A6, {0x13, CHANNEL_1}},
};

void setup() {
  Control_Surface.begin(); 
}

void loop() {
  Control_Surface.loop(); // update
}

```

closed 2020-04-20 by wayna