tttapa/Control-Surface #173: Velocity with piezo Note button

diegogauffin, 2020-04-16:
Hey! Whats up ? I've just found this example to use the signal of a piezo like velocity of a midiNote. I am trying to translate to Control Surface's library but i cant... Do you know how could it be done? It would be awsome to have that posibilty . 

I ll leave you here the link to the youtube's session where he explain detailed : 

https://www.youtube.com/watch?v=0MVZJtWbqsU&list=LL_iPQiJgIu0aS05By-uEbSw&index=7&t=0s 

```cpp

// Your code here

```

tttapa, 2020-04-16:
This is an untested, literal conversion, but that's all I have time for at the moment:

```cpp
/* Midi Piezoelecric Drum Pads
 * Version: 0.1
 * by Notes and Volts
 * www.notesandvolts.com
 * 
 * This software is an ALPHA version
 * and is unsupported.
 * Use at your own risk!
 */

#include <Control_Surface.h>

#define LED 13
#define PADS 3 // How many drum pads?
#define DEBOUNCE 30 // Debounce time (in milli-seconds)

USBMIDI_Interface midi;

// Connect pads in order starting at A0

int sensitivity = 100; // Maximum input range
int threshold = 10; // Minimum input range

unsigned long timer[PADS];
bool playing[PADS];
int highScore[PADS];
MIDIAddress note[PADS] = { // Set drum pad notes here
  36, 40, 53
}; 

void setup() {
  midi.begin();
  for (int x = 0; x < PADS; x++) {
    playing[x] = false;
    highScore[x] = 0;
    timer[x] = 0;
  }
}

void loop() {
  for (int x = 0; x < PADS; x++) {
    int volume = analogRead(x);
    if (volume >= threshold && playing[x] == false) {
      if (millis() - timer[x] >= DEBOUNCE) {
        playing[x] = true;
        playNote(x, volume);
      }
    }
    else if (volume >= threshold && playing[x] == true) {
      playNote(x, volume);
    }
    else if (volume < threshold && playing[x] == true) {
      midi.sendNoteOn(note[x], highScore[x]);
      midi.sendNoteOff(note[x], 0);
      highScore[x] = 0;
      playing[x] = false;
      timer[x] = millis();
    }
  }
  midi.update();
}

void playNote (int pad, int volume) {
  float velocity = ((volume) / float(sensitivity - threshold)) * 127;
  if (velocity > 127) velocity = 127;
  if (velocity > highScore[pad]) highScore[pad] = velocity;
}
```

cferrarini, 2020-09-13:
Tested and Totally Works!!! 
Its easier now to write a class to Piezo and to ESP32 capacitive touch sensors...
Thanks!

LoughGn, 2021-01-11:
Thank you for this, it's amazing.

I am terribly new to coding and I have been staring at these pages for days not getting anywhere.

Is it possible to use a multiplexer with this?  How/where would I define it and the pins as outputs?

Sorry for the newbness...

Thanks!!!!!!!!!

benwadub, 2021-02-20:
hi @cferrarini does your piezo respond to velocity with that code? I just tried it and it only seems to send not on/off here

aamott, 2021-03-21:
I just wrote a NotePiezo module that senses velocity and tested it today. It currently auto-adjusts the max-velocity so that the maxumum range of sensitivity can be used, but that will probably be changed for a set input value. If you're interested, I could push it or post it here for checking.

TrenerKurochkin, 2021-09-25:
> I just wrote a NotePiezo module that senses velocity and tested it today. It currently auto-adjusts the max-velocity so that the maxumum range of sensitivity can be used, but that will probably be changed for a set input value. If you're interested, I could push it or post it here for checking.

Hello, @aamott, can you post it here too. I'm totally noob in coding, and your example would help me a lot) Thank U

TrenerKurochkin, 2021-10-13:
> Sorry I'm a little late! Here's where I posted the last example: [Pull Request 433](https://github.com/tttapa/Control-Surface/pull/433)
> 
> ```c++
> #include <Control_Surface.h>
> 
> // this starts a hairlessMIDI_Interface connection. If you have an arduino pro 
> //    or another that supports the Keyboard library, you can use USBMIDI_Interface midi;
> //    If you use HairlessMIDI_Interface, you have to install hairlessMIDISerial.
> HairlessMIDI_Interface midi;
> 
> // Pin A1 {note 36, CHANNEL_10 is drums}
> NotePiezo piezo1 = { A1, {36, CHANNEL_10} };
> NotePiezo piezo2 = { A2, {38, CHANNEL_10} };
> // the same concept but with a button
> NoteButton button = { 7, {42, CHANNEL_1} };
> 
> 
> void setup() {
>   Control_Surface.begin();
> 
>   // this just sets the minimum threshold automatically so it doesn't register sitting there as a hit.
>   // piezo1.sampleSilence(1000) samples the background noise. +1 adds a little. 
>   // piezo1.setHitThreshold sets the minimum hit to just above the noise we sampled. 
>   piezo1.setHitThreshold(piezo1.sampleSilence(1000)+1);
>   piezo2.setHitThreshold(piezo2.sampleSilence(1000)+1);
> }
> 
> void loop() {
>   Control_Surface.loop();
> }
> ```
> 
> I'll try to add an example of wiring with protecting zener diodes. Piezos can do some damage to the arduino if it isn't protected. It's only a few dollars for a pack of 100 5v or 3.3v zener diodes on a website like aliexpress. It's super easy. Basically, you just solder a 2K-1M resistor to both wires of the piezo, then solder 2 diodes to the wire connecting to the top plate of the piezo (+) with the lines on the diode facing away from it, and solder the end of one directly to the wire on the bottom plate (-) and the other connects to the Arduino 5v or ESP32 3.3v.

Thank U, dear friend. I've tried it, it is super easy to use, but I have a little problem: no Note Off message, so my piezo notes are endless

Wallirec, 2024-10-28:
Pieter, your example works well, but in my Pro micro 9 analog inputs and 4 and 5 do not exist. I have to initialize 11 inputs to use A10, please tell me a more correct way.

And one more question - how to assign channel 10 in the sketch?