tttapa/Control-Surface #174: Sending different messages over interfaces

remcobrantjes, 2020-04-17:
Hi there, this library looks very nicely structured, I just have a question regarding the use of a dual midi interface. I have looked at most examples but couldn't figure this out, sorry if I missed something.

For now I have just two footswitches and an expression pedal (potentiometer) connected to a Leonardo. First I wanted them to control my DAW. That was really easy with this library so great success.

The second thing I want to do is send midi data to another device over 5pin DIN. I guess this would not be so difficult if I follow the dual midi interface example. The thing is that I would like to send different messages to that device. So as an example button 5 would send CC5 over USB and Programchange 10 over the DIN. Is it possible to do that? Because it looks like one button is associated with the same message for all devices. 

Nice to have would be to receive midi data from the device (over 5pin) and send customized data over USB.  Are these things possible with this library? 

Thanks in advance,
Remco

tttapa, 2020-04-18:
> So as an example button 5 would send CC5 over USB and Programchange 10 over the DIN.

Does it have to send both messages at the same time, or is there a "USB" mode that only sends over USB, and a "DIN" mode that only sends over DIN?  
If this is the case, you could use the Bankable features, with the BankType::CHANGE_CABLENB. This allows you to dynamically change the virtual MIDI cable number a MIDI output element sends on. You can then use the MIDI Routing module to route one virtual cable to USB and the other to DIN.

If they have to send two messages at the same time, it's probably easiest to just write some code for it yourself:
- [Custom-MIDI-Output-Element](https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d3b/Custom-MIDI-Output-Element_8ino-example.html)
- [Custom-MIDI-Sender](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d08/Custom-MIDI-Sender_8ino-example.html)

You can of course reuse much of the hardware utilities and the MIDI sending functions provided by the library.

Instead of `Control_Surface.sendCC` you can use `usbmidi.sendCC` and `serialmidi.sendCC`, for example.

remcobrantjes, 2020-04-19:
Thanks a lot I will give it a try. Yes I would like to do it at the same time. 
Meanwhile I did a workaround by defining two buttons on the same pin, a CCButton and a PCButton. Because the midi device will not respond to the CC and my DAW will not respond to the ProgramChange that works. I guess you could also do it by using different midi channels. Not the most elegant way, so in the end I will try with the custom functions I think.