tttapa/Control-Surface #175: rotary encoder acceleration

sucrz, 2020-04-17:
Hi, 

I trying to implement "accelaration" to rotary encoder like on Elektron boxes (when encoder-switch is press, value inc/dec 4 by 4, if switch is not press the encoder inc/dec 1 by 1) 

The idea is to multiply "speedMultiply" by 4 when the switch is press. 

I've modifie couple of file from the lib to add encoder switch pin "spin" 

I think the problem is in the "void update" part, maybe someone can help me a bit. 

```cpp

#pragma once

#if not defined(Encoder_h_) && not defined(IDE)
#error                                                                         \
"The PJRC Encoder library should be included before the Control-Surface "  \
"library. (#include <Encoder.h>)"
#endif

#include <Def/Def.hpp>
#include <Encoder.h>
#include <MIDI_Outputs/Abstract/MIDIOutputElement.hpp>

BEGIN_CS_NAMESPACE

/**
   @brief   An abstract class for rotary encoders that send MIDI events.
*/
template <class Sender>
class MIDIRotaryEncoder : public MIDIOutputElement {
  protected:
    /**
       @brief   Construct a new MIDIRotaryEncoder.

       @todo    Documentation
    */
    MIDIRotaryEncoder(const EncoderPinList &pins,
                      int8_t spin,
                      const MIDIAddress &address,
                      int8_t speedMultiply, uint8_t pulsesPerStep,
                      const Sender &sender)
      : encoder{pins.A, pins.B}, spin(spin), address(address),
        speedMultiply(speedMultiply),
        pulsesPerStep(pulsesPerStep), sender(sender) {}

    // For tests only
#ifndef ARDUINO
    MIDIRotaryEncoder(const Encoder &encoder,
                      int8_t spin,
                      const MIDIAddress &address,
                      int8_t speedMultiply, uint8_t pulsesPerStep,
                      const Sender &sender)
      : encoder{encoder}, spin(spin), address(address), speedMultiply(speedMultiply),
        pulsesPerStep(pulsesPerStep), sender(sender) {}
#endif

  public:
    void begin() final override {}
    void update() final override {
      int Speed;
      int spinV = digitalRead(spin);
      if (spinV == HIGH) {
        Speed = (speedMultiply * 4);
      }
      else {
        Speed = speedMultiply;
      }
      long currentPosition = encoder.read();
      long difference = (currentPosition - previousPosition) / pulsesPerStep;
      if (difference) {
        sender.send(difference * Speed, address);
        previousPosition += difference * pulsesPerStep;
      }
    }

  private:
    Encoder encoder;
    int8_t spin;
    const MIDIAddress address;
    const int8_t speedMultiply;
    const uint8_t pulsesPerStep;
    long previousPosition = 0;

  public:
    Sender sender;
};

END_CS_NAMESPACE

```

tttapa, 2020-04-18:
How did you connect the switch? You didn't enable the internal pull-up resistor.

I'd do something like this:

```cpp
    void begin() final override {
      pinMode(spin, INPUT_PULLUP);
    }
    void update() final override {
      int speed = speedMultiply;
      speed *= digitalRead(spin) == HIGH ? 1 : 4;
      long currentPosition = encoder.read();
      long difference = (currentPosition - previousPosition) / pulsesPerStep;
      if (difference) {
        sender.send(difference * speed, address);
        previousPosition += difference * pulsesPerStep;
      }
    }
```

It expects you to connect the switch between the Arduino pin and ground.

sucrz, 2020-04-18:
Thanks for the reply it work very well!

For the test I've use a ky040 encoder breakout board, each pins (A/B encoder pins) and switch pin are connected to digital Input.

dabonneville, 2020-04-29:
Hi @sucrz, interested in your acceleration addon, can you share it ?
Thanks

sucrz, 2020-04-30:
Hi @dabonneville you'll have to modifiy 2 files from the lib folder in the arduino lib folder : 
(Before modifing any file in the lib make sure to create a copy of the original file and rename it to identify easy the original, something like "filename-original" )

Control-Surface-master\src\MIDI_Outputs\CCRotaryEncoder.hpp

`
#pragma once

#include <MIDI_Outputs/Abstract/MIDIRotaryEncoder.hpp>
#include <MIDI_Senders/RelativeCCSender.hpp>

BEGIN_CS_NAMESPACE

/**
 * @brief   A class of MIDIOutputElement%s that read the input of a **quadrature
 *          (rotary) encoder** and send out relative MIDI **Control Change**
 *          events.
 * 
 * This version cannot be banked.
 *
 * @note    To use this class, include the [PJRC Encoder library]
 *          (https://github.com/PaulStoffregen/Encoder) before the
 *          Control-Surface library.
 *
 * @ingroup MIDIOutputElements
 */
class CCRotaryEncoder : public MIDIRotaryEncoder<RelativeCCSender> {
  public:
    /**
     * @brief   Construct a new CCRotaryEncoder object with the given pins, 
     *          address, channel, speed factor, and number of pulses per step.
     * 
     * @param   pins
     *          A list of the two pins connected to the A and B outputs of the
     *          encoder.  
     *          The internal pull-up resistors will be enabled by the Encoder
     *          library.
     * @param   address
     *          The MIDI address containing the controller number [0, 119], 
     *          channel [CHANNEL_1, CHANNEL_16], and optional cable number 
     *          [0, 15].
     * @param   speedMultiply
     *          A constant factor to increase the speed of the rotary encoder.
     *          The difference in position will just be multiplied by this 
     *          factor. 
     * @param   pulsesPerStep
     *          The number of pulses per physical click of the encoder.
     *          For a normal encoder, this is 4. If you want to increase the
     *          resolution, for the use of Jog wheels, for example, you can go
     *          as 1.  
     *          Whereas a greater speedMultiplier factor will increase the 
     *          speed, increasing the number of pulsesPerStep will result in a 
     *          lower speed.
     * @param   sender
     *          The MIDI sender to use.
     */
    CCRotaryEncoder(const EncoderPinList &pins,
                    int8_t spin,
                    const MIDIAddress &address,
                    int8_t speedMultiply = 1, uint8_t pulsesPerStep = 4,
                    const RelativeCCSender &sender = {})
        : MIDIRotaryEncoder(pins, spin, address, speedMultiply, pulsesPerStep,
                            sender) {}

// For tests only (PJRC Encoder library's copy constructor doesn't work)
#ifndef ARDUINO
    CCRotaryEncoder(const Encoder &encoder, int8_t spin, const MIDIAddress &address,
                    int8_t speedMultiply, uint8_t pulsesPerStep,
                    const RelativeCCSender &sender = {})
        : MIDIRotaryEncoder(encoder, spin, address, speedMultiply, pulsesPerStep,
                            sender) {}
#endif
};

END_CS_NAMESPACE
`
Control-Surface-master\src\MIDI_Outputs\Abstract\MIDIRotaryEncoder.hpp

`
#pragma once

#if not defined(Encoder_h_) && not defined(IDE)
#error                                                                         \
"The PJRC Encoder library should be included before the Control-Surface "  \
"library. (#include <Encoder.h>)"
#endif

#include <Def/Def.hpp>
#include <Encoder.h>
#include <MIDI_Outputs/Abstract/MIDIOutputElement.hpp>

BEGIN_CS_NAMESPACE

/**
   @brief   An abstract class for rotary encoders that send MIDI events.
*/
template <class Sender>
class MIDIRotaryEncoder : public MIDIOutputElement {
  protected:
    /**
       @brief   Construct a new MIDIRotaryEncoder.

       @todo    Documentation
    */
    MIDIRotaryEncoder(const EncoderPinList &pins,
                      int8_t spin,
                      const MIDIAddress &address,
                      int8_t speedMultiply, uint8_t pulsesPerStep,
                      const Sender &sender)
      : encoder{pins.A, pins.B}, spin(spin), address(address),
        speedMultiply(speedMultiply),
        pulsesPerStep(pulsesPerStep), sender(sender) {}

    // For tests only
#ifndef ARDUINO
    MIDIRotaryEncoder(const Encoder &encoder,
                      int8_t spin,
                      const MIDIAddress &address,
                      int8_t speedMultiply, uint8_t pulsesPerStep,
                      const Sender &sender)
      : encoder{encoder}, spin(spin), address(address), speedMultiply(speedMultiply),
        pulsesPerStep(pulsesPerStep), sender(sender) {}
#endif

  public:
    void begin() final override {
      pinMode(spin, INPUT_PULLUP);
    }
    void update() final override {
      int speed = speedMultiply;
      speed *= digitalRead(spin) == HIGH ? 1 : 4;
      long currentPosition = encoder.read();
      long difference = (currentPosition - previousPosition) / pulsesPerStep;
      if (difference) {
        sender.send(difference * speed, address);
        previousPosition += difference * pulsesPerStep;
      }
    }

  private:
    Encoder encoder;
    int8_t spin;
    const MIDIAddress address;
    const int8_t speedMultiply;
    const uint8_t pulsesPerStep;
    long previousPosition = 0;

  public:
    Sender sender;
};

END_CS_NAMESPACE
`

Once you've modified the file you have to modify the encoder déclaration like that : 
CCRotaryEncoder enc = { {A0, A1}, mux.pin(0), MIDI_CC::Bank_Select, 4, 4 };
CCRotaryEncoder enc = { {Pin A, Pin B}, SwitchPin, Midi Address, Speed Multiply, Pulse per step };