tttapa/Control-Surface #178: Arduino Mega 2560 Compatibility - Keyboard Piano Sensitivity

seileremerson, 2020-04-19:
I would like to know if the library is compatible with the arduino mega 2560.
I would like to create a Midi controller, with an old keyboard with sensitive keys. I managed to do it with Midi.h, but I don't know how to activate the sensitivity (speed) of the keys over there.
It consists of double-sided keys with a double matrix.
Would it be possible to use Control-Surface to do this?

tttapa, 2020-04-19:
Yes, the library is compatible with the Arduino Mega, but it has no built-in support for velocity-sensitive keyboard matrices.

seileremerson, 2020-04-21:
I understand .... and is that speed control (sensitivity) possible in the MIDI library?
I had a code rolling as a keypad and MIDI, but I did not manage to activate a double matrix, just a metade, so I did not have a speed (sensitivity).
My code: 

#include <MIDI.h>
#include <Keypad.h>

const byte ROWS = 8; // 8 Linhas
const byte COLS = 8; // 8 Colunas
 
char keys[COLS][ROWS] = {
  {36, 37, 38, 39, 40, 41, 42, 43},
  {44, 45, 46, 47, 48, 49, 50, 51},
  {52, 53, 54, 55, 56, 57, 58, 59},
  {60, 61, 62, 63, 64, 65, 66, 67},
  {68, 69, 70, 71, 72, 73, 74, 75},
  {76, 77, 78, 79, 80, 81, 82, 83},
  {84, 85, 86, 87, 88, 89, 90, 91},
  {92, 93, 94, 95, 96, 97, 98, 99}
};
 
byte rowPins[ROWS] = {22, 26, 30, 34, 38, 42, 46, 50}; //connect to the row pinouts of the kpd
byte colPins[COLS] = {A0, A2, A4, A6, A8, A10, A12, A14}; //connect to the column pinouts of the kpd

Keypad kpd = Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS );

byte pressed = 32;
byte chanel = 0; // MIDI channel to use

void setup() {
 Serial.begin(115200); // set this the same as Hairless
}

void loop() {
 // Fills kpd.key[49] array with up-to 10 active keys.
 // Returns true if there are ANY active keys.
 if (kpd.getKeys())
 {
   for (int i = 0; i < LIST_MAX; i++) // Scan the whole key list.
   {
     if ( kpd.key[i].stateChanged )   // Only find keys that have changed state.
     {
       pressed = kpd.key[i].kchar + 12;
       switch (kpd.key[i].kstate) {  // Report active key state : IDLE, PRESSED, HOLD, or RELEASED
         case PRESSED:
           sendMIDI(chanel | 0x90, pressed, 100);
           break;
          case RELEASED:
           sendMIDI(chanel | 0x80, pressed, 64);
           break;
       }
     }
   }
 }
}  // End loop

void sendMIDI(byte type, byte note, byte velocity){
  Serial.write(type);
  Serial.write(note & 0x7F);
  Serial.write(velocity);
}

tttapa, 2020-04-23:
> I understand .... and is that speed control (sensitivity) possible in the MIDI library?

The velocity sensitivity of your keyboard matrix has nothing to do with what MIDI library you're using. Both the MIDI library and the Control Surface library can send MIDI Note events with velocity just fine. It's just a matter of measuring the velocity of your keyboard matrices.

By the way, the code you posted doesn't use the MIDI library, it just includes its header file without using any of its functions.

lmeucchi, 2020-06-07:
Hello, a solution would be to use this code written by Daniel Moura .... mine is already working very well with a sensitive yamaha keyboard.

This sketch uses the DIO2 library, which performs a quick scan for the inputs and outputs, allowing you to read the two keyboard arrays quickly

Pieter, would it be possible to include this quick scan in your control surface library?

Thank you

[https://github.com/oxesoft/keyboardscanner](url)

```cpp

/*
Moura's Keyboard Scanner: turn you broken (or unused) keyboard in a MIDI controller
Copyright (C) 2017 Daniel Moura <oxe@oxesoft.com>
This code is originally hosted at https://github.com/oxesoft/keyboardscanner
This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.
This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.
You should have received a copy of the GNU General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#include <DIO2.h> // install the library DIO2

#define KEYS_NUMBER 61

#define KEY_OFF               0
#define KEY_START             1
#define KEY_ON                2
#define KEY_RELEASED          3
#define KEY_SUSTAINED         4
#define KEY_SUSTAINED_RESTART 5

#define MIN_TIME_MS   3
#define MAX_TIME_MS   50
#define MAX_TIME_MS_N (MAX_TIME_MS - MIN_TIME_MS)

#define PEDAL_PIN     21

//find out the pins using a multimeter, starting from the first key
//see the picture key_scheme.png to understand how to map the inputs and outputs

//the following configuration is specific for PSR530
//thanks Leandro Meucchi, from Argentina, by the PDF
//take a look at the scheme detailed in PSR530.pdf and modify the following mapping according to the wiring of your keyboard
#define PIN_A1  33
#define PIN_A2  32
#define PIN_A3  31
#define PIN_A4  30
#define PIN_A5  29
#define PIN_A6  28
#define PIN_A7  27
#define PIN_A8  26
#define PIN_A9  25
#define PIN_A10 24
#define PIN_A11 23
#define PIN_A12 22
#define PIN_B1  40
#define PIN_B2  39
#define PIN_B3  38
#define PIN_B4  37
#define PIN_B5  36
#define PIN_B6  35
#define PIN_B7  34
[... 424 lines cut ...]
                }
                break;
            case KEY_SUSTAINED_RESTART:
                if (state_index == 0 && !*signal)
                {
                    *state = KEY_SUSTAINED;
                    digitalWrite(13, HIGH);
                    break;
                }
                if (state_index == 1 && *signal)
                {
                    *state = KEY_ON;
                    send_midi_event(0x80, key, MAX_TIME_MS);
                    send_midi_event(0x90, key, millis() - *ktime);
                }
                break;
            }
            signal++;
        }
        state++;
        ktime++;
    }
}

```