tttapa/Control-Surface #179: CCSelection - how to change the channel

santta, 2020-04-20:
Hi, how to change the channel sending CC in the CCSelection class?

I need to use any channel except 1

```cpp

 
class CCSelection : public Selectable<N> {
  public:
    CCSelection(const Array<MIDIAddress, N> &addresses)
      : addresses{addresses} {}

    void select(setting_t setting) override {
      setting = Selectable<N>::validateSetting(setting);
      Control_Surface.sendCC(addresses[setting], 0x7F);
    }

  private:
    Array<MIDIAddress, N> addresses;
};

CCSelection<4> cc = {{0x7C, 0x7D, 0x64, 0x65}};

tttapa, 2020-04-20:
You can add it to the address array:

```cpp
CCSelection<4> cc = {{
  {0x7C, CHANNEL_6}, 
  {0x7D, CHANNEL_8}, 
  {0x64, CHANNEL_10}, 
  {0x65, CHANNEL_12},
}};
```

[MIDI Addresses](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/dc6/md_pages_Basics.html)

santta, 2020-04-21:
I get an error:

Arduino: 1.8.12 (Windows 10), Плата:"Arduino Leonardo"

leonardo_sketch:77:2: error: could not convert '{{{124, CS::CHANNEL_6}, {125, CS::CHANNEL_8}, {100, CS::CHANNEL_10}, {101, CS::CHANNEL_12}}}' from '<brace-enclosed initializer list>' to 'CCSelection<4>'

exit status 1
could not convert '{{{124, CS::CHANNEL_6}, {125, CS::CHANNEL_8}, {100, CS::CHANNEL_10}, {101, CS::CHANNEL_12}}}' from '<brace-enclosed initializer list>' to 'CCSelection<4>'

santta, 2020-04-21:
could not convert '{{{124, CS::CHANNEL_6}, {125, CS::CHANNEL_8}, {100, CS::CHANNEL_10}, {101, CS::CHANNEL_12}}}' from '<brace-enclosed initializer list>' to 'CCSelection<4>'

tttapa, 2020-04-21:
Apologies, apparently, you need an extra pair of braces: one pair for the CCSelection, two for the array, and one for each individual address.

I fixed it in my previous reply.

santta, 2020-04-21:
Thanks for the answer and help, everything works fine.

closed 2020-04-21 by santta