tttapa/Control-Surface #180: control surface stop audio library

benwadub, 2020-04-20:
hi Pieter, i have a synth project that work fine but when I try to enable the control surface library nothing work, do you know what can interfere between the 2 library please?
```cpp

#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <SerialFlash.h>
#include <Control_Surface.h>
// GUItool: begin automatically generated code
AudioSynthWaveformDc     dc1;            //xy=55,229.9999976158142
AudioSynthWaveformModulated waveformMod9;   //xy=228.57149451119562,475.7143028804235
AudioSynthWaveformModulated waveformMod3;   //xy=231.42856979370117,176.85713577270508
AudioSynthWaveformModulated waveformMod4;   //xy=231.42856979370117,211.85713577270508
AudioSynthWaveformModulated waveformMod5;   //xy=231.42858123779297,255.42856407165527
AudioSynthWaveformModulated waveformMod6;   //xy=231.42858123779297,290.4285640716553
AudioSynthWaveformModulated waveformMod7;   //xy=231.42858123779297,325.4285640716553
AudioSynthWaveformModulated waveformMod8;   //xy=231.42858123779297,362.4285640716553
AudioSynthWaveformModulated waveformMod2;   //xy=232.42856979370117,139.85713577270508
AudioSynthWaveformModulated waveformMod1;   //xy=233.42856979370117,102.85713577270508
AudioSynthNoiseWhite     noise1;         //xy=235.71430587768555,527.1428508758545
AudioSynthNoisePink      pink1;          //xy=244.28573608398438,571.4285678863525
AudioEffectEnvelope      envelope7;      //xy=410.4285697937012,325.8571357727051
AudioEffectEnvelope      envelope6;      //xy=414.4285697937012,289.8571357727051
AudioEffectEnvelope      envelope1;      //xy=415.4285697937012,102.85713577270508
AudioEffectEnvelope      envelope5;      //xy=415.4285697937012,254.85713577270508
AudioEffectEnvelope      envelope8;      //xy=415.4285697937012,362.8571357727051
AudioEffectEnvelope      envelope2;      //xy=416.4285697937012,139.85713577270508
AudioEffectEnvelope      envelope4;      //xy=416.4285697937012,216.85713577270508
AudioEffectEnvelope      envelope3;      //xy=417.4285697937012,178.85713577270508
AudioMixer4              mixer4;         //xy=438.5714530944824,517.1428680419922
AudioEffectEnvelope      envelope9;      //xy=601.4286117553711,477.1428565979004
AudioMixer4              mixer1;         //xy=619.4285697937012,171.85713577270508
AudioMixer4              mixer2;         //xy=625.4285697937012,282.8571357727051
AudioMixer4              mixer3;         //xy=806.7858390808105,194.64279556274414
AudioSynthWaveformDc     dc3;            //xy=900.0000152587891,121.25000190734863
AudioSynthWaveformDc     dc2;            //xy=959.107250213623,435.35712242126465
AudioSynthWaveformModulated waveformMod10;  //xy=1038.750015258789,163.75000095367432
AudioFilterStateVariable filter1;        //xy=1068.5714378356934,290.0000591278076
AudioEffectEnvelope      envelope10;     //xy=1099.4643096923828,432.8571529388428
AudioEffectMultiply      multiply1;      //xy=1225.0000190734863,126.25000071525574
AudioMixer4              mixer7;         //xy=1283.7500190734863,245.00000190734863
AudioEffectDelay         delay1;         //xy=1491.1905250549316,414.9405460357666
AudioEffectFreeverbStereo freeverbs1;     //xy=1511.190486907959,253.51194190979004
AudioMixer4              mixer5;         //xy=1671.1904525756836,140.65477180480957
AudioMixer4              mixer6;         //xy=1672.6188468933105,363.5119228363037
AudioOutputUSB           usb1;
AudioOutputAnalogStereo  dacs1;          //xy=1821.4761123657227,290.65477991104126
AudioConnection          patchCord1(dc1, 0, waveformMod1, 1);
AudioConnection          patchCord2(dc1, 0, waveformMod2, 1);
AudioConnection          patchCord3(dc1, 0, waveformMod3, 1);
AudioConnection          patchCord4(dc1, 0, waveformMod4, 1);
AudioConnection          patchCord5(dc1, 0, waveformMod5, 1);
AudioConnection          patchCord6(dc1, 0, waveformMod6, 1);
AudioConnection          patchCord7(dc1, 0, waveformMod7, 1);
AudioConnection          patchCord8(dc1, 0, waveformMod8, 1);
AudioConnection          patchCord9(dc1, 0, waveformMod9, 1);
AudioConnection          patchCord10(waveformMod9, 0, mixer4, 0);
AudioConnection          patchCord11(waveformMod3, envelope3);
AudioConnection          patchCord12(waveformMod4, envelope4);
AudioConnection          patchCord13(waveformMod5, envelope5);
AudioConnection          patchCord14(waveformMod6, envelope6);
AudioConnection          patchCord15(waveformMod7, envelope7);
AudioConnection          patchCord16(waveformMod8, envelope8);
AudioConnection          patchCord17(waveformMod2, envelope2);
AudioConnection          patchCord18(waveformMod1, envelope1);
AudioConnection          patchCord19(noise1, 0, mixer4, 1);
AudioConnection          patchCord20(pink1, 0, mixer4, 2);
AudioConnection          patchCord21(envelope7, 0, mixer2, 2);
AudioConnection          patchCord22(envelope6, 0, mixer2, 1);
[... 496 lines cut ...]
         

  delay1.delay(0,phs);      //stereoize
  
  dc1.amplitude(shpwm);     //pwm
  dc3.amplitude(shpwm2);    //Lfo waveform pwm
  dc2.amplitude(md);        //mod depth env 

  pink1.amplitude(pnk1);
  noise1.amplitude(nse1);
  freeverbs1.roomsize(rsize);
  mixer5.gain(0, revol);
  mixer6.gain(0, revol);

  
    
    for (byte i = 0; i < POLYPHONY; i++) {
     waveforms[i]->begin(wavfrm);
     waveforms[i]->amplitude(vely); 
    }
   
  usbMIDI.read();
}

```

tttapa, 2020-04-20:
You have to define at least one MIDI interfaces before calling Control_Surface.begin(). You cannot use `usbMIDI` directly if you're using a `USBMIDI_Interface`. You have to use the `USBMIDI_Interface`.

benwadub, 2020-04-20:
I ll try again tonight but I thought I à ready tried it.
But you make me think to something, my synth is on the teensy and the controller too, how would you make them connect together, usb midi like I tried is not the good solution I think?
Din t I need to make 2 serial interface? One out for the pots and one in to go to the synth?

benwadub, 2020-04-21:
hi, ok so I tried with midi pipe but I can't make work the pot to send midi to the synth, do you think it s doable?
```cpp
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <SerialFlash.h>
#include <MIDI.h>
#include <Control_Surface.h>
/*
#define MUXcutoff 1
#define MUXreso 2
#define MUXrevol 3
#define MUXrsize 4
#define MUXatk 5
#define MUXdcy 6
#define MUXsus 7
#define MUXrls 8
#define MUXatkf 9
#define MUXdcyf 10
#define MUXsusf 11
#define MUXrlsf 12
#define MUXmd 13
#define MUXphs 14
#define MUXswc 15
#define MUXswc2 16
#define MUX2nse1 1
#define MUX2pnk1 2
#define MUX2amp2 3
#define MUX2shpwm 4
#define MUX2shpwm2 5
#define MUX2lfoamp 6
#define MUX2lfofreq 7

//MIDI CC control numbers

#define CCcutoff 1
#define CCreso 2
#define CCrevol 19
#define CCrsize 20
#define CCatk 13
#define CCdcy 14
#define CCsus 15
#define CCrls 16
#define CCatkf 9
#define CCdcyf 10
#define CCsusf 11
#define CCrlsf 12
#define CCmd 3
#define CCphs 4
#define CCswc 5
#define CCswc2 22
#define CCnse1 6
#define CCpnk1 7
#define CCamp2 8
#define CCshpwm 24
#define CCshpwm2 23
#define CClfoamp 17
#define CClfofreq 18
*/

USBMIDI_Interface midiusb;
HardwareSerialMIDI_Interface midiser = Serial1;
MIDI_PipeFactory<3> pipes;

// GUItool: begin automatically generated code
AudioSynthWaveformDc     dc1;            //xy=55,229.9999976158142
AudioSynthWaveformModulated waveformMod9;   //xy=228.57149451119562,475.7143028804235
AudioSynthWaveformModulated waveformMod3;   //xy=231.42856979370117,176.85713577270508
AudioSynthWaveformModulated waveformMod4;   //xy=231.42856979370117,211.85713577270508
[... 620 lines cut ...]
  filter1.frequency(cutoff);
  filter1.resonance(reso);
         

  delay1.delay(0,phs);      //stereoize
  
  dc1.amplitude(shpwm);     //pwm
  dc3.amplitude(shpwm2);    //Lfo waveform pwm
  dc2.amplitude(md);        //mod depth env 

  pink1.amplitude(pnk1);
  noise1.amplitude(nse1);
  freeverbs1.roomsize(rsize);
  mixer5.gain(0, revol);
  mixer6.gain(0, revol);

  
    
    for (byte i = 0; i < POLYPHONY; i++) {
     waveforms[i]->begin(wavfrm);
     waveforms[i]->amplitude(vely); 
    }
   
  usbMIDI.read();
}```

tttapa, 2020-04-23:
I still don't understand what you're trying to do. If you have the potentiometers and the synth on the same microcontroller, why do you need to connect them over a virtual MIDI connection? Just read the values of the potentiometer directly.

Your code now contains functions for three different MIDI libraries. That's just asking for trouble. You don't need `MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, MIDI)` if you're already using a `HardwareSerialMIDI_Interface`. You don't need `usbMIDI` if you're already using a `USMIDI_Interface`.

benwadub, 2020-04-23:
Thanks for your interest in my problem!
As I learnt programming with your code I thought it was easier for me to only use your library!
I don t know how to directly read the values with my mux! I m loosed now! :-)

tttapa, 2020-04-27:
You can use the FilteredAnalog class. It's what CCPotentiometer uses under the hood.

https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/dbe/1_8FilteredAnalog_8ino-example.html

If you want to go more low-level, you can call the Arduino function analogRead directly, but it doesn't do any filtering, so it'll be pretty noisy compared to FilteredAnalog.

benwadub, 2020-04-27:
Thanks I ll try it!!