tttapa/Control-Surface #181: Custom ManyButtonsSelector - Sub banks

wayna, 2020-04-20:
Hello,

The midi controller i'm trying to build will be used control a lighting software.
In this software there are six parameters pages which can be called with midi buttons.
Each parameter page has N tabs, switching from one tab to another is done by pressing multiple times the same button (the same midi message is needed)
Each tab contains four parameters, controlled by 4 encoders, so when i switch from on tab to another the encoders must have a different adress, or channel

What i would like to do is adapting the controller behaviour to fit to the software behaviour.

For the example, with 3 parameters, 3 tabs by parameter, and one encoder:
Button 1 - 1 press -> Opens parameter page "Color" - tab 1  -> Encoder channel 1
                  2 press -> Opens parameter  page "Color" - tab 2 -> Encoder channel 2
                  3 press -> Opens parameter  page "Color" - tab 3 -> Encoder channel 3

Button 2 - 1 press -> Opens parameter page "Focus" - tab 1  -> Encoder channel 4
                  2 press -> Opens parameter  page "Focus" - tab 2 -> Encoder channel 5

Button 3 - 1 press -> Opens parameter page "Image" - tab 1  -> Encoder channel 6
                  2 press -> Opens parameter  page "Image" - tab 2 -> Encoder channel 7
                  3 press -> Opens parameter  page "Image" - tab 3 -> Encoder channel 8
                  4 press -> Opens parameter  page "Image" - tab 4 -> Encoder channel 9

Each button acts like a switch selector wrapped -> Loops through N tabs
When i press a button, then another, i recall the last viewed tab of the corresponding button.

I thought about 2 ways to do this:
- Bankable Encoder affected to multiple banks (like in this example, 3 banks with 3 switch selectors), is that doable?
- Create a custom ManyButtons Selector:
If i figure it out right, i'd need to create a custom selector based on [ManyButtonsSelector.hpp](https://github.com/tttapa/Control-Surface/blob/f73200f035ae626046f6edbd501f4662416da96e/src/Selectors/ManyButtonsSelector.hpp) the same way you did in your [Custom-MIDI-Sender.ino](https://github.com/tttapa/Control-Surface/blob/cf133ce061793c517ce755214dfae2796a09bf6b/examples/Extending-the-library/Custom-MIDI-Sender/Custom-MIDI-Sender.ino) sketch, am i right?

What would you think would be the best approach?
Many Thanks

tttapa, 2020-04-21:
> Bankable Encoder affected to multiple banks (like in this example, 3 banks with 3 switch selectors), is that doable?

This should be possible, but only if each of the sub-banks has the same number of banks, which is not the case in your example.

> Create a custom ManyButtons Selector:
If i figure it out right, i'd need to create a custom selector based on ManyButtonsSelector.hpp the same way you did in your Custom-MIDI-Sender.ino sketch, am i right?

I think that would be the easiest ans most flexible approach.

wayna, 2020-04-22:
Okay thanks for the feedback.

I will try to create a custom selector then.