tttapa/Control-Surface #182: BLE conflict on ESP32

fuegovic, 2020-04-22:
BluetoothMIDI_Interface works by itself but if "USBMIDI_Interface" or "HardwareSerialMIDI_Interface" is also present the Bluetooth device is not discoverable anymore. (BLE stop working)

I discovered the problem while trying to use the "MIDI_Pipes-Routing" example. I made some tests and quickly realize that the problem was with the other MIDI interfaces and not with the routing itself. 

Also the default BLE name is too long to work in some OS. From what I understand there's a a bug or something that prevent the use of name longer than 5 characters in IOS (and in windows 10 from my experiences), but it does work in Android.

fuegovic, 2020-04-22:
The error was in front of the monitor all along, I just had to redirect the HardwareSerialMIDI_Interface to "Serial" instead of "Serial1". 

Here's the code for the easiest way to make a DIN5 MIDI to MIDI BLE adapter!

Thanks, the library is amazing!

```
#include <Control_Surface.h>
#include <MIDI_Interfaces/BluetoothMIDI_Interface.hpp>

BluetoothMIDI_Interface MIDI_BLE;
HardwareSerialMIDI_Interface MIDI_SERIAL = Serial;

MIDI_PipeFactory<2> pipes;

void setup() {
  MIDI_BLE >> pipes >> MIDI_SERIAL; // all incoming midi from USB is sent to serial
  MIDI_BLE << pipes << MIDI_SERIAL; // all incoming midi from Serial is sent to USB

  MIDI_SERIAL.begin();
  MIDI_BLE.begin();
}

void loop() {
  MIDI_BLE.update();
  MIDI_SERIAL.update();
}
```

closed 2020-04-22 by fuegovic

tttapa, 2020-04-23:
Glad to hear you got it working!

GiordanoSala, 2023-02-03:
is it possible to use Serial1 (RX2, TX2) ?
Everything works with DIN 5, 6N138 and an OLED for track names, when I transfer the program with HardwareSerialMIDI_Interface MIDI_SERIAL = Serial1;
The OLED screen just won't turn on.
All ok with Serial RX0, TX0.