tttapa/Control-Surface #183: How to acess .getPosition from VPotRing.hpp

kevinagnes, 2020-04-23:
Hi Pieter, I am trying to add support for displaying the Smart Controls using VPOT.
I am having trouble trying to figure out why I can't get the position from the Midi Input Element.
VS Code says: 
'uint8_t CS::MCU::VPotRing_Base<NumValues, Callback>::getValue() const [with unsigned char NumValues = 1; Callback = CS::MCU::VPotEmptyCallback; uint8_t = unsigned char]' is private within this context'

Normally, when I .getValue() from a CCValue Midi Input, I have no problems. Any idea on that?

Cheers, Kevin.

```cpp

//MIDI Output
ankable::BorrowedCCRotaryEncoder encSMART[10] =  // Smart Control using V_POTs
{
  {{bank[0], BankType::CHANGE_CHANNEL}, enc0, {MCU::V_POT_1,  CHANNEL_1}, 2, 4,},
  {{bank[1], BankType::CHANGE_CHANNEL}, enc1, {MCU::V_POT_2,  CHANNEL_1}, 2, 4,},
  {{bank[2], BankType::CHANGE_CHANNEL}, enc2, {MCU::V_POT_3,  CHANNEL_1}, 2, 4,},
  {{bank[3], BankType::CHANGE_CHANNEL}, enc3, {MCU::V_POT_4,  CHANNEL_1}, 2, 4,},
  {{bank[4], BankType::CHANGE_CHANNEL}, enc4, {MCU::V_POT_5,  CHANNEL_1}, 2, 4,},
  {{bank[5], BankType::CHANGE_CHANNEL}, enc5, {MCU::V_POT_6,  CHANNEL_1}, 2, 4,},
  {{bank[6], BankType::CHANGE_CHANNEL}, enc6, {MCU::V_POT_7,  CHANNEL_1}, 2, 4,},
  {{bank[7], BankType::CHANGE_CHANNEL}, enc7, {MCU::V_POT_8,  CHANNEL_1}, 2, 4,},
  {{bank[8], BankType::CHANGE_CHANNEL}, enc8, {           9,  CHANNEL_5}, 2, 4,},
  {{bank[9], BankType::CHANGE_CHANNEL}, enc9, {          10,  CHANNEL_5}, 2, 4,},   
};

// MIDI Display Input
MCU::VPotRing vPotRing[] = 
{
  {1,  CHANNEL_1},
  {2,  CHANNEL_1},
  {3,  CHANNEL_1},
  {4,  CHANNEL_1},
  {5,  CHANNEL_1},
  {6,  CHANNEL_1},
  {7,  CHANNEL_1},
  {8,  CHANNEL_1}, 
}; 

// MAIN Display Update
for (int j=0;j<8;j++)
{
  tft.fillArc( X, Y, Radius, Stroke, 0, vPotRing[j].getPosition() , Color);
}

```

tttapa, 2020-04-24:
The `getValue` method returns the raw MIDI value, not the position. That's why it's private.

You can access all relevant data through the [`getPosition`](https://tttapa.github.io/Control-Surface-doc/Doxygen/da/d50/classMCU_1_1IVPotRing.html#a76b64156640ab02b9676fc19f3427d82), `getMode` and `getCenterLed` methods.

If you want to display the position of the V-pot as intended by the MCU developers, you can use the `getStartOn` and `getStartOff` methods, as demonstrated here:

https://github.com/tttapa/Control-Surface/blob/2b06f004c4d13c6e98d213ddcf0238ccfd5fb775/src/Display/MCU/VPotDisplay.hpp#L26-L29

These methods return the first segment of the range that should be on, and the first segment that should be off, respectively. They take into account both the position and the mode of the V-pot ring.

kevinagnes, 2020-04-25:
Thanks Pieter!
.getPosition() is now working.

Now I am having other problems:
 
1. I am only getting values for VPOT 1 to VPOT 4 and nothing from 5 to 8. (see attached screen)
[image]

2. The display update for the Smart Controls is very slow (12 steps), compared to the EQ Mode 128 steps.

The update is based on the comparison of current from previous values. 
````cpp 
if (cccc[j].getPosition() != oldcccc[j])
````

Do you have any idea on how to work around this issues?

````cpp
Bankable::CCValue<4> cc[10] =
{
  {{bank[0], BankType::CHANGE_ADDRESS}, {ccEnc[0], CHANNEL_9}},
  {{bank[1], BankType::CHANGE_ADDRESS}, {ccEnc[1], CHANNEL_1}},
  {{bank[2], BankType::CHANGE_ADDRESS}, {ccEnc[2], CHANNEL_1}},
  {{bank[3], BankType::CHANGE_ADDRESS}, {ccEnc[3], CHANNEL_1}},
  {{bank[4], BankType::CHANGE_ADDRESS}, {ccEnc[4], CHANNEL_1}},
  {{bank[5], BankType::CHANGE_ADDRESS}, {ccEnc[5], CHANNEL_1}},
  {{bank[6], BankType::CHANGE_ADDRESS}, {ccEnc[6], CHANNEL_1}},
  {{bank[7], BankType::CHANGE_ADDRESS}, {ccEnc[7], CHANNEL_1}},
  {{bank[8], BankType::CHANGE_ADDRESS}, {ccEnc[8], CHANNEL_1}},
  {{bank[9], BankType::CHANGE_ADDRESS}, {ccEnc[9], CHANNEL_1}},
};

MCU::VPotRing cccc[] = 
{
  {1,  CHANNEL_1},
  {2,  CHANNEL_1},
  {3,  CHANNEL_1},
  {4,  CHANNEL_1},
  {5,  CHANNEL_1},
  {6,  CHANNEL_1},
  {7,  CHANNEL_1},
  {8,  CHANNEL_1}, 
};   

void updateSmartControls()
{
  tft.setArcParams(11);
  for (int j=0; j<8; j++)
    {   
      if (cccc[j].getPosition() != oldcccc[j]) // 12 steps
      {

        if (cccc[j].getPosition() < oldcccc[j]) 
                    tft.fillArc(xxPos[j],yyPos[j], circleSize - 5, 6, cccc[j].getPosition(), 0, BLACK);
        if (cccc[j].getPosition() > oldcccc[j]) 
                    tft.fillArc(xxPos[j],yyPos[j], circleSize - 5, 6, 0, cccc[j].getPosition(), smartColour);  
      }    
    }
   tft.setArcParams(127); 
}   

void updateEqControls()
{
    for (int j=0; j<8; j++)
    {   
        if (cc[j].getValue() != oldcc[j]) // 128 steps
        { 
          timer[j] = millis();

          if (cc[j].getValue() < oldcc[j]) 
                        tft.fillArc(xPos[j],yPos[j], circleSize+5, 5,cc[j].getValue(), 0, BLACK);
          if (cc[j].getValue() > oldcc[j]) 
                        tft.fillArc(xPos[j],yPos[j], circleSize+5, 5, 0, cc[j].getValue(), eqColour);  
        }
        oldcc[j] = cc[j].getValue();
    }
}

````

tttapa, 2020-04-25:
> I am only getting values for VPOT 1 to VPOT 4 and nothing from 5 to 8. (see attached screen)

I don't have an explanation for that. Can you verify using a MIDI monitor that your software is sending data for all 8 VPots?

> The display update for the Smart Controls is very slow (12 steps), compared to the EQ Mode 128 steps.

This is a limitation of the Mackie Control Universal protocol. There's nothing you can do about it, except using your own custom CC messages. (Your DAW needs to support this, of course.)

kevinagnes, 2020-04-26:
> Can you verify using a MIDI monitor that your software is sending data for all 8 VPots?

Yes. I receive normal messages. Here is an example for the first and last VPOT:
````
14:34:40.166	From Arduino Due	Control	1	B0 10 02
0	                 To Arduino Due	SysEx		F0 00 00 66 14 12 3A 31 2E 30 F7
14:34:41.439	From Arduino Due	Control	1	B0 17 02
0	                 To Arduino Due	SysEx		F0 00 00 66 14 12 69 36 32 F7
```` 

If I change the MCU::VPotRing Channel to anything other than 1, getPosition() is always 0.

> This is a limitation of the Mackie Control Universal protocol.

I understand.

tttapa, 2020-04-26:
The messages are not V-Pot messages, V-Pot messages use Control Change addresses 0x30 through 0x37, not 0x10 through 0x17.

I've just tried the following sketch, and it seems to work as expected:

```cpp
#include <Control_Surface.h>

MCU::VPotRing vpots[] = {
  {1,  CHANNEL_1},
  {2,  CHANNEL_1},
  {3,  CHANNEL_1},
  {4,  CHANNEL_1},
  {5,  CHANNEL_1},
  {6,  CHANNEL_1},
  {7,  CHANNEL_1},
  {8,  CHANNEL_1}, 
}; 

USBDebugMIDI_Interface midi;

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
  for (auto &vpot : vpots)
    Serial << vpot.getPosition() << '\t';
  Serial << endl;
}
```

> If I change the MCU::VPotRing Channel to anything other than 1, getPosition() is always 0.

The MCU protocol only sends V-Pot data on channel 1.

kevinagnes, 2020-04-26:
> V-Pot messages use Control Change addresses 0x30 through 0x37, not 0x10 through 0x17.

I didn't know that! I will try this tonight :)

I should change V_POT_1-8 to 48-55, right?

[image]

tttapa, 2020-04-26:
No, sending V-Pot data from the Arduino to the computer uses addresses 0x10 through 0x17, receiving V-Pot data uses addresses 0x30 through 0x37.

[image]
[image]

kevinagnes, 2020-04-26:
I think I know what is happening, I am using a MCU Pro and not the older MCU.
Maybe they changed the adress. I will check it tonight! :)

[image] 
[image]
[image]
[image]

cheers!

kevinagnes, 2020-04-27:
Ignore the last post. The problem was another MIDI Input device using the same Address.
Everything good now! :) 
[image]

Thanks Pieter!
Cheers :)

tttapa, 2020-04-27:
Glad to hear you got it working!

I don't think there's any significant difference between MCU Pro and MCU, the protocols should be very similar.