tttapa/Control-Surface #184: BankSelector and 4067 multiplexer

santta, 2020-04-26:
Hi, can I use in banks buttons connected through a 4067 multiplexer?

tttapa, 2020-04-26:
Yes, you can use it in exactly the same way as with other pins. Just supply `yourMultiplexer.pin(0)` as the pin argument to the bank selector constructor.

santta, 2020-04-27:
[image]

If in the test assembly I have 4 buttons for the bank selector ( connected to the arduino pins), and 4 LED 
for indication (connected to the arduino pins). 
Each of the selector buttons changes to a specific channel for all controller elements.

Also connected to my controller:
4 encoders, via Arduino pins. 
4 buttons via Mux4067

What an example sketch might look like?

[image]

tttapa, 2020-04-27:
What have you tried? You'll probably need the ManyButtonsSelectorLEDs class: https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d68/classManyButtonsSelectorLEDs.html

santta, 2020-04-27:
I already use ManyButtonsSelectorLEDs (https://github.com/tttapa/Control-Surface/issues/136) - this is a great selector.
I continue to study your Surface library. And now I'm looking for a way to change the channels sent by Midi CC with this selector

santta, 2020-04-30:
thanks, I was able to make a working sketch

    #include <Encoder.h>
    #include <Control_Surface.h> // Подключение библиотеки
    USBMIDI_Interface midi; // Подключение USB интерфейса

    Bank<4> bank(1);
    ManyButtonsSelectorLEDs<4> bankselector = {
    bank,
    {6, 7, 8, 9}, // button pins
    {10, 11, 12, 13}, // led pins
    };

     Bankable::CCRotaryEncoder enc[] = {
     {{bank, BankType::CHANGE_CHANNEL}, {2, 3}, 0x08, 1, 4},
     };

    CD74HC4067 mux2 = {A5,{A1, A2, A3, A4} };
    Bankable::CCButton button1[] = {
    {{bank, BankType::CHANGE_CHANNEL},  mux2.pin(1), { 0x0E, CHANNEL_1 } },
    };

    Bankable::CCButton butt[] = {
    {{bank, BankType::CHANGE_CHANNEL}, 4, {0x08, CHANNEL_2} }, // 
     };

      void setup() {
     Control_Surface.begin(); // Initialize Control Surface
     }

      void loop() {
      Control_Surface.loop(); // Update the Control Surface
      }

santta, 2020-04-30:
Pieter, thanks for your wonderful library, I enjoy using it.  Perhaps you see errors in my sketch or improvements to improve the sketch?

tttapa, 2020-04-30:
Thanks for sharing your code, it looks alright at first sight.

closed 2020-05-01 by santta