tttapa/Control-Surface #185: position switch

olheadinternet, 2020-04-27:
hey! having a lot of fun and learning a lot with your library.
I am looking to add position switches to my projects. 
I'm not sure this is the proper way to do this...
is there a way to declare value when pushing a cc button
 {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1},
so one could specify the value of each toggle on the position switch?
thanks!

tttapa, 2020-04-27:
What is a position switch? How is it wired? What kind of signals does it output? Do you have a specific part number or datasheet?

olheadinternet, 2020-04-27:
thanks for the reply... 
here is an example of an 8 position switch https://www.newegg.com/p/297-006X-00EB4
here is an example of a 4 position switch https://www.newegg.com/p/297-004Y-00449
in musical terms its like an oscillator selector (both shape and size) on a mini Moog / or virtually shape and octave on abletons analog. in function its just a switch with more than 3 positions... each position is basically just a latching button. But because it has more than 3 position and so does the virtual destination osc shape or step count or step type on some sequencers, I would need to output different values on general_purpose_controller_1 than just 0 and 127.
[image]

tttapa, 2020-04-27:
I see, thanks for the clarification. Do you know what kinds of MIDI messages your DAW is expecting? Is it just split evenly (0-15 is "1", 16-31 is "2", 32-47 is "3", etc)?

tttapa, 2020-04-27:
If that's the case, you could try something like this:

```cpp
#include <Control_Surface.h>

template <setting_t N>
class CCSelection : public Selectable<N> {
  public:
    CCSelection(MIDIAddress address) : address{address} {}

    void select(setting_t setting) override {
      setting = Selectable<N>::validateSetting(setting);
      Control_Surface.sendCC(address, setting * 128 / N + 64 / N - 1);
    }

  private:
    MIDIAddress address;
};

USBMIDI_Interface midi;

CCSelection<8> cc = {{MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}};

ManyButtonsSelector<8> selector = {
  cc,                       // what to select
  {2, 3, 4, 5, 6, 7, 8, 9}, // button pins
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

You can also change the values of the `CCButtons` but that's a less elegant solution in my opinion. 

You'll see that the [third constructor argument](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/daa/classCCButton.html#a8ff77ca457f85462169de01ca7e02238) is the `DigitalCCSender` to use. Its [constructor](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/db5/classDigitalCCSender.html) has two optional arguments to specify the CC values to use when the button is pressed or released.

For instance:

```cpp
CCButton button = {
  2, // pin
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}, // MIDI address
  {0x7F, 0x00} // value when pressed, value when released
};
```

olheadinternet, 2020-04-27:
yes! manybuttonsselector is more elegant... sorry, I'll spend more time in the master documents section before asking next time... I was working from the examples. Nice to know I could do it with ccbuttons. 

its this line that's splitting it evenly?
Control_Surface.sendCC(address, setting * 128 / N + 64 / N - 1);

tttapa, 2020-04-28:
Yes, that's the line. I chose evenly spaced intervals and used the middle of the intervals, rounded towards 0.

olheadinternet, 2020-04-30:
hey, so I'm getting a number of errors when trying to compile the many buttons selector, I'm not sure why... 

I also attempted the CCbutton method... unfortunately the switch I'm using sends the value when pressed of the new position before it sends the value when released of the previous position... is there a easy way to disable value when released?

```cpp
#include <Control_Surface.h>

template <setting_t N>
class CCSelection : public Selectable<N> {
  public:
    CCSelection(MIDIAddress address) : address{address} {}

    void select(setting_t setting) override {
      setting = Selectable<N>::validateSetting(setting);
      Control_Surface.sendCC(address, setting * 128 / N + 64 / N - 1);
    }

  private:
    MIDIAddress address;
};

USBMIDI_Interface midi;

CCSelection<4> cc = {{MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}};

ManyButtonsSelector<4> selector = {
  cc,                       // what to select
  {2, 3, 4, 5}, // button pins
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}

```

```cpp

MANYBUTTONTEST:6:29: error: expected ')' before 'address'
     CCSelection(MIDIAddress address) : address{address} {}
                             ^~~~~~~
MANYBUTTONTEST:6:57: error: expected unqualified-id before '{' token
     CCSelection(MIDIAddress address) : address{address} {}
                                                         ^
MANYBUTTONTEST:14:5: error: 'MIDIAddress' does not name a type
     MIDIAddress address;
     ^~~~~~~~~~~
/Users/zacharyhendel/Documents/Arduino/MANYBUTTONTEST/MANYBUTTONTEST.ino: In member function 'void CCSelection<N>::select(CS::setting_t)':
MANYBUTTONTEST:10:23: error: 'class CS::Control_Surface_' has no member named 'sendCC'
       Control_Surface.sendCC(address, setting * 128 / N + 64 / N - 1);
                       ^~~~~~
MANYBUTTONTEST:10:30: error: 'address' was not declared in this scope
       Control_Surface.sendCC(address, setting * 128 / N + 64 / N - 1);
                              ^~~~~~~
/Users/zacharyhendel/Documents/Arduino/MANYBUTTONTEST/MANYBUTTONTEST.ino: At global scope:
MANYBUTTONTEST:19:72: error: could not convert '{{CS::MIDI_CC::General_Purpose_Controller_1, CS::CHANNEL_1}}' from '<brace-enclosed initializer list>' to 'CCSelection<4>'
 CCSelection<4> cc = {{MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}};
                                                                        ^
exit status 1
expected ')' before 'address'
```

olheadinternet, 2020-04-30:
I changed value when released to 0x8F so it sends 143 and it works as expected 💁‍♀️
feel free to provide feedback, but this works for now.

closed 2020-04-30 by olheadinternet

tttapa, 2020-04-30:
The errors you're getting are because the code was written for the master version of the library, not for the "release" version.  
It should work if you update to the master version.

0x8F is not a valid MIDI value. It'll probably be reduced modulo 0x80 to get it in the valid range.

olheadinternet, 2020-04-30:
updated. What part of the CCSelection can I make unique so I could have multiple settings?

also any note on how I duplicated these is appreciated. 

```cpp

CCSelection<4> cc = {{MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}};

ManyButtonsSelector<4> selector = {
  cc,                       // what to select
  {2, 3, 4, 5}, // button pins
};

CCSelection<4> cc2 = {{MIDI_CC::General_Purpose_Controller_2, CHANNEL_1}};

ManyButtonsSelector<4> selector2 = {
  cc2,                       // what to select
  {6, 7, 8, 9}, // button pins
};

```

tttapa, 2020-04-30:
I'm not sure what you mean. Does the code you posted not work?

olheadinternet, 2020-04-30:
so updated to master release version, and it work.
sorry I was unclear about my follow up question. 

What part of the CCSelection/Setting/Selectable can I make unique so I could have multiple settings for different ManyButtonSelectors.

referring to this (I haven't changed it so I didn't see the need to insert it):
```cpp
template <setting_t N>
class CCSelection : public Selectable<N> {
  public:
    CCSelection(MIDIAddress address) : address{address} {}

    void select(setting_t setting) override {
      setting = Selectable<N>::validateSetting(setting);
      Control_Surface.sendCC(address, setting * 128 / N + 64 / N - 1);
    }

  private:
    MIDIAddress address;
};
```

the duplicated ManyButtonsSelector works just fine... just thought since I was ask one question thought id see if you had a note on how I did that (as I did something there), and if the way I went about it might be why I'm having trouble figuring out how to have a second set of settings.

tttapa, 2020-05-01:
I think your approach is the right one, a separate CCSelection for each setting, and a separate Selector with different pins for each selection.

You could create arrays of CCSelections and Selectors if you wanted to, this reduces some of the boilerplate, but for a small number of settings, your code is fine as well.