tttapa/Control-Surface #186: ESP32 issue

Sixonlon, 2020-04-27:
I tried your examples on a DOIT ESP32 Devkit V1 but it seems that it is not supported. I don't even achieve to compile the example. I tried the OLED-display ones but also the ones in the folder esp32 but none of them compile. Is it due to your current changes and updates or it comes from else where ?

The error is simply : compiling error for the board DOIT ESP32 DevKit V1.

Thanks in advence !

Simon

tttapa, 2020-04-27:
Could you post the full error message? The [CI tests](https://travis-ci.org/github/tttapa/Control-Surface) seem to compile just fine, so I assume there's either a problem with the specific version of the ESP32 Arduino Core you're using, or with the Core configuration of your specific board (the CI tests use the SparkFun ESP32 Thing, IIRC).

I'll see if I can reproduce this problem later today.

Sixonlon, 2020-04-27:
Here is the full message error : 

In file included from C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:46:0,

                 from C:\Users\Portable\Documents\Arduino\libraries\Control-Surface-master\src\MIDI_Outputs\ManyAddresses\CCRotaryEncoder.cpp:1:

C:\Users\Portable\Documents\Arduino\libraries\Encoder/utility/interrupt_pins.h:202:2: error: #error "Interrupts are unknown for this board, please add to this code"

 #error "Interrupts are unknown for this board, please add to this code"

  ^

C:\Users\Portable\Documents\Arduino\libraries\Encoder/utility/interrupt_pins.h:205:2: error: #error "Encoder requires interrupt pins, but this board does not have any :("

 #error "Encoder requires interrupt pins, but this board does not have any :("

  ^

C:\Users\Portable\Documents\Arduino\libraries\Encoder/utility/interrupt_pins.h:206:2: error: #error "You could try defining ENCODER_DO_NOT_USE_INTERRUPTS as a kludge."

 #error "You could try defining ENCODER_DO_NOT_USE_INTERRUPTS as a kludge."

  ^

In file included from C:\Users\Portable\Documents\Arduino\libraries\Control-Surface-master\src\MIDI_Outputs\ManyAddresses\CCRotaryEncoder.cpp:1:0:

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:61:11: error: 'IO_REG_TYPE' does not name a type

  volatile IO_REG_TYPE * pin1_register;

           ^

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:62:11: error: 'IO_REG_TYPE' does not name a type

  volatile IO_REG_TYPE * pin2_register;

           ^

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:63:2: error: 'IO_REG_TYPE' does not name a type

  IO_REG_TYPE            pin1_bitmask;

  ^

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:64:2: error: 'IO_REG_TYPE' does not name a type

  IO_REG_TYPE            pin2_bitmask;

  ^

In file included from C:\Users\Portable\Documents\Arduino\libraries\Control-Surface-master\src\MIDI_Outputs\ManyAddresses\CCRotaryEncoder.cpp:1:0:

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:45:30: error: 'CORE_NUM_INTERRUPT' was not declared in this scope

 #define ENCODER_ARGLIST_SIZE CORE_NUM_INTERRUPT

                              ^

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:135:50: note: in expansion of macro 'ENCODER_ARGLIST_SIZE'

  static Encoder_internal_state_t * interruptArgs[ENCODER_ARGLIST_SIZE];

                                                  ^

In file included from C:\Users\Portable\Documents\Arduino\libraries\Control-Surface-master\src\MIDI_Outputs\ManyAddresses\CCRotaryEncoder.cpp:1:0:

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h: In constructor 'Encoder::Encoder(uint8_t, uint8_t)':

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:82:11: error: 'struct Encoder_internal_state_t' has no member named 'pin1_register'

[... 76 lines cut ...]
                                        ^

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:274:60: error: 'struct Encoder_internal_state_t' has no member named 'pin1_bitmask'

   uint8_t p1val = DIRECT_PIN_READ(arg->pin1_register, arg->pin1_bitmask);

                                                            ^

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:274:72: error: 'DIRECT_PIN_READ' was not declared in this scope

   uint8_t p1val = DIRECT_PIN_READ(arg->pin1_register, arg->pin1_bitmask);

                                                                        ^

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:275:40: error: 'struct Encoder_internal_state_t' has no member named 'pin2_register'

   uint8_t p2val = DIRECT_PIN_READ(arg->pin2_register, arg->pin2_bitmask);

                                        ^

C:\Users\Portable\Documents\Arduino\libraries\Encoder/Encoder.h:275:60: error: 'struct Encoder_internal_state_t' has no member named 'pin2_bitmask'

   uint8_t p2val = DIRECT_PIN_READ(arg->pin2_register, arg->pin2_bitmask);

                                                            ^

tttapa, 2020-04-27:
You seem to be using an outdated version of the PJRC Encoder library that doesn't support the ESP32 yet.

Please try upgrading to the latest version from GitHub: https://github.com/PaulStoffregen/Encoder

Sixonlon, 2020-04-27:
Indeed, it works now.

Thanks a lot !

closed 2020-04-27 by tttapa