tttapa/Control-Surface #188: issue with 4067 mux

olheadinternet, 2020-04-28:
so this works as planned until I get to mux1.pin(8) 
pin(0) ends up controlling both bank_select 0x00 & General_Purpose_Controller_1 0x10
thanks again for your help

```cpp

// #include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

CD74HC4067 mux1 = {
  A0,       // Analog input pin
  {0, 1, 2, 3} // Address pins S0, S1, S2, S3
};
CD74HC4067 mux2 = {
  6,       // Analog input pin
  {0, 1, 2, 3} // Address pins S0, S1, S2, S3
};
 
using namespace MIDI_CC;
// Instantiate a CCPotentiometer object
CCPotentiometer potentiometers[] = {
  {mux1.pin(0),        // Analog pin connected to potentiometer 1
   Bank_Select},     // Controller number of the first potentiometer
  {mux1.pin(1),        // Analog pin connected to potentiometer 2
   Modulation_Wheel},     // Controller number of the second potentiometer
     {mux1.pin(2),        // Analog pin connected to potentiometer 2
   Breath_Controller},     // Controller number of the second potentiometer
     {mux1.pin(3),        // Analog pin connected to potentiometer 2
   Foot_Controller},     // Controller number of the second potentiometer
     {mux1.pin(4),        // Analog pin connected to potentiometer 2
   Portamento_Time},     // Controller number of the second potentiometer
     {mux1.pin(5),        // Analog pin connected to potentiometer 2
   0x06},     // Controller number of the second potentiometer
     {mux1.pin(6),        // Analog pin connected to potentiometer 2
   0x07},     // Controller number of the second potentiometer
     {mux1.pin(7),        // Analog pin connected to potentiometer 2
   0x08},     // Controller number of the second potentiometer
   

>   {mux1.pin(8),        // Analog pin connected to potentiometer 2
>    0x10},     // Controller number of the second potentiometer
> };

using namespace MIDI_Notes;
const uint8_t velocity = 0x7F;
NoteButton buttons[] = {
  {mux2.pin(0), {note(0, 4), CHANNEL_1}, velocity},
  {mux2.pin(1), {note(1, 4), CHANNEL_1}, velocity},
  {mux2.pin(2), {note(2, 4), CHANNEL_1}, velocity},
  {mux2.pin(3), {note(3, 4), CHANNEL_1}, velocity},
  {mux2.pin(4), {note(4, 4), CHANNEL_1}, velocity},
  {mux2.pin(5), {note(5, 4), CHANNEL_1}, velocity},
  {mux2.pin(6), {note(6, 4), CHANNEL_1}, velocity},
  {mux2.pin(7), {note(7, 4), CHANNEL_1}, velocity},
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

```

tttapa, 2020-04-28:
That's most likely because of a problem with the wiring of the address pins of the multiplexers.

olheadinternet, 2020-04-28:
yup silly me... I kept looking. at the input pins of the multiplexer like these far away from the other input pins... but I had what I declared as 3 wired to 4. interesting that the 4067 works as a 8 channel with just 3 pins. thanks.

closed 2020-04-28 by olheadinternet