tttapa/Control-Surface #191: Encoders through mux?

santta, 2020-05-01:
Is it possible to connect rotary encoders through 4067 multiplexers?  or is this not possible due to digital and analog not compatibility of  multiplexors?

tttapa, 2020-05-01:
No, this is not possible for two reasons: 
1. Rotary encoders are handled by the PJRC Encoder library, it has no knowledge of the ExtIO abstraction that's used by the Control Surface library to handle multiplexers. 
2. Rotary encoders need interrupts or need to be polled very frequently to work correctly. ExtIO is relatively slow, as its main purpose is transparent IO access, not speed. If you have nothing else going on in your sketch, it might be fast enough, but I cannot guarantee anything.

Multiplexers can be used for both digital and analog signals without any problems, that's not the issue here.

If you need more encoders, there's some experimental work going on on the `share-encoder` branch:

 - [MCP23017-RotaryEncoder-No-Interrupts](https://tttapa.github.io/Control-Surface-doc/share-encoder/Doxygen/d7/dac/MCP23017-RotaryEncoder-No-Interrupts_8ino-example.html)
 - [MCP23017-RotaryEncoder-Interrupts](https://tttapa.github.io/Control-Surface-doc/share-encoder/Doxygen/dd/d56/MCP23017-RotaryEncoder-Interrupts_8ino-example.html)

The MCP23017 works perfectly on Teensy because it allows you to use I²C inside of interrupt service routines. On most other boards, it still resorts to polling, but it can handle all 16 inputs simultaneously, and it only has to poll the actual inputs if the MCP23017's interrupt pin is low, so it's still pretty fast.

When polling, the same limitations do apply of course, everything in the Control Surface library is handled through cooperative multitasking. If there's anything in your sketch that takes more than a couple of milliseconds to complete, polling will fail, no matter what.

santta, 2020-05-01:
Thanks for the answer, this information is useful to me.

closed 2020-05-01 by santta

MikaBeats, 2021-11-29:
Hello, I would like to know if the library works with I2C encoders like these: https://www.tindie.com/products/saimon/i2cencoder-v2-connect-multiple-encoder-on-i2c-bus/
If it doesn't work, what would you advise me to do to control 26 rotary encoders?  

Thank you.

tttapa, 2021-12-05:
The encoder classes are generic, you can plug in any type of encoder that provides the required `read()` method (and possibly a `begin()` method that is called automatically if available):

https://github.com/tttapa/Control-Surface/blob/e6542665da8997fdf33aee5ff1e0b7383c7ebe00/src/MIDI_Outputs/Abstract/MIDIRotaryEncoder.hpp#L22-L23

MikaBeats, 2022-02-04:
Thank you ! but can you help me to do this part of code please ? i'm beginner and i don't know how to use this correctly.. I'm also looking for what the sender class is but I can't build a code that is good, please help me i'm very lost in the documentation.

thank you, have a nice day.