tttapa/Control-Surface #192: clock & data pins?

olheadinternet, 2020-05-02:
i'm not quite sure where I declare the clock and data pins?
is that necessary?

```cpp

// Include the library
#include <Control_Surface.h>

// Instantiate a MIDI Interface to use
USBDebugMIDI_Interface midi;

// Instantiate a shift register as output for the LEDs
SPIShiftRegisterOut<8> sreg = {
  4,       // Latch pin (ST_CP)
  MSBFIRST, // Byte order
};

using namespace MIDI_Notes;

// Create a LED bar driver that listens for MIDI Note C4 that drives
// the LEDs connected to the eight output pins of the shift register
NoteLEDBar<8> leds = { sreg.pins(), note(C, 4) };

// Initialize the Control Surface
void setup() {
  Control_Surface.begin();
}

// Update the Control Surface
void loop() {
  Control_Surface.loop();
}

```

tttapa, 2020-05-02:
It uses the clock and data pins of the SPI interface.

olheadinternet, 2020-05-02:
so I do st_cp (rclk) on the 74h... to sclk on the Arduino
and data to miso?

tttapa, 2020-05-02:
No, to the data output: MOSI.

See https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d5f/1_8SPI-Blink_8ino-example.html

olheadinternet, 2020-05-02:
now the connections list on the top makes sense.

 so I now realize I should of made clear I'm on a pro micro. 

That blink example runs but with no latch... the RX led also blinks super dimly, which from what I just researched is the SS signal that isn't broken out on the board.

I do get feedback on RX from the midi notes, when running note range leds.

is there a way to manually assign SS?

just to be thorough, I have run this successfully - https://learn.adafruit.com/adafruit-arduino-lesson-4-eight-leds/arduino-code

thank you for dealing with my noobness.

tttapa, 2020-05-02:
The SS pin of the Pro Micro is connected to the RX LED for some reason.

You can specify the latch pin by replacing `SS` by any other output pin of choice on this line:
```cpp
SPIShiftRegisterOut<8> sreg = {SS, MSBFIRST};
```

olheadinternet, 2020-05-03:
ok I figured out my issue... I was looking at a pin out that called o.e. pin 13 'latch' 

[image]

I have now successfully ran Note-LED-Range and noteLEDbar (it seems I had to change USBDebugMIDI_Interface midi to USBMIDI_Interface midi; in the example of the later)

In a different sketch I changed NoteLEDBar to CCLEDBar... it works, but when I hit 127 on the CC value it seems to lock up, RX remains solid (along with the 8 leds) there after. any clue why?

thanks! 

```cpp

// Include the library
#include <Control_Surface.h>

// Instantiate a MIDI Interface to use
USBMIDI_Interface midi;

// Instantiate a shift register as output for the LEDs
SPIShiftRegisterOut<8> sreg = {
  10,       // Latch pin (ST_CP)
  MSBFIRST, // Byte order
};

using namespace MIDI_Notes;

// Create a LED bar driver that listens for MIDI Note C4 that drives
// the LEDs connected to the eight output pins of the shift register

CCLEDBar<8> leds = { sreg.pins(), MIDI_CC::Breath_Controller };

// Initialize the Control Surface
void setup() {
  Control_Surface.begin();
}

// Update the Control Surface
void loop() {
  Control_Surface.loop();
}

```

tttapa, 2020-05-03:
That does seem to be an off-by-one error in the library, thanks for letting me know.

It should be fixed in d73bacfe0afbbc0748226b201150e99e6c031b2a.

olheadinternet, 2020-05-04:
do I re-download the master?

closed 2020-05-04 by olheadinternet