tttapa/Control-Surface #195: return to previous value / ccbuttons

olheadinternet, 2020-05-05:
hi! 
I have a pot and momentary button mapped to the same midi cc... I would like the momentary button to send 127 and then return back to the previous value as set by the pot. is there a way to get the pot value as the release value of the ccbutton or a release value that is equivalent to return to previous?

```cpp

CCPotentiometer potentiometers [] = {  
  {mux1.pin(0),        // Analog pin connected to potentiometer 1
   {MIDI_CC::Bank_Select, CHANNEL_2 }},   
  
  {mux1.pin(1),        // Analog pin connected to potentiometer 2
  {MIDI_CC::Modulation_Wheel, CHANNEL_2 }},
  
  {mux1.pin(2),        // Analog pin connected to potentiometer 
  {MIDI_CC::Breath_Controller, CHANNEL_2}},
  
CCButton button[] =
  {7, // pin
  {MIDI_CC::Breath_Controller, CHANNEL_2}, // MIDI address
  {0x7F, 0x00}}, // value when pressed, value when released
  {8, // pin
  {MIDI_CC::Modulation_Wheel, CHANNEL_2}, // MIDI address
  {0x7F, 0x00}}, // value when pressed, value when released
  {9, // pin
  {MIDI_CC::Bank_Select, CHANNEL_2}, // MIDI address
  {0x7F, 0x00}}, // value when pressed, value when released
};

```

tttapa, 2020-05-06:
I think it would be best to implement this yourself, see [Custom-MIDI-Output-Element.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d3b/Custom-MIDI-Output-Element_8ino-example.html).  
You can use the `Button` class as demonstrated in this example, and the `FilteredAnalog` class for your potentiometer, this is what `CCPotentiometer` etc. use under the hood:

https://github.com/tttapa/Control-Surface/blob/c03f0783545aaa93c481e7d9575de0e543db8ec3/src/MIDI_Outputs/Abstract/MIDIFilteredAnalog.hpp#L36-L39

Also see this example: [FilteredAnalog.ino](https://tttapa.github.io/Arduino-Helpers/Doxygen/d3/dbe/1_8FilteredAnalog_8ino-example.html)

olheadinternet, 2020-05-14:
thanks! will go through the custom midi elements... for this I realized I could just route a button to the pot signal itself and do it electronically outside of code.

closed 2020-05-14 by olheadinternet