tttapa/Control-Surface #197: AppleMIDI example: Questions

ciapamusche, 2020-05-08:
Hi im tryng to get work AppleMIDI example.

i have an error

Arduino:1.8.12 (Windows 10), Scheda:"DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

In file included from C:\Users\Damiano\Documents\Arduino\libraries\Control-Surface-master\examples\3. MIDI Interfaces\AppleMIDI\AppleMIDI.ino:99:0:

C:\Users\Damiano\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Interfaces/Wrappers/FortySevenEffects.hpp:9:18: fatal error: MIDI.h: No such file or directory

compilation terminated.

exit status 1

Any advices?

I have other questions.

1)Is possible to use the AppleMIDI on ESP8266? Is a wifi communication so i think is possible.

2) an off topic but i dont want to open new unuseful issues:  I have a roland EV-5 pedal. How can i connect? thank you very much for help

tttapa, 2020-05-08:
> C:\Users\Damiano\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Interfaces/Wrappers/FortySevenEffects.hpp:9:18: fatal error: MIDI.h: No such file or directory

The [AppleMIDI](https://github.com/lathoub/Arduino-AppleMIDI-Library) library depends on the FortySevenEffects [MIDI](https://github.com/FortySevenEffects/arduino_midi_library) library. You have to install that first.

> 1. Is possible to use the AppleMIDI on ESP8266? Is a wifi communication so i think is possible.

That's a question for the AppleMIDI developers, I've never tried it on an ESP8266, but it seems to work fine on an ESP32, and although there are no examples for the ESP8266 specifically, their GitHub page claims that it's compatible.

> 2. an off topic but i dont want to open new unuseful issues: I have a roland EV-5 pedal. How can i connect? 

A quick Google search brings up this schematic:

[image]

If it's correct, you can connect the shield to ground, the ring to Vcc (3.3V), and the tip to the analog input. 

An ESP8266 doesn't have a 3.3V ADC. Some boards have a voltage divider on-board to account for  that, but it's not ideal for measuring potentiometers.  
If your board doesn't have a voltage divider, connecting the pedal directly to the ESP8266's ADC pin will destroy it.

ciapamusche, 2020-05-08:
Sorry for imprecision I have a NodeMCU which have internal voltage divider and can manage until 3.3v

I will try the AppleMIDI on it

ciapamusche, 2020-05-16:
Update. The library didnt work with ESP8266. in contact with developer he adapt a branch for solve the problem. 
https://github.com/lathoub/Arduino-AppleMIDI-Library/issues/85

So if you want use NodeMCU or ESP8266 you have to download the branch feat/2.1.0

ciapamusche, 2020-05-16:
Pasting the modified code to get ESP8266 Work. Changing wifi libraries

`/**
 * This example demonstrates how to use the AppleMIDI library to use Control
 * Surface over the network.
 * 
 * This example requires the 
 * [AppleMIDI](https://github.com/lathoub/Arduino-AppleMIDI-Library) and 
 * [MIDI](https://github.com/FortySevenEffects/arduino_midi_library) libraries.
 * 
 * @boards  ESP32
 * 
 * Connections
 * -----------
 * 
 * The on-board LED will be used, as well as the push button on GPIO0. These 
 * should be present on most ESP32 boards, if this is not the case for your 
 * board, connect an LED (+ series resistor) and a push button to the relevant
 * pins (in the "MIDI Elements" section below). For more details, see the 
 * @ref NoteButton.ino and @ref 1.Note-LED.ino examples.
 * 
 * WiFi Credentials
 * ----------------
 * 
 * Open the tab `WiFi-Credentials.example.h`, enter your WiFi credentials, and
 * rename the file to `WiFi-Credentials.h`.
 * 
 * Behavior
 * --------
 * 
 * Upload the code to the ESP32, and open the Serial monitor. You should see
 * output like this
 * 
 * ~~~
 * Connecting to Your WiFi Network ...
 * Connected!
 * IP address: 192.168.1.35
 * mDNS responder started (ESP32.local)
 * ~~~
 * 
 * Next, connect to the device using your DAW or other MIDI software. If the 
 * software supports mDNS (Apple Bonjour), you can use `ESP32.local`, 
 * otherwise, you'll have to use the IP address.
 * 
 * When the connection is successful, you'll see the following message in the
 * Serial monitor:
 * 
 * ~~~
 * Connected to session Your Session Name
 * ~~~
 * 
 * When the button is pushed, a MIDI note on message for note C4 (middle C) is
 * sent.  
 * When the ESP32 receives a MIDI note message for that note, it turn on/off the
 * LED accordingly.
 * 
 * RTP MIDI Bridge (Linux)
 * -----------------------
 * 
 * If you're on Linux, you can use the rtpmidi-bridge application in the example
 * folder. You'll need to install Node.js and NPM.
 * 
 * First, install the necessary dependencies and build tools:
 * 
 * ~~~sh
 * sudo apt install build-essential libasound2-dev libavahi-compat-libdnssd-dev
 * ~~~
 * 
 * Then install the dependencies using NPM:
 * 
[... 101 lines cut ...]
         << F("Connected!") << endl
         << F("IP address: ") << WiFi.localIP() << endl;

  // Set up mDNS responder:
  if (!MDNS.begin(AppleMIDI.getName()))
    FATAL_ERROR(F("Error setting up MDNS responder!"), 0x0032);
  Serial << F("mDNS responder started (") << AppleMIDI.getName() << ".local)" 
         << endl;
  MDNS.addService("apple-midi", "udp", AppleMIDI.getPort());

  // Set up some AppleMIDI callback handles
  AppleMIDI.setHandleConnected(onAppleMidiConnected);
  AppleMIDI.setHandleDisconnected(onAppleMidiDisconnected);
  AppleMIDI.setHandleError(onAppleMidiError);

  // Initialize Control Surface (also calls MIDI.begin())
  Control_Surface.begin();
}

// ---------------------------------- Loop ---------------------------------- //

void loop() {
  // Update all MIDI elements and handle incoming MIDI
  Control_Surface.loop();
}`

ciapamusche, 2020-05-16:
Pasting working code for ESP8266 Boards.  Modified the Wifi libraries.

/**
 * This example demonstrates how to use the AppleMIDI library to use Control
 * Surface over the network.
 * 
 * This example requires the 
 * [AppleMIDI](https://github.com/lathoub/Arduino-AppleMIDI-Library) and 
 * [MIDI](https://github.com/FortySevenEffects/arduino_midi_library) libraries.
 * 
 * @boards  ESP8266
 * 
 * Connections
 * -----------
 * 
 * The on-board LED will be used, as well as the push button on GPIO0. These 
 * should be present on most ESP8266 boards, if this is not the case for your 
 * board, connect an LED (+ series resistor) and a push button to the relevant
 * pins (in the "MIDI Elements" section below). For more details, see the 
 * @ref NoteButton.ino and @ref 1.Note-LED.ino examples.
 * 
 * WiFi Credentials
 * ----------------
 * 
 * Open the tab `WiFi-Credentials.example.h`, enter your WiFi credentials, and
 * rename the file to `WiFi-Credentials.h`.
 * 
 * Behavior
 * --------
 * 
 * Upload the code to the ESP8266, and open the Serial monitor. You should see
 * output like this
 * 
 * ~~~
 * Connecting to Your WiFi Network ...
 * Connected!
 * IP address: 192.168.1.35
 * mDNS responder started (ESP8266.local)
 * ~~~
 * 
 * Next, connect to the device using your DAW or other MIDI software. If the 
 * software supports mDNS (Apple Bonjour), you can use `ESP32.local`, 
 * otherwise, you'll have to use the IP address.
 * 
 * When the connection is successful, you'll see the following message in the
 * Serial monitor:
 * 
 * ~~~
 * Connected to session Your Session Name
 * ~~~
 * 
 * When the button is pushed, a MIDI note on message for note C4 (middle C) is
 * sent.  
 * When the ESP32 receives a MIDI note message for that note, it turn on/off the
 * LED accordingly.
 * 
 * RTP MIDI Bridge (Linux)
 * -----------------------
 * 
 * If you're on Linux, you can use the rtpmidi-bridge application in the example
 * folder. You'll need to install Node.js and NPM.
 * 
 * First, install the necessary dependencies and build tools:
 * 
 * ~~~sh
 * sudo apt install build-essential libasound2-dev libavahi-compat-libdnssd-dev
 * ~~~
 * 
 * Then install the dependencies using NPM:
 * 
[... 101 lines cut ...]
         << F("Connected!") << endl
         << F("IP address: ") << WiFi.localIP() << endl;

  // Set up mDNS responder:
  if (!MDNS.begin(AppleMIDI.getName()))
    FATAL_ERROR(F("Error setting up MDNS responder!"), 0x0032);
  Serial << F("mDNS responder started (") << AppleMIDI.getName() << ".local)" 
         << endl;
  MDNS.addService("apple-midi", "udp", AppleMIDI.getPort());

  // Set up some AppleMIDI callback handles
  AppleMIDI.setHandleConnected(onAppleMidiConnected);
  AppleMIDI.setHandleDisconnected(onAppleMidiDisconnected);
  AppleMIDI.setHandleError(onAppleMidiError);

  // Initialize Control Surface (also calls MIDI.begin())
  Control_Surface.begin();
}

// ---------------------------------- Loop ---------------------------------- //

void loop() {
  // Update all MIDI elements and handle incoming MIDI
  Control_Surface.loop();
}

alomdj00, 2021-01-26:
Hola @ciapamusche  he estado  tratando de usar el esp8266, y no he logrado compilar, instale el ide nuevamente, descargue todo lo que se me ocurrio y no compilo, veo que nombras una rama FEAT, la he buscado en el repositorio de applemidi, de fortyseveneffects y de controlsurface y no la encuentro, podrias indicarme a que libreria pertenece esta rama o como lograste compilar para esp8266, agradezco mucho tu ayuda. Un saludo

ciapamusche, 2021-01-27:
Hi Which error you get?

have you try the branch indicated?

alomdj00, 2021-01-27:
Hola, este es el error que obtengo 

https://github.com/tttapa/Control-Surface/issues/372

He usado la version master y la new input de controlsurface.
AppleMidi  ya no tiene ramas.
Fortyseveneffects tampoco tiene ramas.

No se cual es la rama feat, o a que libreria pertenece la rama feat  para buscarla  y probar