tttapa/Control-Surface #198: Questions about components

wingedrhino, 2020-05-15:
Hi,

I tried building a controller out of what pots I had at home, and these were all sub-optimal (22k, 47k, 100k, and 470k pots). I have erratic readings each time I touch my MIDI device, and the flood of messages slows things down considerably. I'd now like to go out and source more _appropriate_ components, for a fast and jitter-free operation. Can you provide some general suggestions on the sort of components I should be buying? Here's a few questions I have:

1. What's an ideal resistance for a CCPotentiometer? I've heard that 10k or lower is better. Is 5k a good balance?

2. Does the library support mapping a linear potentiometer into a non-linear potentiometer, and possibly select the mapping curves? If not, would you recommend linear or audio pots for the knobs?

3. Would an ARM based microcontroller result in a faster operation, assuming that all connections are direct (i.e, no external muxes used)?

4. Would you recommend an ARM microcontroller if I WAS using a couple of external muxes for the analog inputs?

5. I'm considering 24 potentiometers on one of my builds (to control all drawbars of setBfree, and also the reverb & distortion level). At how many analog inputs does Control Surface start to noticeably degrade in performance? Ideally I want sub-30ms response times, preferably sub-10ms.

tttapa, 2020-05-15:
1. The maximum recommended output impedance of devices you can connect to the ADC of most Arduinos is 10kΩ. The worst-case impedance of a 22kΩ potentiometer is 11kΩ. Personally, I use 5kΩ potentiometers.
2. Yes, see [CCPotentiometer-Map.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d26/CCPotentiometer-Map_8ino-example.html). Do keep in mind that this will amplify the noise.
3. ARM only refers to the CPU core, it doesn't tell you anything about the speed of the ADC and other IO. In general, however, most ARM-based Arduinos will be significantly faster than their AVR counterparts, both in CPU performance and IO speed. There's a very wide range of ARM boards, to know the actual numbers, you have to look it up in the appropriate datasheet.
4. It depends. I built a controller using an Arduino Leonardo with 12 potentiometers, and it works just fine, without any noise or erratic readings and it's plenty fast. An ARM board like a Teensy will be much faster, but that extra speed is "wasted" if you don't need it.
5. It depends on all kinds of factors, both in hardware and in software, so I cannot tell you that.

To improve noise filtering, you can do a couple of things: 
- change the [filter datatype](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d69/namespaceAH.html#ae42145751d1331f921750738659d5d6b) to `uint32_t` to increase the filter precision.
- increase the [filter factor](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d69/namespaceAH.html#a74cef1adb51d95add8b2bea344b6c755) from 2 to 3 or even 4 to change the amount of filtering. This also increases lag, but this isn't an issue as long the sampling rate is high enough.

If that doesn't help, you'll probably have to look at your hardware. Are you using a breadboard or did you solder the wires?

wingedrhino, 2020-05-15:
1. Thanks! I'll buy a few 5k pots and if they're good I'll get a few dozens more of the same stuff.

2. Makes sense. I should probably just buy an even mix of linear and logarithmic pots, so I can switch things around a little!

3, 4 & 5: Yeah I understand my questions came off as rather vague! From what I can see now after some reading, for any setup involving inputs polled in a loop, the main limiting factors would be 1) the clockspeed, 2) the speed of the analogRead() operation (which seems to be atleast 2x faster on the Due vs the Uno), and 3) the number of inputs. But I should be able to surpass these limitations via other means for inputs that need to be faster, perhaps via slave controllers that send out interrupts when _they_ detect a signal change.

About those techniques you mentioned to increase filter precision - can I set them for each knob separately?

The way I did up my current hardware was by directly melting holes and sticking pots out of an ice cream tub! I soldered the wires onto the pots and switches, but they were connected together (in a loose bundle) by knotting the wires, and then to the Leonardo via DuPont connectors.

I also had some trimpots to test with, that I mounted on a breadboard (this version was completely solderless). I noticed that the loose connections between the pots and the breadboard, and in a couple of cases, within the pot's screw itself, caused a lot of jitters.

All of this was from a 10-year old electronics kit that was probably not very well handled by a younger me. But it is what I had at home. For my next build I'm going to source fresh components and mount everything directly on a perfboard, with adequate soldering!

closed 2020-05-21 by wingedrhino