tttapa/Control-Surface #199: LED map to potentiometer

DJMathijs16, 2020-05-19:
I want to map the position of my softpot to the ledbar, so where I'll put my finger, the led (1 LED only) has to go on at that position.
When I touch my softpot now, it doesn't matter at what position, only the first upper led goes on. 
[image]

Here's my code
```
#include <FastLED.h>
#include <Encoder.h>
#include <Control_Surface.h>
USBMIDI_Interface midi;
int touchsliderPin = A1;

CCPotentiometer touchslider = {
  A1,                                   // Analog pin connected to potentiometer
  {0x08, CHANNEL_1}, // Control Change 8 on Channel 1
};

// Define the array of leds.
Array<CRGB, 8> leds = {};

// The data pin with the strip connected.
constexpr uint8_t ledpin = 26;

CCRangeFastLED<leds.length> TouchStripLedBar = {leds, 0x08};

void setup() {
  digitalWrite(touchsliderPin, HIGH); //enable pullup resistor

  FastLED.addLeds<NEOPIXEL, ledpin>(leds.data, leds.length);
  FastLED.setCorrection(TypicalPixelString);
  TouchStripLedBar.setBrightness(128);

  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
  FastLED.show();
}
```

tttapa, 2020-05-19:
`CCRangeFastLED` is used to listen to a range of _addresses_, not ranges of values.

You could try something like this:

```cpp
#include <FastLED.h>
#include <Control_Surface.h>

// Callback that lights a single LED in a LED strip based on the
// received CC value.
class CCFastLEDBarCallback : public SimpleNoteCCValueCallback {
  public:
    // Constructor
    CCFastLEDBarCallback(CRGB *leds, uint8_t numLeds, 
                         CRGB (*colormapper)(uint8_t, uint8_t))
        : leds(leds), numLeds(numLeds), colormapper(colormapper) {}
       
    // Called once upon initialization.
    void begin(const INoteCCValue &input) override { updateAll(input); }

    // Called each time a MIDI message is received and an LED has to be updated.
    // @param   input
    //          The NoteCCRange or NoteCCValue object this callback belongs to.
    //          This is the object that actually receives and stores the MIDI
    //          values.
    // @param   index
    //          The index of the value that changed. (zero-based)
    void update(const INoteCCValue &input, uint8_t index) override {
        // Get the MIDI value that changed [0, 127]
        uint8_t value = input.getValue(index);
        uint8_t ledIndex = value * numLeds / 128;
        // Apply the color mapper to convert the value to a color
        CRGB newColor = CRGB(colormapper(value, index));
        // Update the LED colors
        for (uint8_t i = 0; i < ledIndex; ++i)
          leds[i] = CRGB::Black;
        leds[ledIndex] = newColor;
        for (uint8_t i = ledIndex + 1; i < numLeds; ++i)
          leds[i] = CRGB::Black;
        dirty = true;
    }

    // Returns true if the dirty flag is set and the LED strip has to
    // be updated. Resets the flag when called.
    bool getDirty() {
      bool d = dirty;
      dirty = false;
      return d;
    }

  private:
    CRGB *leds;
    uint8_t numLeds;
    CRGB (*colormapper)(uint8_t, uint8_t);
    bool dirty = true;
};

// Simple adapter class
class CCFastLEDBar : public GenericNoteCCRange<MIDIInputElementCC, 1,
                                               CCFastLEDBarCallback> {
  public:
    CCFastLEDBar (CRGB *leds, uint8_t numLeds, MIDIAddress address, 
                  CRGB (*colormapper)(uint8_t, uint8_t))
      : GenericNoteCCRange<MIDIInputElementCC, 1, CCFastLEDBarCallback>{
            address,
            {leds, numLeds, colormapper},
    } {}

    bool getDirty() { return callback.getDirty(); }
};

// -------------------------------------------------------------------------- //

// Map a CC value and an LED index to a color value.
CRGB colormapper(uint8_t value, uint8_t index) {
  uint8_t hue = increaseBitDepth<8, 7, uint8_t>(value); // map 0-127 → 0-255
  return CHSV(hue, 255, 255);
}

// The data pin with the strip connected.
constexpr uint8_t ledpin = 26;
// FastLED buffer.
Array<CRGB, 16> leds = {};
// LED bar that listens for CC data
CCFastLEDBar ccleds = { leds.data, leds.length, 0x08, colormapper };

// The pin with the potentiometer connected
constexpr pin_t potpin = A1;
// Potentiometer that sends CC data
CCPotentiometer ccpot = { potpin, 0x08 };

void setup() {
  FastLED.addLeds<NEOPIXEL, ledpin>(leds.data, leds.length);
  FastLED.setCorrection(TypicalPixelString);
  FastLED.setBrightness(128);

  Control_Surface.begin(); // Initialize Control Surface
  pinMode(potpin, INPUT_PULLUP);
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
  if (ccleds.getDirty())
    FastLED.show();
}
```

It's (very) roughly based on this example: https://tttapa.github.io/Control-Surface-doc/Doxygen/dd/da5/Custom-Note-LED-Input-Element-Callback_8ino-example.html

DJMathijs16, 2020-05-20:
Thank you verry much. I'll try this.

Malensky, 2020-07-22:
Hi, I am trying to grasp this example and adapt it for my purposes. In my case I need each led in the neopixel strip to respond to a different CC number with the colors corresponding to the changing values as in your example... if it makes sense. Not sure how to achieve that so any help would be much appreciated. 
Thank you so much for your work!

tttapa, 2020-07-23:
I think the [`CCRangeFastLED`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/dee/classCCRangeFastLED.html) class does what you want.

Malensky, 2020-07-27:
Hi, thanks so much, that almost solved it for me, after a week of efforts, I am almost there. Usng your example and changing the RainbowColorMapper to DefaultColorMapper I made the Neopixels change colors, each according to its corresponding potentiometer. This however seems to be happening at a much quicker rate, while I need them to go through all the "Rainbow" within one full turn of the knob... if it makes sense. Could you tell me where to look? Here is my code, thanks!

 
#include <FastLED.h>
// Must be before Control Surface to enable FastLED features of Control Surface
#include <Control_Surface.h>
USBMIDI_Interface midiusb;

MIDI_PipeFactory<1> pipes;
 
// Define the array of leds.
Array<CRGB, 11> leds = {};
// The data pin with the strip connected.
constexpr uint8_t ledpin = 4;

// Custom callback to handle output for a selector.
class MySelectorCallback {
  public:
    // Constructor
    MySelectorCallback(pin_t redLED, pin_t greenLED, pin_t blueLED)
      : redLED(redLED), greenLED(greenLED), blueLED(blueLED) {}

    // Begin function is called once by Control Surface.
    // Use it to initialize everything.
    void begin() {
      pinMode(redLED, OUTPUT);
      pinMode(greenLED, OUTPUT);
      pinMode(blueLED, OUTPUT);
      show(0);
    }

    // Update function is called continuously by Control Surface.
    // Use it to implement things like fading, blinking ...
    void update() {}

    // Update function with arguments is called when the setting
    // changes.
    // Use it to update the LEDs.
    void update(setting_t oldSetting, setting_t newSetting) {
      (void) oldSetting; // unused in this example
      show(newSetting);
    }

  private:
    // Show the color of the given setting.
    void show(setting_t setting) {
      uint8_t color = getColor(setting);
      digitalWrite(redLED, color & 0b001 ? HIGH : LOW);
      digitalWrite(greenLED, color & 0b010 ? HIGH : LOW);
      digitalWrite(blueLED, color & 0b100 ? HIGH : LOW);
    }

    // Convert the given setting to a 3-bit RGB color value.
    static uint8_t getColor(setting_t setting) {
      switch (setting) {
        case 0: return 0b111;
        case 1: return 0b011;
        case 2: return 0b100;
        case 3: return 0b001;
        case 4: return 0b101;
        case 5: return 0b010;
        default: return 0b000;
      }
    }

  private:
    // Member variables to remember the pin numbers of the LEDs.
    pin_t redLED, greenLED, blueLED;
};

 
USBMIDI_Interface midi;

// Create 6 banks with one track per bank.
Bank<6> bank = 1;

// Create a selector that uses our custom callback, to control the bank.
GenericIncrementDecrementSelector<6, MySelectorCallback> selector = {
  bank,         // bank to manage
  {9, 7, 8}, // red, green, blue LED pins 
                // (this is the MySelectorCallback constructor defined above)
  {2, 3},       // incr/decr button pins
  Wrap::Wrap,   // wrap around when reaching setting 6
};

// Instantiate an analog multiplexer
CD74HC4067 mux = {
  A0,       // Analog input pin
  {10, 14, 15, 16} // Address pins S0, S1, S2
};

// Create an array of potentiometers that send out
// MIDI Control Change messages when you turn the
// potentiometers connected to the eight input pins of
// the multiplexer
Bankable::CCPotentiometer volumePotentiometers[] = {
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(0), {MIDI_CC::Sound_Controller_1, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(1), {MIDI_CC::Sound_Controller_2, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(2), {MIDI_CC::Sound_Controller_3, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(3), {MIDI_CC::Sound_Controller_4, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(4), {MIDI_CC::Sound_Controller_5, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(5), {MIDI_CC::Sound_Controller_6, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(6), {MIDI_CC::Sound_Controller_7, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(7), {MIDI_CC::Sound_Controller_8, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(8), {MIDI_CC::Sound_Controller_9, CHANNEL_1}},  
};

Bankable::CCButtonLatched<6> b2 = {
  {bank, BankType::CHANGE_CHANNEL},     // bank configuration
  mux.pin(9),                                   // digital pin
  {MIDI_CC::Sound_Controller_10, CHANNEL_1},                  // address
};

Bankable::CCButtonLatched<6> b3 = {
  {bank, BankType::CHANGE_CHANNEL},     // bank configuration
  mux.pin(10),                                   // digital pin
  {MIDI_CC::General_Purpose_Controller_5, CHANNEL_1},                  // address
};

 
// Create a functor that maps the velocity and the index of a note to a color.
struct RainbowColorMapper {
  CHSV operator()(uint8_t value, uint8_t index) const {
    return CHSV(255 * index / leds.length, 255, 2 * value);
  }
};
 
using namespace MIDI_CC;
CCRangeFastLED<leds.length, DefaultColorMapper> midiled = {leds, Sound_Controller_1};

 
void setup() {

  //Control_Surface >> pipes >> Control_Surface;
  Control_Surface >> pipes >> Control_Surface;
  //Control_Surface << pipes << midiusb; 
  
  // See FastLED examples and documentation for more information.
  FastLED.addLeds<NEOPIXEL, ledpin>(leds.data, leds.length);
  FastLED.setCorrection(TypicalPixelString);
  midiled.setBrightness(20);
  Control_Surface.begin();
}
 
void loop() {
  Control_Surface.loop();
  FastLED.show();
}

tttapa, 2020-07-27:
The default color mapper is just the default color palette used by the Launchpad MIDI controllers, it's not meant to go around the color wheel.

If you want to make the color depend on the value of the MIDI messages, you could use something like this:
```cpp
// Create a functor that maps the velocity and the index of a note to a color.
struct RainbowColorMapper {
 CHSV operator()(uint8_t value, uint8_t index) const {
   (void) index;
   return CHSV(increaseBitDepth<8, 7, uint8_t>(value), 255, 255);
 }
};

using namespace MIDI_CC;
CCRangeFastLED<leds.length, RainbowColorMapper> midiled = {leds, Sound_Controller_1};
```
You'll have to understand the HSV color space and the `CHSV` type of the FastLED library: https://github.com/FastLED/FastLED/wiki/Pixel-reference#setting-hsv-colors-

Malensky, 2020-07-28:
Thank you, thank you so much! It works! The only thing I can't figure out is how to make the leds bankable, that is to respond to a different midi channel each time I change banks. The code now is:

 
#include <FastLED.h>
// Must be before Control Surface to enable FastLED features of Control Surface
#include <Control_Surface.h>
USBMIDI_Interface midiusb;

MIDI_PipeFactory<7> pipes;
 
// Define the array of leds.
Array<CRGB, 11> leds = {};
// The data pin with the strip connected.
constexpr uint8_t ledpin = 4;

// Custom callback to handle output for a selector.
class MySelectorCallback {
  public:
    // Constructor
    MySelectorCallback(pin_t redLED, pin_t greenLED, pin_t blueLED)
      : redLED(redLED), greenLED(greenLED), blueLED(blueLED) {}

    // Begin function is called once by Control Surface.
    // Use it to initialize everything.
    void begin() {
      pinMode(redLED, OUTPUT);
      pinMode(greenLED, OUTPUT);
      pinMode(blueLED, OUTPUT);
      show(0);
    }

    // Update function is called continuously by Control Surface.
    // Use it to implement things like fading, blinking ...
    void update() {}

    // Update function with arguments is called when the setting
    // changes.
    // Use it to update the LEDs.
    void update(setting_t oldSetting, setting_t newSetting) {
      (void) oldSetting; // unused in this example
      show(newSetting);
    }

  private:
    // Show the color of the given setting.
    void show(setting_t setting) {
      uint8_t color = getColor(setting);
      digitalWrite(redLED, color & 0b001 ? HIGH : LOW);
      digitalWrite(greenLED, color & 0b010 ? HIGH : LOW);
      digitalWrite(blueLED, color & 0b100 ? HIGH : LOW);
    }

    // Convert the given setting to a 3-bit RGB color value.
    static uint8_t getColor(setting_t setting) {
      switch (setting) {
        case 0: return 0b111;
        case 1: return 0b011;
        case 2: return 0b100;
        case 3: return 0b001;
        case 4: return 0b101;
        case 5: return 0b010;
        default: return 0b000;
      }
    }

  private:
    // Member variables to remember the pin numbers of the LEDs.
    pin_t redLED, greenLED, blueLED;
};

 
USBMIDI_Interface midi;

// Create 6 banks with one track per bank.
Bank<6> bank = 1;

// Create a selector that uses our custom callback, to control the bank.
GenericIncrementDecrementSelector<6, MySelectorCallback> selector = {
  bank,         // bank to manage
  {9, 7, 8}, // red, green, blue LED pins 
                // (this is the MySelectorCallback constructor defined above)
  {2, 3},       // incr/decr button pins
  Wrap::Wrap,   // wrap around when reaching setting 6
};

// Instantiate an analog multiplexer
CD74HC4067 mux = {
  A0,       // Analog input pin
  {10, 14, 15, 16} // Address pins S0, S1, S2
};

// Create an array of potentiometers that send out
// MIDI Control Change messages when you turn the
// potentiometers connected to the eight input pins of
// the multiplexer
Bankable::CCPotentiometer volumePotentiometers[] = {
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(0), {MIDI_CC::Sound_Controller_1, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(1), {MIDI_CC::Sound_Controller_2, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(2), {MIDI_CC::Sound_Controller_3, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(3), {MIDI_CC::Sound_Controller_4, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(4), {MIDI_CC::Sound_Controller_5, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(5), {MIDI_CC::Sound_Controller_6, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(6), {MIDI_CC::Sound_Controller_7, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(7), {MIDI_CC::Sound_Controller_8, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, mux.pin(8), {MIDI_CC::Sound_Controller_9, CHANNEL_1}},  
};

Bankable::CCButtonLatched<6> b2 = {
  {bank, BankType::CHANGE_CHANNEL},     // bank configuration
  mux.pin(9),                                   // digital pin
  {MIDI_CC::Sound_Controller_10, CHANNEL_1},                  // address
};

Bankable::CCButtonLatched<6> b3 = {
  {bank, BankType::CHANGE_CHANNEL},     // bank configuration
  mux.pin(10),                                   // digital pin
  {MIDI_CC::General_Purpose_Controller_5, CHANNEL_1},                  // address
};

 
// Create a functor that maps the velocity and the index of a note to a color.
struct RainbowColorMapper {
 CHSV operator()(uint8_t value, uint8_t index) const {
   (void) index;
   return CHSV(increaseBitDepth<8, 7, uint8_t>(value), 255, 120);
 }
};
 
using namespace MIDI_CC;
CCRangeFastLED<leds.length, RainbowColorMapper> midiled = {leds, Sound_Controller_1};
 
void setup() {

//Control_Surface >> pipes >> Control_Surface;
  Control_Surface >> pipes >> midiusb;
  Control_Surface << pipes << midiusb; 
  
  // See FastLED examples and documentation for more information.
  FastLED.addLeds<NEOPIXEL, ledpin>(leds.data, leds.length);
  FastLED.setCorrection(TypicalPixelString);
  midiled.setBrightness(80);
  Control_Surface.begin();
}
 
void loop() {
  Control_Surface.loop();
  FastLED.show();
}

tttapa, 2020-07-28:
You can use the Bankable version of CCRangeFastLED: https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/dc3/classBankable_1_1CCRangeFastLED.html