tttapa/Control-Surface #200: Can I mix different button types in banks?

ermyril, 2020-05-22:
So I'm playing with banks using [MIDI_controller-97](https://tttapa.github.io/Control-Surface-doc/Doxygen/de/d26/MIDI_controller-97_8ino-example.html) as an example, and I'm interested in possibility of mixing instances of CCButton and CCButtonLatched across banks so I can have an arbitrarily set of regular buttons and toggle buttons. Ideally with an ability to swap them at runtime (like 10s hold of a button will change it from CCButton to CCButtonLatched), is it even possible?

I'm trying to do a simple foot controller with 8 buttons - 6 of them sending CC signals and other two used for a bank selection.
One possibility that I see is to distinct button rows like top 3 buttons will act as toggles (CCButtonLatched) and bottom row will act like a regular buttons, put them into two different bank sets and just switch both of their banks simultaniously. 

But again ideally I want to have ability to mix them across one set like so
```
═══════════   ──╮             ═════════════════
CCButton        │              Button 1
──────────      │  Bank 1      ────────────────
CCButtonLatched │              Button 2
═══════════   ──┤             ═════════════════
CCButton        │              Button 1
──────────      │  Bank 2      ────────────────
CCButton        │              Button 2
═══════════   ──┤             ═════════════════
CCButtonLatched │              Button 1
──────────      │  Bank 3      ────────────────
CCButton        │              Button 2
═══════════   ──┤             ═════════════════
CCButtonLatched │              Button 1
──────────      │  Bank 4      ────────────────
CCButtonLatched │              Button 2
═══════════   ──┤             ═════════════════
```

P.S. Huge thanks for the library! It's a tremendous work and the source code is so neat and inspiring! despite my complete lack of experience with c++

ermyril, 2020-05-22:
UPD: 
I was able to mix different types of buttons across banks (no need to use two separate bank sets as I wrote earlier)
Still don't have a solid understanding how it works though :D

Will tinker around it more and try to change them at runtime

```
Bank<4> bank(4);
...
Bankable::CCButtonLatching toggleButtons[] = {
  {bank, mux.pin(3), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}},
  {bank, mux.pin(4), {MIDI_CC::General_Purpose_Controller_2, CHANNEL_2}},
};

Bankable::CCButton regularButtons[] = {
  {bank, mux.pin(5), {MIDI_CC::General_Purpose_Controller_3, CHANNEL_3}},
  {bank, mux.pin(6), {MIDI_CC::General_Purpose_Controller_4, CHANNEL_4}},
};
```

ermyril, 2020-05-22:
UPD2:
Was able to change the behaviour like so, is this the right way to do it?

```
Bank<4> bank(1);
...

Bankable::CCButtonLatching toggleButtons[] = {
  {bank, mux.pin(3), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}},
};

Bankable::CCButton regularButtons[] = {
  {bank, mux.pin(3), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}},
};

void setup() {
  ...
  regularButtons[0].disable();
  ...
}

void loop() {
  ...

  if (pushbutton.update() == Button::Falling) {
    if (regularButtons[0].isEnabled()) {
      regularButtons[0].disable();
      toggleButtons[0].enable();
    } else {
      regularButtons[0].enable();
      toggleButtons[0].disable();
    }
  }
}

```

tttapa, 2020-05-22:
Banks are used to change the address or channel of a MIDI element, they cannot (at least not directly) change the function of a MIDI element.  
That doesn't mean that you cannot mix different types of MIDI elements in a single bank, you can combine all kinds of elements in a single bank.

To change the function of a button, you can simply create two buttons, one with each function (e.g. latching and non-latching), and enable/disable them accordingly, like you have done in your last code snippet.  
A disadvantage could be that handling bank changes while pressing one of the buttons could become an issue.

An alternative would be to use ManyAddress elements, where you can explicitly set some elements to an invalid address in some banks, this is demonstrated here:
https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d79/One-Pot-Both-PB-and-CC_8ino-example.html
(The example is for potentiometers, but the same applies to other elements as well, including switches/buttons.)

ermyril, 2020-05-24:
@tttapa, thanks!

Tried both variants, ManyAddress seems to be what I need, because when I'm disabling/enabling buttons they change their behaviour across all banks which is not what I want. Although I can try to detect bank changes and maintain their state by enabling/disabling buttons according to some map on every bank change. Haven't thought about it, will try it tomorrow.

Tried to use ManyAddress, and functionality is closest to what I need, but there is some problems: there is no Bankable::ManyAddresses::CCButtonLatched class, I've tried to hack around and implement it but with no luck.
And more significant problem is changing behaviour in runtime, from what I've understood - each element that goes into Bankable::ManyAddresses::NoteButton<4> array(?) is passed to the constructor, so it'll be no luck if I'll try something like that. and I haven't found any methods that allow these addresses to change on runtime

```
Bankable::ManyAddresses::NoteButton<4> toggleButtons[] = {
  {bank, mux.pin(3), {toggleAddresses[0][0], toggleAddresses[0][1], toggleAddresses[0][2], toggleAddresses[0][3]}},
  {bank, mux.pin(4), {toggleAddresses[1][0], toggleAddresses[1][1], toggleAddresses[1][2], toggleAddresses[1][3]}},
  ...
};
```

My second thought was trying to read configuration from eeprom, and initialising Bankable::ManyAddresses::NoteButton<4> toggleButtons[] after reading config, but I didn't had much luck with it either. 

So I'll try approach with detecting bank changes for now

tttapa, 2020-05-26:
> Although I can try to detect bank changes and maintain their state by enabling/disabling buttons according to some map on every bank change.

You can use [`Bank::getSelection()`](https://tttapa.github.io/Control-Surface-doc/Doxygen/db/dbd/classBank.html#a883bbbd61bff47fdc24105ba663d622b) to get the current bank setting.

> there is no Bankable::ManyAddresses::CCButtonLatched class, I've tried to hack around and implement it but with no luck.

I'll try adding it when I find the time, but I have exams right now, so I really don't have time at the moment.

> And more significant problem is changing behaviour in runtime, from what I've understood - each element that goes into Bankable::ManyAddresses::NoteButton<4> array(?) is passed to the constructor, [...]

The idea is that you use the bank functionality to change the addresses.

> My second thought was trying to read configuration from eeprom, and initialising Bankable::ManyAddresses::NoteButton<4> toggleButtons[] after reading config, but I didn't had much luck with it either.

Yes, you can do that, see #129.

ermyril, 2020-05-31:
Thanks!

I've managed to do it via managing state using ```Bank::getSelection()``` 

Thought that I'll finish the whole thing by now, but dealing with protoboards and a few chained 595 shift registers appeared to be more challenging than I anticipated. Cursed the whole "RGB led per button" idea a couple of times already. Seems like a Pareto principle in action :)

closed 2020-06-11 by ermyril