tttapa/Control-Surface #201: Solicitud de datos a daw para faders motorizados

alomdj00, 2020-05-23:
Saludos amigo, gracias por tu gran trabajo.

Estoy haciendo uso de tu completa libreria para crear mi controlador midi DIY. Actualmente me he tropezado con algunas dificultades.

Quisiera obtener la posición de los faders MCU en los daws, he tratado con .getPosicion, getValue y no me ha funcionado. Tal vez obteniendo estos valores sea posible usar faders motorizados.

tttapa, 2020-05-26:
I'll see if I can add it when my exams are over.

alomdj00, 2020-05-27:
Genial, ire avanzando en lo demas, exitos   en tus examenes

4dvn, 2020-11-05:
> I'll see if I can add it when my exams are over.

Me too [image]

tttapa, 2020-11-05:
This has been added to the new-input branch, see https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/d0/d6c/Pitch-Bend-Value_8ino-example.html.

4dvn, 2020-11-06:
> This has been added to the new-input branch, see https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/d0/d6c/Pitch-Bend-Value_8ino-example.html.

Hi Petter, have you update my topic request of pots & buttons matrices banks with 1 button?