tttapa/Control-Surface #202: How to use 14bit pots/encoders?

Kmanggaard, 2020-05-24:
Hello,
Fantastic library! I'm having trouble finding documentation of 14bit implementation. I see the module exists, but don't know how to use it. Can you point me in the right direction?

tttapa, 2020-05-26:
For pitch bend, you can use PBPotentiometer, for 14-bit control change, you can use something like this:
```cpp
#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBDebugMIDI_Interface midi;

class CC14Potentiometer : public MIDIFilteredAnalog<ContinuousCCSender14<10>> {
  public:
    CC14Potentiometer(pin_t analogPin, MIDIAddress address)
      : MIDIFilteredAnalog(analogPin, address, {}) {}
};

// Instantiate a CC14Potentiometer object
CC14Potentiometer potentiometer = {
  A0,                                   // Analog pin connected to potentiometer
  {MIDI_CC::Channel_Volume, CHANNEL_1}, // Channel volume of channel 1
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

Kmanggaard, 2020-05-26:
Many thanks. It was the hi-res CC I was looking for to keep controls on one
channel.

Kind regards / venlig hilsen

Klaus Manggaard