tttapa/Control-Surface #203: DUAL MIDI interface "no match for operator "

Malensky, 2020-05-26:
Hello,

I have successfully built an Ardunio Leonardo-based midi controller with your amazing library, kudos for that! It has 6 pots, 2 momentary switches, a rotary encoder and a bank-selector switch. It is fully fucntional over USB-midi, however wanted it to send midi over 5-din MIDI at the same time. I tried using the DUAL-interface example but it gives me an error while compiling. I am pretty new at this but as far as I understand it does not recognize part of the library... Apologies if I speak nonsense.

You may see the code and the error message below. Sincere thanks for all your work and for any assistance with this!

Kind regards,
Malen

/** 
 * This is the code I used for my specific MIDI controller.  
 * 
 * @boards  Leonardo
 * 
 * Connections
 * -----------
 * 
 * I used an Arduino Leonardo with:
 * - 4 faders on A0-A3
 * - 8 potentiometers on A4-A11
 * - 1 rotary encoder on pins 0-1
 * - 4 toggle switches on pins 2, 3, 5 and 7
 * - 1 toggle switch (for bank select) on pin 11
 * - 1 LED on pin 13
 *
 * Behavior
 * --------
 * 
 * - When bank select is 0, faders 1-4 are channel volumes 1-4, potSide 1-4 are 
 *   pan/balance of channel 1-4, switches 1-4 are mute channel 5-8.
 * - When bank select is 1, faders 1-4 are channel volumes 5-8, potSide 1-4 are 
 *   pan/balance of channel 5-8, switches 1-4 are mute channel 1-4.
 * - PotTop 1-4 are not in the bank, because I use them as effect or EQ
 *   parameters, and they are channel independent.
 * - Enc1 is used for scrolling.
 *
 * Mapping
 * -------
 * 
 * Create a custom mapping using the MIDI learn option in your DAW.
 *
 * Written by PieterP, 07-09-2017
 * https://github.com/tttapa/Control-Surface
 */

#include <Encoder.h>
// Encoder has to be included before Control_Surface
#include <Control_Surface.h>

USBMIDI_Interface usbmidi;
HardwareSerialMIDI_Interface serialmidi = {Serial1, MIDI_BAUD};

// Create a MIDI pipe factory to connect the MIDI interfaces to Control Surface
BidirectionalMIDI_PipeFactory<2> pipes;

// If the jog wheels or other encoders are too slow in your software, increase
// this value.
// (It will be multiplied with the actual speed of the encoder, as the name
// implies.) Default is 1.
const int speedMultiplier = 1;

Bank<2> bank(1); // A bank with four channels, and 2 bank settings

Bankable::CCPotentiometer faders[] = {
  {{bank, BankType::CHANGE_CHANNEL}, A0, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A1, {MIDI_CC::General_Purpose_Controller_2, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A2, {MIDI_CC::General_Purpose_Controller_3, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A3, {MIDI_CC::General_Purpose_Controller_4, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A4, {MIDI_CC::General_Purpose_Controller_5, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A5, {MIDI_CC::General_Purpose_Controller_6, CHANNEL_1}},
[... 368 lines cut ...]

 operator|(TrueMIDI_SinkSource &sinksource,

 ^~~~~~~~

C:\Users\Malen Malenov\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Interfaces/MIDI_Pipes.hpp:545:1: note:   template argument deduction/substitution failed:

C:\Users\Malen Malenov\Desktop\Arduino MIDI Projects\Control-Surface-master\examples\1. MIDI Output\Other\Dr-Mumago-MIDI-Rover\Dr-Mumago-MIDI-Rover.ino:82:3: note:   cannot convert 'CS::Control_Surface' (type 'CS::Control_Surface_') to type 'CS::TrueMIDI_SinkSource&'

   Control_Surface | pipes | serialmidi;

   ^~~~~~~~~~~~~~~

Using library Encoder at version 1.4.1 in folder: C:\Users\Malen Malenov\Documents\Arduino\libraries\Encoder 
Using library Control-Surface-master at version 1.1.1 in folder: C:\Users\Malen Malenov\Documents\Arduino\libraries\Control-Surface-master 
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Using library SoftwareSerial at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial 
Using library MIDIUSB at version 1.0.3 in folder: C:\Users\Malen Malenov\Documents\Arduino\libraries\MIDIUSB 
exit status 1
no match for 'operator|' (operand types are 'CS::Control_Surface_' and 'CS::BidirectionalMIDI_PipeFactory<2> {aka CS::MIDI_PipeFactory<2, std::pair<CS::MIDI_Pipe, CS::MIDI_Pipe> >}')

Invalid library found in C:\Program Files (x86)\Arduino\libraries\arcore-master: no headers files (.h) found in C:\Program Files (x86)\Arduino\libraries\arcore-master
Invalid library found in C:\Program Files (x86)\Arduino\libraries\MIDIUSB: no headers files (.h) found in C:\Program Files (x86)\Arduino\libraries\MIDIUSB
Invalid library found in C:\Program Files (x86)\Arduino\libraries\arcore-master: no headers files (.h) found in C:\Program Files (x86)\Arduino\libraries\arcore-master
Invalid library found in C:\Program Files (x86)\Arduino\libraries\MIDIUSB: no headers files (.h) found in C:\Program Files (x86)\Arduino\libraries\MIDIUSB

tttapa, 2020-05-26:
I'm unable to reproduce this using 28ed279942c5fbe289151901182a303edf46bf5c and Arduino IDE version 1.8.9, AVR Core 1.8.2.

Malensky, 2020-05-26:
Hello, thank you for your quick reply!
So, any guess where it migth go sideways?
Thanks again,
Malen

tttapa, 2020-05-26:
```
C:\Users\Malen Malenov\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Interfaces/MIDI_Pipes.hpp:477:32: note: candidate: CS::BidirectionalMIDI_Pipe& CS::operator|(CS::TrueMIDI_SinkSource&, CS::BidirectionalMIDI_Pipe&)

 inline BidirectionalMIDI_Pipe &operator|(TrueMIDI_SinkSource &sinksource,

                                ^~~~~~~~

C:\Users\Malen Malenov\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Interfaces/MIDI_Pipes.hpp:477:32: note:   no known conversion for argument 1 from 'CS::Control_Surface_' to 'CS::TrueMIDI_SinkSource&'
```
This part of the error message indicates that `Control_Surface_` doesn't inherit from `TrueMIDI_SinkSource`, but clearly, it does in the latest version:
https://github.com/tttapa/Control-Surface/blob/28ed279942c5fbe289151901182a303edf46bf5c/src/Control_Surface/Control_Surface_Class.hpp#L26-L27

The only explanation I can come up with is that you're not using the latest `master` version.

Malensky, 2020-05-26:
Thank you! I will look into it!
Again, brilliant work!
Warmly,
Malen

closed 2020-05-26 by Malensky

Malensky, 2020-05-26:
Yes, just removed and reinstalled my Arduino IDE and the included libraries. It works now!!! Many thanks!!!!

tttapa, 2020-05-26:
Glad to hear!