tttapa/Control-Surface #205: adding led to mute button

benwadub, 2020-05-27:
hi Pieter, i d like to add a led to my mute button, could you tell me how I could write it on this mux where mute respond to cc 94, I d like that my led light when unmute and that it stop when mute is on, thanks
```cpp
CD74HC4067 mux1 = {
  A3,       // numéro de broche de l'arduino
  {2, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};

Bank<2> bank2 = {-1}; // active 2 bank avec 8 adresses par bank
//démarre un multiplexeur pour 8 boutons mute
Bankable::CCButtonLatched<2> buttonmute[] = { //ces lignes déclarent le mux boutons fil rouge
  {{bank2,BankType::CHANGE_ADDRESS}, mux1.pin(7), {94, CHANNEL_1}},//numéro de bank correspondant/indique que le changement de bank change le canal midi (+8 car on a 8 adresses par bank)/pin sur laquelle le controleur est branché/numéro de cc/numéro de canal midi
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(6), {94, CHANNEL_2}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(5), {94, CHANNEL_3}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(4), {94, CHANNEL_4}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(3), {94, CHANNEL_5}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(2), {94, CHANNEL_6}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(1), {94, CHANNEL_7}},
  {{bank2,BankType::CHANGE_ADDRESS},mux1.pin(0), {94, CHANNEL_8}},
};
```

tttapa, 2020-05-27:
If you want the state of the button, use [`CCButtonLatched::getState`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d1d/classBankable_1_1CCButtonLatched.html#a08078bc1b235f535287ad0bcb16d7dad) and use that to turn on/off the LED. If you want the state of the control in software, you can use [`CCValueLED`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/dca/classBankable_1_1CCValueLED.html). It doesn't matter that you're using a multiplexer, the procedure is the same regardless.

benwadub, 2020-05-27:
so if I want a led on mux pin 0 that respond to the midi send from the button on pin 7 I could write it like this?:
```cpp
Bankable::CCButtonLatched::getState<2> buttonmute[] = { //ces lignes déclarent le mux boutons fil rouge
  {{bank2,BankType::CHANGE_ADDRESS}, mux1.pin(7), {94, CHANNEL_1}},
...
{{bank2,BankType::CHANGE_ADDRESS}, mux1.pin(0), {94, CHANNEL_1}},
...
 CCValueLED
```

tttapa, 2020-05-27:
No, you cannot use a multiplexer for outputs.

benwadub, 2020-05-28:
Ok so each led wil use a teensy pin?

tttapa, 2020-06-03:
Yes, unless you use other ExtIO options such as shift registers or other LED drivers.

benwadub, 2020-06-03:
I ll try to investigate and discover the shift register thanks again I ll go deeper in my learning process :-)

benwadub, 2021-11-09:
hi i made my led to get state using mcu protocol, I have a strange thing, when I push a button or toggle with my mouse on the screen the led goes on or off no problem but I can t have more than one led lighting, if I mute track one led 1 light, if I mute track 2 led 2 light but led one become off
cpp```////////////////LEDS
using namespace ExtIO;
const pin_t latchPin = 10;  // Pin connected to ST_CP of 74HC595
const pin_t dataPin  = 11;  // Pin connected to DS of 74HC595
const pin_t clockPin = 13;  // Pin connected to SH_CP of 74HC595
ShiftRegisterOut<8> sreg = { dataPin, clockPin, latchPin, MSBFIRST,  };

  NoteLED leds  [] = {
  {sreg.pin(0), {MCU::MUTE_1, CHANNEL_1}},
  {sreg.pin(1), {MCU::MUTE_2, CHANNEL_1}},
  {sreg.pin(2), {MCU::MUTE_3, CHANNEL_1}},
  {sreg.pin(3), {MCU::MUTE_4, CHANNEL_1}},
  {sreg.pin(4), {MCU::MUTE_5, CHANNEL_1}},
  {sreg.pin(5), {MCU::MUTE_6, CHANNEL_1}},
  {sreg.pin(6), {MCU::MUTE_7, CHANNEL_1}},
  {sreg.pin(7), {MCU::MUTE_8, CHANNEL_1}},

};```
here is how I wrote it

tttapa, 2021-11-13:
This sounds like a problem with your DAW. Try the `USBDebugMIDI_Interface` to make sure that it's not an Arduino problem: send `90 10 7F` `90 11 7F` and the first two LEDs should turn on.

Alternatively, use this example to see what your DAW is sending: https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d85/Mackie-Control-Universal-Reverse-Engineering_8ino-example.html