tttapa/Control-Surface #206: Esp32 pin layout.

bmarshall91, 2020-06-03:
I'm trying to make a 4 footswitch BLE Midi footswitch to send program changes. Just having a problem working out how the pins relate to the pin out diagram for the ESP32 Dev kit 1. 
I know it's probably simple but I'm struggling to find the answer. 

Thanks in advance

tttapa, 2020-06-03:
The Control Surface library uses the standard Arduino pin numbers (i.e. the ones you use with `digitalWrite` etc.). These should be listed on the pinout diagram for your specific board. On some diagrams, they may be prefixed with `GPIO` (general purpose input/output).

For example, if you have a button connected to the pin labeled `GPIO5`, you could use the following in your code:

```cpp
// Instantiate a PCButton that reads the input from a push button and sends out
// a MIDI Program Change message when it's pressed.
PCButton pcBtn = {
  5,                                 // pin
  {MIDI_PC::Steel_Drums, CHANNEL_1}, // address
};
```

bmarshall91, 2020-06-03:
Ah simple.

So if I did:

// Instantiate a selector that reads three buttons and controls the program// changerManyButtonsSelector<3> programSelector = { programChanger, {{2, 3, 4}},};

This would be GPIO pins 2, 3 and 4?

Thank you for taking the time to answer a basic problem.

tttapa, 2020-06-03:
> So if I did:
> ```
> // Instantiate a selector that reads three buttons and controls the program
> // changer
> ManyButtonsSelector<3> programSelector = { programChanger, {{2, 3, 4}},};
> ```
> This would be GPIO pins 2, 3 and 4?

Exactly.

closed 2020-06-04 by bmarshall91