tttapa/Control-Surface #207: How controlling a plugin switch with 3 positions with the same CC

luca-art, 2020-06-03:
Hi,
I spent many hours and days reading this wonderfull library. I just did a Jupiter-8 controller (Arturia plugin) Everything works ok but I could not find out what command I must use if I want to move (in plugin) a switch with 3 positions with the same CC. I could do it with an encoder and a pot but no luck with an on-off-on switch neither with a rotary switch (like those typical eight positions ones)
Could anybody help me? Is it possible to do it?
Thank you very much

tttapa, 2020-06-03:
I think #185 covers what you're after.

tttapa, 2020-06-04:
Don't worry, I don't blame you, I don't expect everyone to dig through hundreds of issues.

luca-art, 2020-06-04:
Hi, I am very happy. Following the post (#185) you said, I get it!
At first I was very frustrated (I have a basic skill in programming) because I thought I would need 3 phisical connectors in the switch (plus ground) for the 3 inputs (2, 3 and 4). Anyway, at last i have modified the hexadecimal value in each input, instead of to put the same value for on and off I put another value 0x40 (for the third position) to off. I can not explain it better (I am from Spain) I show you the lines of code:

CCButton buttons [] = {
  {2, // pin
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}, // MIDI address
  {0x00, 0x40}}, // value when pressed, value when released
  {3, // pin
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}, // MIDI address
  {0x7F, 0x40}}, // value when pressed, value when released
};

I am sorry but I do not remember how to insert code properly
Thank you again

luca-art, 2020-06-04:
Well, to clarify a little bit, the 0x40 value would be the second (middle) position in the plugin switch