tttapa/Control-Surface #208: Example not compiling with Platformio

brennansang, 2020-06-04:
Hey there,
Thanks so much for your work on this, I'm incredibly excited to work with your library. Unfortunately, I'm running into an issue when compiling with Platformio while using a Teensy 3.6.

To try and isolate the issue, I started a new project using just the [Send Midi Notes example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d8d/Send-MIDI-Notes_8ino-example.html) — but I end up getting the same error.

The error that I can't seem to get past is:

`src/main.cpp:13:7: error: 'MIDIAddress' does not name a type`

Everything compiles fine using Teensyduino, but compiling fails using Atom/Platformio. Any idea what I'm missing?

Thanks again for your time,  
Brennan

Here's the example code I'm using.

```cpp
#include <Control_Surface.h>

// Instantiate the MIDI over USB interface
USBMIDI_Interface midi;

// Push button connected between pin 2 and ground.
// Note message is sent when pressed.
Button pushbutton = {2};

using namespace MIDI_Notes;

// MIDI address of the note to send
const MIDIAddress noteAddress = {note(C, 4), CHANNEL_1};
// The velocity of the note events
const uint8_t velocity = 0x7F;

void setup() {
  pushbutton.begin(); // enables internal pull-up
  midi.begin();       // initialize the MIDI interface
}

void loop() {
  midi.update(); // read and handle or discard MIDI input

  pushbutton.update();                              // read the button state
  if (pushbutton.getState() == Button::Falling)     // if the button is pressed
    midi.sendNoteOn(noteAddress, velocity);         // send a note on event
  else if (pushbutton.getState() == Button::Rising) // if the button is released
    midi.sendNoteOff(noteAddress, velocity);        // send a note off event
}

```
Here's my platformio.ini

```[env:teensy]
platform = teensy
board = teensy36
framework = arduino
lib_deps = Control Surface
build_flags = -D USB_MIDI_SERIAL
lib_ignore = MIDIUSB```

Here's the output from when I try to compile the example code.

```Processing teensy (platform: teensy; board: teensy36; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy36.html
PLATFORM: Teensy 4.9.0 > Teensy 3.6
HARDWARE: MK66FX1M0 180MHz, 256KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES:
 - framework-arduinoteensy 1.152.0 (1.52)
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 94 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Control Surface> 1.1.1
|   |-- <SoftwareSerial> 1.0
|   |-- <Encoder>
|   |-- <SPI> 1.0
|   |-- <FastLED> 3.3.1
|   |   |-- <SoftwareSerial> 1.0
|   |   |-- <SPI> 1.0
|   |   |-- <DmxSimple> 3.1
|   |-- <Audio> 1.3
|   |   |-- <SerialFlash>
|   |   |   |-- <SPI> 1.0
|   |   |-- <SPI> 1.0
|   |   |-- <SD> 1.2.2
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
Building in release mode
Compiling .pio/build/teensy/src/main.cpp.o
src/main.cpp:13:7: error: 'MIDIAddress' does not name a type
 const MIDIAddress noteAddress = {note(C, 4), CHANNEL_1};
       ^
src/main.cpp: In function 'void loop()':
src/main.cpp:27:21: error: 'noteAddress' was not declared in this scope
     midi.sendNoteOn(noteAddress, velocity);         // send a note on event
                     ^
src/main.cpp:29:22: error: 'noteAddress' was not declared in this scope
     midi.sendNoteOff(noteAddress, velocity);        // send a note off event
                      ^
*** [.pio/build/teensy/src/main.cpp.o] Error 1
========================== [FAILED] Took 3.60 seconds ==========================```

tttapa, 2020-06-04:
PlatformIO uses the 1.1.1 version, which doesn't have `MIDIAddress` yet. 
You could force PlatformIO to use the master version, or you can use the version-specific documentation and examples here:
https://tttapa.github.io/Control-Surface-doc/index.html

The examples should be included with the library as well.

brennansang, 2020-06-04:
Well, that would explain it. Is there a downside to forcing PlatformIO to use the master version? 

Thanks for your time.

tttapa, 2020-06-04:
Not really, master should work at all times, but the API might not be 100% stable.  
I simply don't have much time at the moment to prepare an "official" release.

DylanBrianBenton, 2020-06-06:
How  do you force PIO to use the latest? i used the below and 1.1.1 is the only one that installs.
platformio lib -g install  https://github.com/tttapa/Control-Surface/archive/master.zip

tttapa, 2020-06-06:
I don't use PIO. Last time I needed it, I simply searched for PIO's Control-Surface folder and manually replaced it by the master version.

agittins, 2022-02-19:
When PlatformIO adds Control-Surface it adds an entry like `tttapa/Control Surface@^1.2.0-4` to `platformio.ini`.

What worked for me was changing that declaration to pull directly from the master branch in the git repo. You probably don't need to add the `master` tag, but I like that it makes it explicit. My `platformio.ini` for a current project looks a bit like this:

```
[env:d1]
platform = espressif8266
board = d1
framework = arduino
lib_deps = 
	#tttapa/Control Surface@^1.2.0-4
	https://github.com/tttapa/Control-Surface.git#master
	lathoub/AppleMIDI@^3.1.2
lib_ignore = MIDIUSB
```