tttapa/Control-Surface #209: Change the values of the Encoder

DaniilAndreevich, 2020-06-05:
#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

using namespace MIDI_Notes;

// Instantiate a CCAbsoluteEncoder object
//CCAbsoluteEncoder enc = {
//  {2, 3},       // pins
//  MIDI_CC::Channel_Volume, // MIDI address (CC number + optional channel)
//  1,            // optional multiplier if the control isn't fast enough
//};

CCRotaryEncoder enc[] = {
  {{2, 3},MIDI_CC::General_Purpose_Controller_1_LSB, 1,}, // pins, MIDI CC number, optional multiplier if the control isn't fast enough
  {{4, 5},MIDI_CC::General_Purpose_Controller_2_LSB, 1,},
  {{6, 7},MIDI_CC::General_Purpose_Controller_3_LSB, 1,},
  {{8, 9},MIDI_CC::General_Purpose_Controller_4_LSB, 1,}
};

NoteButton button[] = {
  {10,{note(C, 4), CHANNEL_1}}, //push button on pin 0 of multiplexer, note C4 on MIDI channel 1
  {11,{note(D, 4), CHANNEL_1}},
  {12,{note(E, 4), CHANNEL_1}},
  {13,{note(F, 4), CHANNEL_1}},
};

// Similarly, for Pitch Bend
//    PBAbsoluteEncoder enc = {
//      {2, 3},    // pins
//      CHANNEL_1, // MIDI channel
//      1,       // large multiplier because Pitch Bend has high resolution
//    };

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
 

Hi all, I use this code for 4 encoders, everything works as I need when turning to the left gives a value of 1 and when turning to the right produces a value of 127 how to change the value 1 to 0 ?

tttapa, 2020-06-05:
I've never come across that particular encoding for CC values. Are you sure that's what you want?

If so, sending relative CC values is handled by the RelativeCCSender class, you can add your own mode if you want to:
https://github.com/tttapa/Control-Surface/blob/master/src/MIDI_Senders/RelativeCCSender.hpp  
Existing modes are explained here: [**relativeCCmode**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d24/RelativeCCSender_8hpp.html#a793b4d05f50a2c3af3ddeeef50e7a71d)  
Alos see the RotaryEncoder.ino example, it shows how to select the mode.

DaniilAndreevich, 2020-06-05:
> Я никогда не сталкивался с этой конкретной кодировкой для значений CC. Ты уверен, что это то, чего ты хочешь?
> 
> Если это так, отправка относительных значений CC обрабатывается классом RelativeCCSender, вы можете добавить свой собственный режим, если хотите:
> https://github.com/tttapa/Control-Surface/blob/master/src/MIDI_Senders/RelativeCCSender.hpp
> Существующие режимы объяснены здесь: [**relativeCCmode**](https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d24/RelativeCCSender_8hpp.html#a793b4d05f50a2c3af3ddeeef50e7a71d)
> Alos видит RotaryEncoder.в Примере ino показано, как выбрать режим.

I'm sorry, but I'm not experienced in this business, Yes, this is what I need, so that when you turn to the left, the value is 0, and to the right, the value is 127, is there any way to correct this value in the library? or some other way of this decision ?

tttapa, 2020-06-06:
You add a new mode to the `relativeCCmode` enum, and handle that case in the `mapRelativeCC` function. You'll have to decide what to do with values smaller and larger than -1 or 1. The parameter `value` of `mapRelativeCC` is the delta value, from -15 to 15, excluding 0. When you turn 4 steps to left, `value` will be -4, for example, and you have to translate it to a CC data value.

DaniilAndreevich, 2020-06-07:
I inserted this command in the library, but it resulted in all values equal
to 0 . how do I make the left value remain 0 and the right value become 127

// send a Control Change MIDI event
            Control_Surface.MIDI().sendCC(address, 0x00);
            delta -= thisDelta;
        }

tttapa, 2020-06-07:
> it resulted in all values equal to 0 

Yes, that's exactly what you told your code to do.

You could try something like this:

```cpp
        while (delta != 0) {
            int8_t thisDelta = delta > 0 ? +1 : -1;
            uint8_t msgVal = delta > 0 ? 0 : 127;
            // send a Control Change MIDI event
            Control_Surface.sendCC(address, msgVal);
            delta -= thisDelta;
        }
```

DaniilAndreevich, 2020-06-07:
Thank you very much, this is exactly what I need