tttapa/Control-Surface #210: SPI Blink Issue

DylanBrianBenton, 2020-06-06:
Hi,

Sorry if this is a really stupid question. Im trying the SPI blink example using a Teensy 4.0 and a 74HC595. I want to be able to use each output on the 595 as a normal output. For some reason i cant get it working, but my question is how can i specify the SPI pins used?  The T4.0 has 3 sets of SPI busses and I would have though they need to be initalised before it worked?

My current pinout is as below, but I've just fried my pin 13 (SCLK) so ill be using the secondary bus.

[https://i.imgur.com/w3SYirE.png](url)

[https://www.pjrc.com/teensy/card10a_rev1_web.png](url)
[https://www.pjrc.com/teensy/card10b_rev1_web.png](url)

```cpp
#include <Arduino_Helpers.h> // Include the Arduino Helpers library.
#include <AH/Hardware/ExtendedInputOutput/SPIShiftRegisterOut.hpp>
 
using namespace ExtIO; // Bring the ExtIO pin functions into your sketch
 
// Instantiate a shift register with the SPI slave select pin as latch pin, most
// significant bit first, and a total of 8 outputs.
SPIShiftRegisterOut<8> sreg = {SS, MSBFIRST};
 
const pin_t ledPin = sreg.pin(0); // first pin of the shift register
 
void setup() {
  sreg.begin();            // Initialize the shift registers
  pinMode(ledPin, OUTPUT); // You don't even need this line, since
                           // shift registers are always outputs
}
 
void loop() {
  // Toggle the state of the LED every 1/2 second
  digitalWrite(ledPin, HIGH);
  delay(500);
  digitalWrite(ledPin, LOW);
  delay(500);
}
```

tttapa, 2020-06-06:
You have to do that through the Teensy libraries, Control Surface simply uses the main SPI object. 

https://www.pjrc.com/teensy/td_libs_SPI.html#altpins

DylanBrianBenton, 2020-12-27:
Hey Mate, 

Sorry just coming back to all these ideas. Would I need to extend SPIShiftRegisterOut to change which bus it used (Teensy 4.x has 3 available) using the PJRC options? If I forked the library would it be easy enough for me to modify the constructor to include the abillity to specify all the other pins?

DylanBrianBenton, 2020-12-27:
Well, if not a way to specify the SPI1 or SPI2 object instead 👍

tttapa, 2020-12-27:
For a quick and dirty workaround, you could change the `SPI` object used here: https://github.com/tttapa/Control-Surface/blob/master/src/AH/Hardware/ExtendedInputOutput/SPIShiftRegisterOut.ipp
You could pass a reference or a pointer to the SPI object to use to the `SPIShiftRegisterOut` constructor and store it as a member variable.

This will be improved in the upcoming 2.0 release, but I don't have much time to work on it until early February.

DylanBrianBenton, 2020-12-27:
Thanks mate. A little out of my depth but I have 2 friends who are programmers so I'll ask them. 

Just to clear it up. I could point SPI.h from here [https://github.com/PaulStoffregen/SPI](url) and then change the constructor to allow it to change to SPI1, SPI2 etc?

tttapa, 2020-12-27:
The quick hack is to replace all occurrences of the `SPI` object by `SPI1` or `SPI2`, e.g. here:
https://github.com/tttapa/Control-Surface/blob/feee931d44d5f20fb7913b171895316e5819549d/src/AH/Hardware/ExtendedInputOutput/SPIShiftRegisterOut.ipp#L19
Replace `SPI.begin();` by `SPI1.begin();`, idem for `SPI.beginTransaction`, `SPI.transfer`, `SPI.endTransaction`.

Alternatively, don't hardcode the `SPI` object, but store a pointer or reference to any object of type `SPIClass` (or `decltype(SPI)` in general) as a member variable of the `SPIShiftRegisterOut` class. You can then initialize this member in the constructor, so the user can specify the SPI interface to use when creating a `SPIShiftRegisterOut` instance.
Explaining C++ pointers and classes is outside of the scope of this issues section, of course, but your programmer friends might be able to help with that.