tttapa/Control-Surface #211: How can I recieve and send MIDI-through messages with BidirectionalMIDI_PipeFactory?

Malensky, 2020-06-06:
Hi, I have attached a MIDI input circuit to my midi controller and I would like to make it recieve and pass on MIDI messages from another one, alongsied its own messages. Could you help me understand how? Thank you!

Here is the code: 

#include <Encoder.h>
// Encoder has to be included before Control_Surface
#include <Control_Surface.h>

USBMIDI_Interface usbmidi;
HardwareSerialMIDI_Interface serialmidi = {Serial1, MIDI_BAUD};

// Create a MIDI pipe factory to connect the MIDI interfaces to Control Surface
BidirectionalMIDI_PipeFactory<2> pipes;

// If the jog wheels or other encoders are too slow in your software, increase
// this value.
// (It will be multiplied with the actual speed of the encoder, as the name
// implies.) Default is 1.
const int speedMultiplier = 1;

Bank<2> bank(1); // A bank with four channels, and 2 bank settings

Bankable::CCPotentiometer faders[] = {
  {{bank, BankType::CHANGE_CHANNEL}, A1, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A2, {MIDI_CC::General_Purpose_Controller_2, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A3, {MIDI_CC::General_Purpose_Controller_3, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A4, {MIDI_CC::General_Purpose_Controller_4, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A0, {MIDI_CC::Channel_Volume, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A5, {MIDI_CC::Pan, CHANNEL_1}},
};

Bankable::CCButton switches[] = {
  {{bank, BankType::CHANGE_CHANNEL}, 4, {MIDI_CC::General_Purpose_Controller_7, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, 12, {MIDI_CC::General_Purpose_Controller_8, CHANNEL_1}},
};

// Instantiate a CCAbsoluteEncoder object
CCAbsoluteEncoder enc = {
  {8, 7},       // pins
  MIDI_CC::General_Purpose_Controller_5, // MIDI address (CC number + optional channel)
  4,            // optional multiplier if the control isn't fast enough
};

SwitchSelector selector = {bank, 11};

void setup() {

  Control_Surface | pipes | usbmidi;
  Control_Surface | pipes | serialmidi;
  
  Control_Surface.begin();
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() { // Refresh all inputs
  Control_Surface.loop();
  digitalWrite(LED_BUILTIN, digitalRead(11));
}

tttapa, 2020-06-06:
The code you posted routes all messages from Control Surface (i.e. the CCPotentiometers etc.) to the serial and USB interfaces, and routes all messages from these interfaces to Control Surface (which doesn't do anything in your case, since you're not using any MIDI Input Elements).

If you want to route data from the serial interface to USB, you have to add another pipe and another route. See [MIDI Routing](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/dc4/MIDI_Pipes-Routing_8ino-example.html).

Malensky, 2020-06-06:
Hi, thank you so much, that solved it! Now it serves all my purposes - the code is below. Do you see any potential problems with those pipes?

Thanks once again!
Malen

#include <Encoder.h>
// Encoder has to be included before Control_Surface
#include <Control_Surface.h>

USBMIDI_Interface midiusb;
HardwareSerialMIDI_Interface midiser = Serial1;

// Create a MIDI pipe factory to connect the MIDI interfaces to Control Surface

MIDI_PipeFactory<6> pipes;

// If the jog wheels or other encoders are too slow in your software, increase
// this value.
// (It will be multiplied with the actual speed of the encoder, as the name
// implies.) Default is 1.
const int speedMultiplier = 1;

Bank<2> bank(1); // A bank with four channels, and 2 bank settings

Bankable::CCPotentiometer faders[] = {
  {{bank, BankType::CHANGE_CHANNEL}, A1, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A2, {MIDI_CC::General_Purpose_Controller_2, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A3, {MIDI_CC::General_Purpose_Controller_3, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A4, {MIDI_CC::General_Purpose_Controller_4, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A0, {MIDI_CC::Channel_Volume, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, A5, {MIDI_CC::Pan, CHANNEL_1}},
};

Bankable::CCButton switches[] = {
  {{bank, BankType::CHANGE_CHANNEL}, 4, {MIDI_CC::General_Purpose_Controller_7, CHANNEL_1}},
  {{bank, BankType::CHANGE_CHANNEL}, 12, {MIDI_CC::General_Purpose_Controller_8, CHANNEL_1}},
};

// Instantiate a CCAbsoluteEncoder object
CCAbsoluteEncoder enc = {
  {8, 7},       // pins
  MIDI_CC::General_Purpose_Controller_5, // MIDI address (CC number + optional channel)
  4,            // optional multiplier if the control isn't fast enough
};

SwitchSelector selector = {bank, 11};

void setup() {
  
  midiser >> pipes >> midiusb; 
  midiser << pipes << midiser; 
  Control_Surface >> pipes >> midiusb;
  Control_Surface << pipes << midiusb;
  Control_Surface >> pipes >> midiser;
  Control_Surface << pipes << midiser;
  
  Control_Surface.begin();
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() { // Refresh all inputs
  Control_Surface.loop();
  digitalWrite(LED_BUILTIN, digitalRead(11));
}

tttapa, 2020-06-07:
The pipes look alright, you have 6 pipes, and you make 6 connections, so that's fine.

Malensky, 2020-06-07:
Yes, thanks! One final thing I cannot quite figure out: I have connected a led to the tx pin to reflect serial activity and it works fine as long as I have the usb connected to a computer, but if it is only powering the led is still. Any ideas? Thank you, promise not to bother you anymore. Best wishes, Malen

tttapa, 2020-06-08:
I'm not sure I understand what you mean. What board are you using?

Malensky, 2020-06-08:
It is Arduino Leonardo, so it is also a USB-MIDI controller. Seems like the controller is getting some sort of feedback from the PC while sending messages over USB-MIDI and the LED blinks on key turn, button press, etc. (the LED is attached to the serial TX pin). When Leonardo is only connected to power supply and sending messages over serial the LED does not blink. What do you think?

tttapa, 2020-06-08:
It could be this issue: https://github.com/arduino-libraries/MIDIUSB/issues/46

Malensky, 2020-06-08:
Thanks, I will follow that through! All your support is much appreciated!
Kind regards,
Malen

closed 2020-06-08 by Malensky