tttapa/Control-Surface #213: Can t find cd74hct4067 through hole

benwadub, 2020-06-08:
Hi, I can t find any 4067 through hole, can another 16 channel mix work with the library without changing anything in code?
Just to learn why is the 4067 used in every midi project? What does it have that other don t have?

tttapa, 2020-06-09:
A through-hole version of a 16-channel multiplexer would be much larger than the SMD version. You can use a breakout board such as [this one](https://www.sparkfun.com/products/9056).

Multiplexers are very simple devices, you should be able to use a different part without issues in most cases, but I can't guarantee anything.

The 4067 is simply the most popular option, which acts as a positive feedback loop, many projects use it, so people trying to reproduce these projects use the same chip, and DIY electronics stores stock them.

benwadub, 2020-06-10:
I use some like this but I wanted to make pcb and Ias I never soldered smd I m a bit worry and jlcpcb are out of stock of these mux to sell pcb with it already soldered! I ll experiment with others to test!
Thanks for the answer!