tttapa/Control-Surface #214: center detent potentiometer and hysteresis

PseudotsugaMensk, 2020-06-12:
Hi,

I am using a potentiometer with a center detent.
The goal is to send a MIDI CC with a value of 63, when the potentiometer is in center position.
I use the map function to assign a "space" in the middle position, which should always correspond to the value 63. But depending on from which direction and speed in which I turn the poti, sometimes the value 62 is getting sent. I suspect, that the hysteresis function, which is applied after the map function, is doing this. 

I would appreciate a suggestion how to properly handle this situation.
Otherwise great library, love it.

### Code for mapping function:
constexpr analog_t minimumValue = 255;
constexpr analog_t maximumValue = 16383 - 255; 

analog_t mappingFunction(analog_t raw) {
  raw = constrain(raw, minimumValue, maximumValue);
  if (raw < 7600)
  {
    return map(raw, 0, 7600, 0, 8192); 
  }
  else if (raw > 8700)
  {
    return map(raw, 8700, 16383, 8192, 16383);
  }
  else
  {
    return 8100;
  }
}

tttapa, 2020-06-12:
This should work:
```cpp
constexpr analog_t minimumValue = 255;
constexpr analog_t maximumValue = 16383 - 255; 
constexpr analog_t margin = 600;

analog_t mappingFunction(analog_t raw) {
  raw = constrain(raw, minimumValue, maximumValue);
  if (raw <= 8192 - margin)
    return map(raw, minimumValue, 8192 - margin, 0, 8192 + 64); 
  else if (raw >= 8192 + margin)
    return map(raw, 8192 + margin, maximumValue, 8192 + 64, 16383);
  else
    return 8192 + 64;
}
```
I don't think it's mathematically 100% correct, but I have an exam tomorrow, so I don't have time to look into it in too much depth right now.

PseudotsugaMensk, 2020-06-12:
It works, and I see what you did there.
Except it results in 64, so I changed your "+ 64" to "- 64" and the result is a steady 63.

Thanks a lot and I wish you the best for your exam.

closed 2020-06-13 by PseudotsugaMensk

tttapa, 2020-06-13:
For future reference: adding or subtracting 64 works because it's half a step, so it's halfway the hysteresis interval.  
The full range is [0, 16384), and a CCPotentiometer has 128 levels, so a single step of the CC value is 16384/128 = 128, and half a step = 128/2 = 64.

josehdx, 2022-09-25:
> This should work:
> 
> ```c++
> constexpr analog_t minimumValue = 255;
> constexpr analog_t maximumValue = 16383 - 255; 
> constexpr analog_t margin = 600;
> 
> analog_t mappingFunction(analog_t raw) {
>   raw = constrain(raw, minimumValue, maximumValue);
>   if (raw <= 8192 - margin)
>     return map(raw, minimumValue, 8192 - margin, 0, 8192 + 64); 
>   else if (raw >= 8192 + margin)
>     return map(raw, 8192 + margin, maximumValue, 8192 + 64, 16383);
>   else
>     return 8192 + 64;
> }
> ```
> 
> I don't think it's mathematically 100% correct, but I have an exam tomorrow, so I don't have time to look into it in too much depth right now.

Where should i inster this code? i would like to use it with the PBPotentiometer class, but im not sure if i should insert it on "PBPotentiometer.hpp" or the "main.cpp"

Thanks

josehdx, 2022-09-25:
By the way, my case is weird because, when i turn the wheel one side or the other, it always return to "7353" or CC57 so i need to add and substract 839 depending the direction