tttapa/Control-Surface #216: Bank function simply don't work on Teensy LC.

SwooshMihu, 2020-06-12:
I have tried using every Bank example provided with the proper wiring yet none of them seem to work.
I will take the simplest example, using the Bank.io example with 2 potentiometer and the 2 buttons connected to the according A0 A1 , 2 and 3 pins and by evaluating the MIDI output using MIDI-OX monitor, the bank function does not work.

Output of MIDI-OX 
[image]

Just to make sure that the wiring is correct, i've uploaded a program to test all of them as inputs and they work just fine.

tttapa, 2020-06-12:
Please use the following sketch to ensure that the selector buttons are working:
```cpp
#include <Control_Surface.h> // Include the Control Surface library

template <setting_t N>
class TestSelectable : public Selectable<N> {
  void select(setting_t newsetting) override {
    Serial << "Selection " << newsetting << endl;
  }
};

TestSelectable<4> sel;

IncrementDecrementSelector<4> selector = {
    sel,
    {2, 3},     // push button pins (increment, decrement)
    Wrap::Wrap, // Wrap around
};

void setup() {
  Serial.begin(115200);
  while(!Serial);
  selector.begin();
}
 
void loop() {
  selector.update();
}
```
It should print the new selection to the serial monitor when a button is pressed.

SwooshMihu, 2020-06-12:
Without pressing any buttons, it does this
[image]
Might it be due to imperfect contacts? I have no clue at this point.
I have tested the wiring of the buttons using  this function
[image]
and it worked fine.

tttapa, 2020-06-12:
Are your buttons normally open or normally closed? If the selection changes without you pressing the buttons, this means that pin 3 is always low, any idea why that would be the case?

SwooshMihu, 2020-06-12:
I am using a basic arduino pushbutton with this wiring
[image]

I have tried with and without the resistor with the same result..

SwooshMihu, 2020-06-12:
Fixed it, it was a wiring problem, thanks for your time!
I have just removed this wire
[image]

tttapa, 2020-06-12:
Leave out the resistor, and connect the bottom lead of the button to ground, not 5V.

tttapa, 2020-06-12:
> I have just removed this wire

That doesn't solve the problem, it leaves your button floating halfway.

SwooshMihu, 2020-06-12:
yep,yep, done as you say and it works. Thank you again!

closed 2020-06-14 by SwooshMihu