tttapa/Control-Surface #218: Unable to use relative rotary encoder

SwooshMihu, 2020-06-14:
I am trying to create a custom MIDI controller that i will use in conjuction with MIDI2LR(https://github.com/rsjaffe/MIDI2LR) to create a physical interface with Adobe Lightroom. Right now i am trying to set up a rotary encoder to use as an alternative to a potentiometer and i am having some difficulties with it. I am trying to define the encoder as being relative, using the RotaryEncoder.ino program( i have tried using  RelativeCCSender::setMode(relativeCCmode::MACKIE_CONTROL_RELATIVE); 
as well as
RelativeCCSender::setMode(relativeCCmode::TWOS_COMPLEMENT);
and i get no output whatsoever, neither in MIDI-OX, nor in MIDI2LR.

If i use the  AbsoluteRotaryEncoder.ino program, instantiating the encoder as a CCAbsoluteEncoder
i still get no output but if i instantiate it as PBAbsoluteEncoder, i do get the following output in MIDI-OX:
[image]
The program used:
#include <Encoder.h> // Include the Encoder library.
#include <Control_Surface.h> // Include the Control Surface library
USBMIDI_Interface midi;
// Similarly, for Pitch Bend
    PBAbsoluteEncoder enc = {
      {2, 3},    // pins
      CHANNEL_1, // MIDI channel
      127,       // large multiplier because Pitch Bend has high resolution
    };
void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}
void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

The rotary encoder that i am using is this : https://www.tme.eu/ro/en/details/ps1010-20/rotary-switches/ctr/ps1010-20-kq15a6-0-000/ , wired according to their diagram.

tttapa, 2020-06-14:
The MIDI-OX screenshot makes me believe you wired the rotary encoder incorrectly.  
From what I can gather on the product page, you have to wire pins 1 and 4 of the encoder to ground, and pins 2 and 3 to pins 2 and 3 of the Arduino.

SwooshMihu, 2020-06-15:
After rewiring, this is the result i get using PitchBend
[image]
Yet no output can be seen when instantiating the encoder as CCAbsoluteEncoder nor as CCRotaryEncoder

tttapa, 2020-06-15:
You could try using the Encoder library directly, it'll be easier to debug: https://github.com/PaulStoffregen/Encoder/blob/master/examples/Basic/Basic.pde 

I'm pretty sure that this is a hardware problem rather than an issue with the libraries involved.

SwooshMihu, 2020-06-15:
Right, using the encoder library directly i can see this on the serial output so i guess it's working fine?
[image]
Yet i get no output from it in MIDI-OX, nor in MIDI2LR

tttapa, 2020-06-15:
No, that's not fine, when you turn the encoder, you should see the position printed, not just -1, 0 and 1. Please post your exact wiring diagram and a picture of the wiring.

SwooshMihu, 2020-06-15:
This is the wiring that i have used to get the -1, 0, 1 result 
[image]
I do not have a multimeter right now to test if the connections are ok. I will also get an EC11 encoder to test that one aswell.

SwooshMihu, 2020-06-19:
I have simply ditched that encoder and picked up a new one with this wiring 
[image]
and it works without problems. thanks for your time

closed 2020-06-19 by SwooshMihu