tttapa/Control-Surface #219: How to get MIDI input values (from DAW)/ 1 LED 2 states

felipegaspari, 2020-06-17:
Hello Pieter/everybody,

your library is amazing, but it is so big it is hard to follow. I'm new to C++ and Arduino, but I've been studying your library for weeks.

I managed to hack the Custom-MIDI-output-element to get the code below working (I'm very bad at this atm, if anyone is thinking as using it as an example). The idea is that pressing a button will MCU::SELECT a track, and when longpressed it
will send MCU::REC_RDY to arm the track.

Question 1: How can I get the value of a MIDI event coming from the DAW to use with my own code (for example MCU::REC_RDY)?

Question 2: NoteValueLed works great, but I'd like to use a single LED for this 2 states; if track is selected, the LED should be ON, and if REC_RDY, it would blink. 

Any orientation would be great. 
Keep up the great work!

Felipe

```cpp
#include <Control_Surface.h>

BEGIN_CS_NAMESPACE

class SelectArmNoteButton : public MIDIOutputElement {
  public:
 
    SelectArmNoteButton(pin_t pin, const MIDIAddress &address, MIDIAddress MIDIRecRdy, uint8_t velocity)
      : button{pin}, address{address}, MIDIRecRdy{MIDIRecRdy}, velocity{velocity} {}

  public:

    // Initialize: enable the pull-up resistor for the button
    // This method is called once by `Control_Surface.begin()`.
    void begin() final override {
      button.begin();

    }

    void update() final override {
      AH::Button::State state = button.update();               // Read the button

      if (state == AH::Button::Released) {
  
      }

      else if (state == AH::Button::Rising) {                // if released
        //   Control_Surface.sendNoteOff(address, velocity); // → note off
        longPressState = Initial;
        sent = false;
      }
      else if (state == AH::Button::Falling) {                      // if pressed
        Control_Surface.sendNoteOn(address, velocity);  // → note on
      }
      else if  (state == AH::Button::Pressed) {
        auto now = millis();
        if ((button.stableTime(now) >= 500) && (sent == false)) {
          Control_Surface.sendNoteOn(MIDIRecRdy, velocity);
          auto now = millis();
          sent = true;

        }
      }
    }

     AH::Button::State getButtonState() const { return button.getState(); }
  
  private:
    AH::Button button;
    const MIDIAddress address;
    const MIDIAddress MIDIRecRdy;

    uint8_t velocity;
    enum {
      Initial,
      LongPress,
    } longPressState = Initial;

    bool sent = false;

};

END_CS_NAMESPACE

// :::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: //

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

using namespace MIDI_Notes;

// Instantiate a MyNoteButton object
SelectArmNoteButton button = {
  34,                       // Push button on pin 5
  {MCU::SELECT_1, CHANNEL_1}, // Note C4 on MIDI channel 1
  {MCU::REC_RDY_1, CHANNEL_1},
  0x7F,                    // Maximum velocity
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface (calls button.begin())
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface (calls button.update())
}
```

tttapa, 2020-06-17:
I think this would be the easiest approach:

```cpp
#include <Control_Surface.h>

USBDebugMIDI_Interface midi;

class CustomLED : Updatable<> {
  public:
    CustomLED(MIDIAddress selected, MIDIAddress armed, pin_t ledPin, unsigned long ledPeriod)
      : selected(selected), armed(armed), ledPin(ledPin), blinkTimer(ledPeriod) {}

    void begin() override {
      ExtIO::pinMode(ledPin, OUTPUT);
      ExtIO::digitalWrite(ledPin, ledState);
    }

    void update() override {
      if (armed.getValue() > 0) {
        if (!blinking) {
          blinkTimer.begin();
          blinking = true;
          ledState = LOW;
        }
        if (blinkTimer) {
          ledState = ledState == HIGH ? LOW : HIGH; // toggle the state
          ExtIO::digitalWrite(ledPin, ledState);
        }
      } else if (selected.getValue() > 0) {
        blinking = false;
        ExtIO::digitalWrite(ledPin, HIGH);
      } else {
        blinking = false;
        ExtIO::digitalWrite(ledPin, LOW);
      }
    }

  private:
    NoteValue selected;
    NoteValue armed;
    pin_t ledPin;
    Timer<millis> blinkTimer;

    PinStatus_t ledState = LOW;
    bool blinking = false;

};

CustomLED led = {
  {MCU::SELECT_1, CHANNEL_1},
  {MCU::REC_RDY_1, CHANNEL_1},
  LED_BUILTIN,
  250, // ms
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

felipegaspari, 2020-06-17:
Amazing! This works perfectly, and it is a great example as well. 
Thank you Pieter!

closed 2020-06-17 by felipegaspari